blueman-project/blueman #1449: Cant connect MS keys from Logitech

jpstijn, 2020-11-30:
blueman: 2.1.2
BlueZ:
Distribution:
Desktop environment: Ubuntu 20.04.1 LTS

I am using the blueman bluetooth manager to connect a MS Anywhere 2 mouse en MS keys Logitech keyboard but i cant pair my MS keys. 

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-11-30:
Don't you mean MX Keys? Then I guess you're missing some firmware, see #1412. I'm really puzzled why some missing firmware doesn't keep the adapter from functioning but from handling the pairing process for that one specific device... 😄

infirit, 2020-11-30:
>I'm really puzzled why some missing firmware doesn't keep the adapter from functioning but from handling the pairing process for that one specific device...

This is what happens with my old BCM20702A1 dongle. If I remember correctly the firmware unlocks Bluetooth 4 functionality. So pairing will work but the moment I tried to use any of the Bluetooth 4 functionality it fails.

cschramm, 2020-12-01:
@infirit What seems to be failing here and in #1412 is pairing the MX Keys and it does not even result in an obvious error (there might be something in the bluetoothd logs, of course), just not show the passkey, but, yes, an existing (fallback) driver / firmware not handling forwarding a certain event, could be the explanation. :thinking:

jpstijn, 2020-12-01:
> Don't you mean MX Keys? Then I guess you're missing some firmware, see #1412. I'm really puzzled why some missing firmware doesn't keep the adapter from functioning but from handling the pairing process for that one specific device... 😄

Yes i did mean MX keys

jpstijn, 2020-12-01:
Its indeed some sort of the same issue i do not get prompt for a key and i just says it failed.

closed 2021-02-08 by github-actions[bot]