blueman-project/blueman #1450: How to properly clear/delete/reset the "Recent Connections" auto-connect-profiles list

ysooqe, 2020-12-02:
blueman: 2.1.3
BlueZ: 5.55
Distribution: Arch Linux
Desktop environment: lxqt + i3

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi there, I would like to ask for some help through this ticket (which I hope is alright according to the README).

I have a question regarding the recent connections plugin. I have installed a new bluetooth adapter (Intel module). Now, there are some auto connect profiles shown in the recent connections list, but hovering over them gives the tooltip "Adapter for this connection is not available", which I suppose is related to the changed bluetooth adapter.

Now the question is, how do I properly clear/delete/reset this list? 

I have tried many things already. Uninstalled and reinstalled everything bluetooth-related, switched back to the old adapter and tried to delete the entries, tried to look through the code of this repo to find where these profiles are loaded from (in order to delete them manually), but unfortunately, none of these approaches got me to a solution.

If there is any device-specific information that is needed in order to achieve this, please let me know so I can provide this information.

infirit, 2020-12-02:
You can't from the blueman UI. I am traveling ATM so can't give you the
commands to clear them until tonight. You can however install dconf-editor
and reset the key under org.blueman.plugins.recentconns.

I'll probably add a reset menu entry at some point to make this easier.

ysooqe, 2020-12-02:
Thank you very much for your answer! Is there a difference between using dconf-editor and using commands?

infirit, 2020-12-02:
> Thank you very much for your answer! Is there a difference between using dconf-editor and using commands?

There are boring technical details but for you as the user the command will give the same result as clicking reset in dconf-editor.

Run the below from a terminal and it will clear all recent connection from the menu.

```
gsettings reset org.blueman.plugins.recentconns recent-connections
```

closed 2021-02-09 by github-actions[bot]