blueman-project/blueman #1451: Bluetooth crashes after pausing audio; requires machine reboot to work again.

chellyxia, 2020-12-03:
* [Confirmed] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

blueman: 2.1.3
BlueZ: 5.55-1
Distribution: Linux 5.8.18-1; MANJARO 20.2
Desktop environment: Xfce 4.14

Bluetooth Hardware: Intel Dual Band Wireless-AC 3168NGW (w/ integrated BT 4.2)

I have an issue with my Bluetooth earphones where whenever audio stops playing, there’s a random 30% chance it will crash the Bluetooth service and corrupt it; meaning I cannot use Bluetooth any longer unless I reboot my entire system. 

More precisely, if I'm watching a Youtube video on Firefox with Audio using my bluetooth earphones, and pause the video, there's a chance either: the bluetooth earphones disconnect (I can reconnect the earphones), or the bluetooth service crashes entirely (I **must** reboot the machine). The behavior isn't consistent. Sometimes this does happen, other times it will keep on working as intended.

This doesn't happen with my other bluetooth headphones; only the earphones.

If I had to guess, I'd say it's some sort of "dirty disconnect" which is corrupting -something-. The same issue occurs if, for example, I let the earphones' battery die without disconnecting them first.

This has been happening for a couple of years (ever since I switched to Manjaro). I've posted on the Manjaro boards a few times, but could never get it solved.

On my end, I'd be happy with a solution which doesn't require me to restart the machine every time this issue occurs.

Rebooting `blueman-applet` returns: 
`00.54.52 WARNING  DiscvManager:109 update_menuitems: warning: Adapter is None`

`systemctl restart bluetooth` does manage to restart the service, but when I try to boot up `blueman-applet`, it immediately crashes again with the same error as above. 

Output of `rfkill list`:

`0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
6: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no`

`Bluetoothctl list` is empty.

Pastebin for `blueman-applet --loglevel debug`: https://pastebin.com/HTKkDezS

cschramm, 2020-12-03:
If I get you right, no adapters are available once this happens, no killswitch is active, restarting bluetoothd does not help, and rebooting the system does. It's an issue in kernel / driver / firmware / hardware then.

I'd expect reloading `btusb` with `rmmod` and `modprobe` to bring back the adapter (most probably unpowered, but `bluetoothctl list` should show it). If not, also try `rmmod btintel`, I guess it stays loaded otherwise. `dmesg` should hopefully have some insights (either when the modules get loaded, especially information about missing firmware, or when the corruption takes place).

chellyxia, 2020-12-04:
Thanks for your reply.

> If I get you right, no adapters are available once this happens, no killswitch is active, restarting bluetoothd does not help, and rebooting the system does.

That is all correct.

`rmmod btusb` and `rmmod btintel` followed by `modprobe btusb` and `modprobe btintel` don't fix the issue sadly. The same errors persist.

`Bluetoothctl list` is remains empty. For reference, when BT is working, `bluetoothctl list` outputs:
 `Controller 14:4F:8A:2C:1A:BD user-pc [default]`

I was able to catch the exact moment the bluetooth errors out in `dmesg`. At 815, I paused a video playing audio. I hope this will be helpful.

```
[  815.529311] usb 1-7: USB disconnect, device number 4  
[  815.582481] audit: type=1130 audit(1607075454.739:119): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'  
[  815.822499] usb 1-7: new full-speed USB device number 5 using xhci_hcd    
[  815.963708] usb 1-7: New USB device found, idVendor=8087, idProduct=0aa7, bcdDevice= 0.01  
[  815.963710] usb 1-7: New USB device strings: Mfr=0, Product=0, SerialNumber=0  
[  818.022529] Bluetooth: hci0: command 0x0c03 tx timeout  
[  821.569249] audit: type=1131 audit(1607075460.726:120): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'  
[  826.128804] Bluetooth: hci0: sending initial HCI reset command failed (-110)  
```

At `rmmod btusb` and `modprobe btusb`, the `dmesg` is:

```
[ 1065.489084] usbcore: deregistering interface driver btusb
[ 1075.220541] usbcore: registered new interface driver btusb
[ 1077.429810] Bluetooth: hci0: command 0x0c03 tx timeout
```

There are no logs for `btintel`.

infirit, 2020-12-04:
It starts at the first line, or maybe a bit earlier. The USB disconnect to me looks like the bluetooth adapter failed and then disappears/reset and never recovers properly. Really sounds like firmware corruption/bug or in the worst case hw failure.

If the firmware got corrupted on disk somehow you can try is uninstalling the linux-firmware package and then reinstalling it.

In either case there is not much we can do about this from blueman, you may want to post on the manjaro/arch forums and see if other users have similar problems.

cschramm, 2020-12-04:
Yep, USB device vanishes, comes back but then the chip is unresponsive (reset command times out).

I'm wondering if there shouldn't be anything about an attempt to re-download the firmware after the USB device comes back.

You could try to enable some kernel debugging:
`echo 8 >/proc/sys/kernel/printk`
`echo 'file btusb.c +p' >/sys/kernel/debug/dynamic_debug/control`
`btintel.c` and `hci_intel.c` might also make sense.

closed 2021-02-11 by github-actions[bot]