blueman-project/blueman #1454: Status Indicator Icon Missing

ShrirajHegde, 2020-12-03:
blueman:2.1.3
BlueZ: 5.55
Distribution: Arch Linux
Desktop environment:Gnome

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Status Indicator Icon Missing  in gnome 3.38 [app indicator support extension](https://extensions.gnome.org/extension/615/appindicator-support/) installed

infirit, 2020-12-03:
Without blueman-applet this is at most an educated guess. Blueman can use
libappindicator but you will need to enable the plugin and have
libappindicator installed.

ShrirajHegde, 2020-12-03:
[Solved]  with libappindicator-gtk3, I am wondering why it isn't even in optional dependencies

cschramm, 2020-12-03:
That is a good question. I guess it should be, yes. I found a short discussion in https://bugs.archlinux.org/task/56900. 🤷

closed 2020-12-03 by cschramm