blueman-project/blueman #1456: New window created whenever a device connects / disconnects

WhyNotHugo, 2020-12-04:
blueman: current master, @borisfaure g212aa8fc
BlueZ: 5.55
Distribution: ArchLinux
Desktop environment: swaywm (Wayland)

Whenever I connect or disconnect a device, a new window is opened indicating the action I've just performed:

[image]

This window lingers indefinitely until I close it (it also responds to <kbd>Esc</kbd>.

infirit, 2020-12-04:
That is the fallback window when people do not have a notification daemon running. This specific notification is from a new plugin "ConnectionNotifier".

@cschramm, should we add a timer that closes these automatically?

WhyNotHugo, 2020-12-04:
A notification deamon is up and running.

It looks like the fallback is triggering _despite_ the daemon being present. Can I provide anything to help figure out what's wrong here?

I also used d-feet and executed `GetCapabilities` on it, maybe this says something...?

[image]

I inspected the code a bit, and there seems to be a `forced_fallback` flag, and for some reason, it's `True` out of the box.

```
> from blueman.main.Config import Config
> Config('org.blueman.general')['notification-daemon']
False
```

---

In any case, I don't think a window in the middle of the screen is a good fallback -- if no notification daemon is present, it might make sense to just show nothing.

WhyNotHugo, 2020-12-04:
Oh, checked with `dconf-editor` and this setting was set to `False` on my system -- I've no idea why, but the issue seemed to be between the keyboard and the back of my chair, sorry for the noise!

Just as a followup, the "Disconnected" notification has the proper icon, but the reconnection one (that shows battery percent), does not.

It also shows 100% battery, but my mouse's non-standard (logitech) API indicates 20%..

cschramm, 2020-12-04:
> should we add a timer that closes these automatically?

@infirit: Would make sense, I guess.

> if no notification daemon is present, it might make sense to just show nothing.

While we started adding some rather verbose notifications lately, they are super important at least for agent information, so just not showing them is not an option in general. We could think about distinguishing something like notifications levels, though.

> the "Disconnected" notification has the proper icon, but the reconnection one (that shows battery percent), does not.

@WhyNotHugo: What do you mean? That one should show the `battery` icon from your icon theme.

> It also shows 100% battery, but my mouse's non-standard (logitech) API indicates 20%..

Oh gosh... Are you saying the device has standard and non-standard battery interfaces and they provide completely different data? :facepalm: Which mouse is it?

WhyNotHugo, 2020-12-04:
> While we started adding some rather verbose notifications lately, they are super important at least for agent information, so just not showing them is not an option in general. We could think about distinguishing something like notifications levels, though.

Why don't you just run a notification-daemon?

I meant to say: I don't think it makes sense to implement notifications _inside_ blueman as a fallback. A notification daemon should be expected to be up and running if someone wants notifications.

> @WhyNotHugo: What do you mean? That one should show the battery icon from your icon theme.

The top notification is the "connected" notification. The bottom one is the "disconnected" notification.

[image]

> Oh gosh... Are you saying the device has standard and non-standard battery interfaces and they provide completely different data? facepalm Which mouse is it?

It's a Logitech Vertical MX. [`solaar`](https://github.com/pwr-Solaar/Solaar) indicates the battery is 20%, and that sounds about right considering how long it's been since I've last charged it.

I used `d-feet` and, effectively, the mouse reports 100% battery (via the proper device's `org.bluez.Battery1`).

I'd say blueman isn't at fault here, the mouse is to blame. Projects like Solaar _do_ have interesting databases of bluetooth devices and slightly non-standard interfaces though :thinking:

infirit, 2020-12-04:
> Oh gosh... Are you saying the device has standard and non-standard battery interfaces and they provide completely different data? facepalm Which mouse is it?

Oh yes but this is with their own wireless dongle/protocol. My M720 Triathlon always reports 100 while the hidpp driver report 55, but I found this also is very unreliable. Battery info is a mess and the main reason I kept pushing back in #947 :cry:

cschramm, 2020-12-04:
@WhyNotHugo: So there's no icon at all? Do you have one of the icon themes that we require, see https://github.com/blueman-project/blueman/blob/master/Dependencies.md?

If those devices always show 100 %, why do they implement the interface at all then? :unamused: My MX Master 2S actually work well. :shrug:

WhyNotHugo, 2020-12-04:
It's Logitech's own protocol, but I'm not using a dongle; it's connected via my laptops Bluetooth.

> So there's no icon at all? 

The "disconnected" notification had the Ivón, the "connected" notification does not.

(I just power cycled the mouse for the screenshot).

I do have `adwaita-icon-theme`.

cschramm, 2020-12-05:
My icon theme knowledge is limited, but it could be that just having adwaita-icon-theme is not enough but it also needs to be the one in use or one of its parent icon themes. :confused: What icon theme do you use?

closed 2021-02-11 by github-actions[bot]