blueman-project/blueman #1459: Blueman install itself in wrong folder

pbakiewicz, 2020-12-05:
blueman: latest
BlueZ:
Distribution: arch
Desktop environment:

In both cases when I'm trying to either install via arch package manager, or directly from source with `sudo make install` blueman installs itself in `/usr/lib/python3.9/site-packages` wheres my python version is 3.8. 

So when I'm trying to run `blueman-manager` it says that `blueman module not found`, and rightly so cos my 3.8 python interpreter looks for packages in `/usr/lib/python3.8/site-packages` not in python3.9.

cschramm, 2020-12-05:
The current arch package is built for 3.9, yes. https://github.com/archlinux/svntogit-community/commit/d364fcc54183b57538ebb90ba52c2c6e5ed09da9#diff-3e341d2d9c67be01819b25b25d5e53ea3cdf3a38d28846cda85a195eb9b7203a (Your system seems to be in an inconsistent state, i.e. the blueman package is up-to-date but the python package is not.)

If you're building with 3.8 at /usr/bin/python3 , it should install to 3.8, yes. Grep for `checking for python` in the configure output.

pbakiewicz, 2020-12-09:
`rg -B 3 'checking for python' config.log `

```478:configure:11127: checking for python
479-configure:11145: found /usr/bin/python
480-configure:11157: result: /usr/bin/python
481:configure:11175: checking for python version
482-configure:11182: result: 3.8
483:configure:11194: checking for python platform
484-configure:11201: result: linux
485:configure:11227: checking for python script directory
486-configure:11262: result: ${prefix}/lib/python3.8/site-packages
487:configure:11271: checking for python extension module directory
```

It's not with my inconsistent state. It indeed found python 3.8. I don't have 3.9 installed. Yet it still wants to put blueman  in `3.9/site-packages`  :/

infirit, 2020-12-09:
I am inclined to agree with @cschramm, most likely you have installed blueman installed twice. My guess is you have the blueman package installed and installed it manually. If you want to build the latest from git I strongly suggest you use below PKGBUILD I use and build you own package from now on. Stick the below in a directory in your home and run ``makepkg -si``.

```
_pkgname=blueman
pkgname=${_pkgname}-git
pkgver=2.1.r421.g212aa8fc
pkgrel=1
pkgdesc="GTK+ Bluetooth Manager"
arch=('x86_64')
url="https://github.com/blueman-project/blueman"
license=('GPL')
depends=('bluez' 'bluez-libs' 'gtk3' 'libnm' 'python-cairo' 'python-gobject')
makedepends=('cython' 'gettext' 'iproute2' 'git')
optdepends=('dnsmasq: Network Access Point (NAP) support'
            'iproute2: Network Access Point (NAP) support'
            'networkmanager: Dial Up Networking (DUN) and Personal Area Networking (PAN) support'
            'pulseaudio-bluetooth: audio devices support')
conflicts=('blueman')

source=('blueman::git+https://github.com/blueman-project/blueman.git')
sha1sums=('SKIP')

pkgver() {
  cd "${_pkgname}"
  git describe --long --tags | sed 's/\([^-]*-g\)/r\1/;s/-/./g'
}

prepare() {
  cd "${srcdir}/${_pkgname}"
  #patch --strip=1 --input="${srcdir}/1382.patch"
  ./autogen.sh

  # Use correct D-Bus config location
  sed -i '/^dbusdir =/ s/sysconfdir/datadir/' data/configs/Makefile.{am,in}
}

build() {
  cd ${_pkgname}
  ./configure \
    --prefix=/usr \
    --sysconfdir=/etc \
    --libexecdir=/usr/lib/${_pkgname} \
    --disable-schemas-compile \
    --with-dhcp-config='/etc/dhcpd.conf'
  make
}

package() {
  cd "${srcdir}/${_pkgname}"
  make DESTDIR="$pkgdir" install
  install -d -o root -g 102 -m 750 "$pkgdir/usr/share/polkit-1/rules.d"
}
```

cschramm, 2020-12-10:
Ah, right, if you have the blueman package installed, the stuff in `/usr/lib/python3.9/site-packages` is completely expected and /usr/bin/blueman probably points to it. `undefined symbol: Py_EnterRecursiveCall` is an indication that the `_blueman.so` was built for 3.9 but you're trying to load it with 3.8.

closed 2021-02-16 by github-actions[bot]