blueman-project/blueman #1460: list of supported types of devices in particular webcam

stefanocudini, 2020-12-05:
hi

this software running on ubuntu can support bluetooth webcam?

cschramm, 2020-12-05:
blueman doesn't have any specific handling for video devices but you should be able to connect to it with blueman. I have no clue if and how such devices are supported in general. Cameras are usually supported via the videodev module and often uvcvideo but I guess non of them have any specific Bluetooth support either.

closed 2020-12-06 by stefanocudini

cschramm, 2020-12-06:
Out of curiosity I checked some code searches (GitHub :roll_eyes:, GitLab, Debian) for references to the video source profile, `00001303-0000-1000-8000-00805F9B34FB`, `VIDEO_SOURCE_PROFILE_ID`, `0x1303`, and found absolutely nothing promising. I'd say there just is no support for Bluetooth cameras for Linux systems. In case you find something, please let us know here.