blueman-project/blueman #1466: Bluetooth Group Network automatic connection inconsistent

rhyek, 2020-12-13:
blueman: 2.1.3-2ubuntu1.1
BlueZ: ?
Distribution: Lubuntu 20.10
Desktop environment:

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-12-13:
Duplicate of #239. There is an auto-connect "backend" in the current master (#853) and a pull request for the frontend parts in #1414.

closed 2020-12-13 by cschramm

rhyek, 2020-12-13:
@cschramm thank you for responding. I'll write this here as I don't want to pollute the linked issues.

I am not very familiar with this project, but I managed to somewhat install from source successfully so I could have the autoconnect plugin. After `make install` I was running into a few issues that I guess just had to do with a mismatched desktop environment or distro to your development environment (like the blueman module install location and others). I had given up and just ran `sudo apt install blueman`, but noticed the plugin was now available! So I guess I have it running.

Anyways, the autoconnect plugin seems to work for the most part. It consistently reconnects to the Group Network service (shared internet) when running a power cycle on the minipc, but disabling bluetooth on my MBP for ~20s, then reenabling it works, but it takes around 4m 30s and not the 60s mentioned in the plugin description. I tested this twice. In these cases I find it is better for me to do a power cycle on the mini pc and have the auto reconnect work immediately.

cschramm, 2020-12-14:
The plugin currently tries to auto-connect on start and every minute (as there's no good way to get to know when the remote device becomes available).

It would make perfect sense to also try it when an adapter becomes available, yes.

Regarding the 4, 5 minutes it takes after re-enabling Bluetooth: Can you please provide the debug output of blueman-applet for that?