blueman-project/blueman #1471: problem with telephoning

p-m, 2020-12-16:
blueman: 2.1.4
BlueZ: 5.48
Distribution: openSUSE Leap-15.2
Desktop environment: Fvwm

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi,

I would like to connect my Nokia N9 to my desktop-system for telephoning.
I can connect it with blueman-manager and audio from the N9-musicplayer plays nicely on my desktop. But when using the phone, bluetooth is not used. (In my car, it's exactly the other way round...)
How could I telephone via bluetooth please?

TIA for any hints, Peter

cschramm, 2020-12-17:
Different profiles need to be used for that. For the phone it's HFP / HSP, for music it's A2DP.

The thing that handles the audio connection is probably pulseaudio and you can switch the profiles it uses. This is also available in blueman-manager by right-clicking on the device.

(Regarding the car, it's rather typical for older / cheaper systems to only support HFP / HSP and no A2DP.)

p-m, 2020-12-17:
Thanks for your comments. It seems, that the car provides the HFP profile, but blueman does not.
And I don't see any possibility to activate HFP neither in blueman-manager nor in pavucontrol...
Where could I find it please?
TIA for any hints, Peter

cschramm, 2020-12-17:
As I said, it's pulseaudio that does all the audio handling. pavucontrol has the profile selection in the `Configuration` tab and blueman-manager provides it as an `Audio Profile` submenu in the device menu.

p-m, 2020-12-18:
I was confused because the N9 never shows HFP in the list of supported profiles for my desktop, only HSP.
And there is no "Audio Profile" submenu in my blueman-manager version...
But after re-pairing the smartphone and using pavucontrol for switching to HSP/HFP it seems to work now.
So, thanks for your efforts, Peter

closed 2020-12-18 by cschramm