blueman-project/blueman #1475: Pairing of audio device (Teufel Bluetooth Headset) fails

chgauert, 2020-12-23:
blueman: 2.1.4
BlueZ: 5.55
Distribution: LFS
Desktop environment: XFCE

Pairing of audio device (Teufel Bluetooth Headset) fails 
Blueman tries to read the pin from pin-code-database, there it returns the pin value for the one audio device defined there (max:4) - based on the value it creates a random value and uses this to pair with the headset.
But the Teufel Bluetooth Headset expects 0000 as pin.
I guess the code in BluezAgent.py using PIN_SEARCHES is not that ideal - because it gets down to just look for same type and there it gets the wrong value in this case
How is this supposed to work correctly? 
If i add the value for the new device to the pin-code-database its working as expected - but i would either expect a hint to the user to add the device to the database or some kind of fallback mechanism i.e. if code is not found show the Pin-Request dialog. In this case here the dialog is not shown because max:4 is returned as pin due to the one existing audio device

* [ x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-12-23:
That is a very good question. :sweat_smile: The search paths at https://github.com/blueman-project/blueman/blob/99e78b74aace7203efd99031f615041c940a1884/blueman/main/applet/BluezAgent.py#L60 should probably specify that the attributes that are not matched must not be present (unfortunately ElementTree does not support the XPath expression for that). At least it does not make sense to match an entry by type only if it also has name or oui specified.

I wonder why this did not bite us yet as the type-only search path should match in many cases. (Well, the pin code lookup was broken in 2.1.beta1 - 2.1.3. I wonder why that did not bite us as well.) The answer probably lies in the cases where the PIN lookup even comes into play and that should be legacy pairing (pre Bluetooth 2.1) with a device that has a display and either no actual keyboard or is the initiator if it has a keyboard. That is where it gets weird as the types contained in our database file are keyboard, mouse, audio, printer, network, and joypad. I would not typically expect such devices (except printer, maybe, but Bluetooth printers are rare anyway) to actually have a display.

@infirit?

infirit, 2021-03-02:
A lot of time BlueZ's autopair plugin will handle the cases where it the pin is 0000. I tried to replicate what gnome-bluetooth does but I obviously failed :disappointed: . I did test it with the devices I had at hand at the time.

@chgauert can you post the full log from blueman-applet please as it shows what type of authoerisation bluemans agent is requested to perform and what it found in the pindb.

PR #1614 (2021-11-08): Drop PIN database
This clearly seems to do more harm than good:

* The special values NULL, KEYBOARD, and ICADE do not actually get handled but provided as strings.
* The lookups are too unspecific, so that any devices that are classified as mouse, audio, or network matches an entry typically not made for it.
* On a match, the user cannot enter a PIN even if the automatic one turns out to be wrong.

On the other hand it does not seem very relevant:

* The whole mechanism is only for Legacy Pairing (Bluetooth <= 2.0) with devices that use a fixed or random PIN.
* Providing 0000 (or any 4-digit  [... cut]
commit 4238d43f (2021-11-08): Drop PIN database

This clearly seems to do more harm than good:

* The special values NULL, KEYBOARD, and ICADE do not actually get handled but provided as strings.
* The lookups are too unspecific, so that any devices that are classified as mouse, audio, or network matches an entry typically not made for it.
* On a match, the user cannot enter a PIN even if the automatic one turns out to be wrong.

On the other hand it does not seem very relevant:

* The whole mechanism is only for Legacy Pairing (Bluetooth <= 2.0) with devices that use a fixed or random PIN.
* Providing 0000 (or any 4-digit [... cut]
commit 5e597913 (2022-02-08): Drop PIN database

This clearly seems to do more harm than good:

* The special values NULL, KEYBOARD, and ICADE do not actually get handled but provided as strings.
* The lookups are too unspecific, so that any devices that are classified as mouse, audio, or network matches an entry typically not made for it.
* On a match, the user cannot enter a PIN even if the automatic one turns out to be wrong.

On the other hand it does not seem very relevant:

* The whole mechanism is only for Legacy Pairing (Bluetooth <= 2.0) with devices that use a fixed or random PIN.
* Providing 0000 (or any 4-digit [... cut]
closed 2021-11-08 by cschramm