blueman-project/blueman #1483: Failure when attempting to pair Logitech Mouse M535

Hector-A85, 2021-01-17:
blueman: 2.08-1+deb10u1
BlueZ: 5.50-1.2~deb10u1
Distribution: Debian 10.7 Buster
Desktop environment: LXDE

Please find attached a file with the CLI output of bluetoothctl. As you can see the M535 Mouse is discovered, reported "not connected" and immediately deleted. This output confirms the GUI behaviour which constantly flashes the mouse ID as buttons are pressed or the mouse moved, but never stops to make it available for pairing. 

hector@hector-XPS13-9350:~$ sudo systemctl start bluetooth.service
[sudo] password for hector: 
hector@hector-XPS13-9350:~$ bluetoothctl
Agent registered
[bluetooth]# scan on
Discovery started
[CHG] Controller 30:52:CB:80:3F:CE Discovering: yes
[bluetooth]# trust 30:52:CB:80:3F:CE
Device 30:52:CB:80:3F:CE not available
[bluetooth]# scan on
Failed to start discovery: org.bluez.Error.InProgress
[NEW] Device 34:88:5D:B4:CC:01 Bluetooth Mouse M336/M337/M535
[CHG] Device 34:88:5D:B4:CC:01 Connected: no
[DEL] Device 34:88:5D:B4:CC:01 Bluetooth Mouse M336/M337/M535
[NEW] Device 34:88:5D:B4:CC:01 Bluetooth Mouse M336/M337/M535
[CHG] Device 34:88:5D:B4:CC:01 Connected: no
[DEL] Device 34:88:5D:B4:CC:01 Bluetooth Mouse M336/M337/M535
[NEW] Device 34:88:5D:B4:CC:01 Bluetooth Mouse M336/M337/M535
[CHG] Device 34:88:5D:B4:CC:01 Connected: no
[DEL] Device 34:88:5D:B4:CC:01 Bluetooth Mouse M336/M337/M535
[bluetooth]# trust 34:88:5D:B4:CC:01
Device 34:88:5D:B4:CC:01 not available
[bluetooth]# pair 34:88:5D:B4:CC:01
Device 34:88:5D:B4:CC:01 not available
[bluetooth]# 

As you can see above, both the Controller and the Device are reported as "not available". Am I missing something?

Will appreciate all help you can provide.

Best regards

Hector Acevedo

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2021-01-17:
Passing the controller to the `trust` command cannot work as it expects a device. When you pass the device, it's not know as it just got removed again.

I don't remember such a behavior where the device immediately gets removed again but I'd expect to find a first hint in the bluetoothd log or maybe the kernel log.

Hector-A85, 2021-01-19:
Thanks for replying and your suggestions. I take that you mean the contents of /lib/systemd/system/bluetooth.service so here it goes:

[Unit]
Description=Bluetooth service
Documentation=man:bluetoothd(8)
ConditionPathIsDirectory=/sys/class/bluetooth

[Service]
Type=dbus
BusName=org.bluez
ExecStart=/usr/lib/bluetooth/bluetoothd
NotifyAccess=main
#WatchdogSec=10
#Restart=on-failure
CapabilityBoundingSet=CAP_NET_ADMIN CAP_NET_BIND_SERVICE
LimitNPROC=1
ProtectHome=true
ProtectSystem=full

[Install]
WantedBy=bluetooth.target
Alias=dbus-org.bluez.service

If this is not what I should be looking for, please provide some directions to find the correct information.

cschramm, 2021-01-20:
bluetoothd log should be available on your system via `sudo journalctl -u bluetooth.service`, kernel log is typically `sudo dmesg`.

Hector-A85, 2021-01-20:
Thanks. Here goes bluetooth log. Kernel log is very long and since the bluetooth log is already showing an error with the sap driver and sap-server maybe this is enough at this very moment. Please advise if I have to also send the kernel log.

-- Logs begin at Wed 2021-01-20 20:21:30 -03, end at Wed 2021-01-20 20:30:06 -03. --
Jan 20 20:21:31 hector-XPS13-9350 systemd[1]: Starting Bluetooth service...
Jan 20 20:21:31 hector-XPS13-9350 bluetoothd[560]: Bluetooth daemon 5.50
Jan 20 20:21:31 hector-XPS13-9350 systemd[1]: Started Bluetooth service.
Jan 20 20:21:31 hector-XPS13-9350 bluetoothd[560]: Starting SDP server
Jan 20 20:21:31 hector-XPS13-9350 bluetoothd[560]: Bluetooth management interface 1.14 initialized
Jan 20 20:21:31 hector-XPS13-9350 bluetoothd[560]: Sap driver initialization failed.
Jan 20 20:21:31 hector-XPS13-9350 bluetoothd[560]: sap-server: Operation not permitted (1)
Jan 20 20:21:44 hector-XPS13-9350 bluetoothd[560]: Endpoint registered: sender=:1.62 path=/MediaEndpoint/A2DPSource
Jan 20 20:21:44 hector-XPS13-9350 bluetoothd[560]: Endpoint registered: sender=:1.62 path=/MediaEndpoint/A2DPSink

cschramm, 2021-01-21:
Did you try to to pair there? So nothing shows up? Nothing in dmesg as well?

Then I'd try with debugging enabled in bluetoothd, something like `sudo systemctl stop bluetooth.service`, then `sudo bluetoothd -d -n`.

Hector-A85, 2021-01-22:
Used dmesg to access kernel log. Only references to bluetooth as follows:

[    5.355836] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.355839] Bluetooth: BNEP filters: protocol multicast
[    5.355845] Bluetooth: BNEP socket layer initialized

Then debugging bluetooth.d yielded:

hector@hector-XPS13-9350:~$ sudo systemctl stop bluetooth.service
hector@hector-XPS13-9350:~$ sudo bluetoothd -d -n
bluetoothd[1484]: Bluetooth daemon 5.50
bluetoothd[1484]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[1484]: src/main.c:parse_config() Key file does not have key “MinEncKeySize” in group “GATT”
D-Bus setup failed: Name already in use
bluetoothd[1484]: Unable to get on D-Bus
hector@hector-XPS13-9350:~$

Hector-A85, 2021-01-27:
Any comments on the above? Thanks

cschramm, 2021-01-27:
`D-Bus setup failed: Name already in use` looks like there was a bluetoothd running. May automatically start again for whatever reason. You can try to temporarily disable it (`sudo systemctl disable bluetooth.service`) before stopping to avoid that.

Hector-A85, 2021-01-27:
Any news on the above? Thanks

Hector-A85, 2021-01-27:
Sorry I asked again. Missed your reply. Here goes what I did:

hector@hector-XPS13-9350:~$ sudo systemctl disable bluetooth.service
[sudo] password for hector: 
Synchronizing state of bluetooth.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable bluetooth
Removed /etc/systemd/system/dbus-org.bluez.service.
Removed /etc/systemd/system/bluetooth.target.wants/bluetooth.service.
hector@hector-XPS13-9350:~$ 

However, the Bluetooth icon still appears in the System Tray and right-clicking on it shows previous menu unchanged. Moreover, behavior is still the same. Do I have to reboot after the above commands?

Hector-A85, 2021-01-27:
I shutdown the computer and after restarting the bluetooth icon was gone and service unavailable. Then restarted it with:

sudo systemctl restart bluetooth

Nothing changed and the mouse behaves just as before. Pairing is impossible as the appearance is too brief and assumedly is deleted immediately.

cschramm, 2021-01-28:
`sudo systemctl disable bluetooth.service` to disable the daemon, then `sudo systemctl stop bluetooth.service` to stop it, then `sudo bluetoothd -d -n` to start it in foreground, try the process where the device keeps vanishing from bluetoothctl and blueman-manager and check the output. When you're done `sudo systemctl enable bluetooth.service` to re-enable the service and `sudo systemctl start bluetooth.service` to immediately start if it that's not done automagically anyway.

Hector-A85, 2021-01-28:
Tried all the above, no success. Here's the relevant output from the foreground:

bluetoothd[5419]: src/adapter.c:connected_callback() hci0 device 34:88:5D:B4:CC:01 connected eir_len 37
bluetoothd[5419]: src/device.c:device_create() dst 34:88:5D:B4:CC:01
bluetoothd[5419]: src/device.c:device_new() address 34:88:5D:B4:CC:01
bluetoothd[5419]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_34_88_5D_B4_CC_01
bluetoothd[5419]: src/device.c:device_set_class() /org/bluez/hci0/dev_34_88_5D_B4_CC_01 0x000580
bluetoothd[5419]: src/adapter.c:dev_disconnected() Device 34:88:5D:B4:CC:01 disconnected, reason 3
bluetoothd[5419]: src/adapter.c:adapter_remove_connection() 
bluetoothd[5419]: src/adapter.c:adapter_remove_connection() Removing temporary device /org/bluez/hci0/dev_34_88_5D_B4_CC_01
bluetoothd[5419]: src/device.c:device_remove() Removing device /org/bluez/hci0/dev_34_88_5D_B4_CC_01
bluetoothd[5419]: src/device.c:btd_device_unref() Freeing device /org/bluez/hci0/dev_34_88_5D_B4_CC_01
bluetoothd[5419]: src/device.c:device_free() 0x564c5266bfc0
bluetoothd[5419]: plugins/policy.c:disconnect_cb() reason 3
bluetoothd[5419]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 34:88:5D:B4:CC:01 type 0 status 0xe
bluetoothd[5419]: src/adapter.c:resume_discovery() 

This keeps repeating for every movement of the mouse. The two last commands restart the service but nothing has changed for better.

I have noticed that the foreground output contains several lines that refer to Network Manager. I do not use it and use Wicd instead.

cschramm, 2021-01-29:
This is exactly what we wanted to see. I'm just giving you directions how to get closer to your problem, so it's completely expected that the problem does not magically go away.

The log shows disconnection reason 3. That means that the remote device terminated the connection.

What I don't really get is that a connection attempt is immediately initiated. From the log I'd say it's the mouse that does it. You do set it into pairing mode beforehand, don't you?

Hector-A85, 2021-01-29:
Yes, I was able to pair it and use quite some time ago. Then I had the misfortune to reinstall Debian (due to another problems) and after that the reported problem appeared.

Hector-A85, 2021-02-03:
Hello Christopher,

Any ideas on how to deal with this issue? Thanks

cschramm, 2021-02-03:
As it looks like the mouse is trying to connect, it seems to know your computer and I guess it's even paired from its perspective. I'd try to find a way to reset it or temporarily pair with another computer if nothing else is available.

Hector-A85, 2021-02-09:
Hello Christopher!
Good news! The mouse is now working perfectly. Your last tip pointed me to do some googling and found that these mice have a reset button on the underside. You have to turn it on and press the reset button for some time (I did it for about 20 seconds). After that, turn it off and on again. Blueman will recognize it immediately and I was able to pair and trust it. Finally turn the mouse off and on again.
Please mark this query as solved and many thanks for pointing me in the right direction.
Hector

closed 2021-02-09 by cschramm