blueman-project/blueman #1486: Blueman-Applet 2 tray icons

MartinX3, 2021-01-24:
blueman: 2.1.4
BlueZ: 5.55
Distribution: Arch Linux / EndeavourOS
Desktop environment: XFCE 4.16

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

hkleinsorgen, 2021-01-24:
Same issue here, slightly different environment:
blueman: 2.1.2-1ubuntu0.2
bluez: 5.53-0ubuntu3
Distro: Linux Mint 20 Ulyana
Desktop: Cinnamon 4.6.7

`ps -Af |  grep blueman`
```
1762    1418  0 12:04 ?        00:00:00 /usr/bin/python3 /usr/bin/blueman-applet
1858       1  0 12:04 ?        00:00:00 /usr/bin/python3 /usr/bin/blueman-tray
1864       1  0 12:04 ?        00:00:00 /usr/bin/python3 /usr/bin/blueman-tray
```
So only one applet, but two tray processes. Killing one of the blueman-tray processes gets rid of the extra icon.

Killing the applet and starting it manually results in the same log as above, and only one tray process

MartinX3, 2021-01-24:
@hkleinsorgen If I read the other bug ticket correctly there was a fix in 2.1.4?
If yes, maybe your problem is fixed if you update blueman/bluez
Or you suffer from my problem existing on the latest version for my linux distribution.

hkleinsorgen, 2021-01-24:
Ah ok, sorry, didn't see that a similar issue was reported for 2.1.2. Can't upgrade to 2.1.4 yet due to missing dependencies (Python 3.9), so I can't test this newer version.

MartinX3, 2021-01-24:
What I see now is that it isn't always duplicated.
This time I got a boot without it, while I got it previously 4 times straight.
Seems to be somehow random.

cschramm, 2021-01-24:
Unfortunately the mechanism to enforce single instances in blueman 2.1, namely [the `check_single_instance` function](https://github.com/blueman-project/blueman/blob/2-1-stable/blueman/Functions.py#L292), suffers from race conditions. In current master this is replaced completely as it uses GApplications now (via https://github.com/blueman-project/blueman/pull/1304).

It think there was some path where you end up with two icons after initially starting a single blueman-applet process, but I'm not 100 % sure. Did you check if there might be competing autostart mechanisms? You can just disable the autostart entry in your desktop settings and see if blueman still starts on boot. Do you use the AppIndicator extension? Does it influence the behavior if you disable it?

MartinX3, 2021-01-24:
Sadly if I deactivate the autostart entry I have no applet and if I remove the StatusTray I have no icons at all :)
If `AppIndicator` is something else (The plugins are translated into my german language) I think I never used it.

cschramm, 2021-01-25:
I mean the blueman-applet plugin. The icon's right click menu has an item "Erweiterungen" and the first one in the list is AppIndicator. If it's activated, blueman uses libappindicator to show an icon. If not, it uses a GtkStatusIcon. That's also the fallback if an AppIndicator icon does not work. Using one or the other might or might not have an influence on the behavior but I do not really expect any.

MartinX3, 2021-01-30:
Sadly even with `AppIndicator` activated it worked now the last days as intended. :(

MartinX3, 2021-04-01:
/remove stale

closed 2021-04-09 by github-actions[bot]