blueman-project/blueman #1489: Failed to connect to Keychron keyboard

apiraino, 2021-01-31:
blueman: 2.1.4-1+b1
BlueZ: from dkpg:
- bluez 5.55-3
- bluez-firmware 1.2-4
- bluez-hcidump 5.55-3
- bluez-obexd 5.55-3
- bluez-tools 2.0~20170911.0.7cb788c-4

Distribution: debian bullseye 10.7
Desktop environment: wayland/sway

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The device is found but cannot pair. Here's the info
```
[bluetooth]# info DC:2C:26:EC:D5:EF
Device DC:2C:26:EC:D5:EF (public)
	Name: Keychron K2
	Alias: Keychron K2
	Class: 0x00000540
	Icon: input-keyboard
	Paired: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
```

bluetoothd logs: https://paste.debian.net/1183456/

The keyboard has a double way to connect: USB-c and bluetooth. When used via USB-C works fine.

How do I debug this? How do I understand the error? What does an `connect_failed_callback()` exiting with `status 4` mean?

thanks

cschramm, 2021-02-03:
Doesn't mean anything specific. Should be just `MGMT_STATUS_CONNECT_FAILED` and is used for any unspecific connection error (https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/net/bluetooth/mgmt.c?id=e6ed8b78eae53788c5e80304b401c23896f86287#n2982). Did you check dmesg yet? In he past we've seen the error e.g. in cases where firmware was missing.

apiraino, 2021-02-10:
hi @cschramm thanks for the feedback. I have no other useful log to report. 

I had this keyboard successfully paired it until a couple of weeks ago on my old Ubuntu. Then I've reinstalled the OS (now a Debian) and I can't pair it anymore, so I can exclude a hardware issue. Might be a package missing? Who knows, errors are totally non descriptive and I can't help a lot in debugging.

I wonder why I can't pair it anymore, do I have to "reset" the hardware, like in [this other issue](https://github.com/blueman-project/blueman/issues/1483#issuecomment-776201363)? I would do that only if necessary, because I'd lose the customizatons.

Should I try reaching out the Bluez folks? Their [mailing list](http://www.bluez.org/development/lists/) seems for developers and patches only, the other option is on IRC.

Thanks again for your help, really appreciated :+1:

cschramm, 2021-02-10:
There are cases where stuff is missing but you should usually get something out of `dmesg` like https://github.com/blueman-project/blueman/issues/1115#issuecomment-541458959.

If there's nothing in dmesg so far, I'd give enabling debug messages from the kernel parts a try with something like `echo 'module bluetooth +p' | sudo tee /sys/kernel/debug/dynamic_debug/control` (or for any `bt*` module).

apiraino, 2021-02-10:
ok I should have dumped some debug. The messages are obscure, not like what I expected so (I might have done something wrong or have a different configuration). 

The log starts shortly before the keyboard is probed and stops right after the pairing fails with status 4. If you search for the device address (`dc:2c:26:ec:d5:ef`) you can see the relevant events but I can't glean more than that.

do you see anything interesting?

<details><summary>log dump</summary>

[52479.860677] channel 2 len 60
[52479.887914] hci0 hcon 00000000b159f556 handle 256
[52479.887921] hcon 00000000b159f556
[52479.920121] hci0 evt 2
[52479.920237] hci0 orig refcnt 9
[52479.920256] hci0 orig refcnt 8
[52479.920300] length 0
[52480.204438] hci0 urb 000000000662b560 status 0 count 31
[52480.204636] hci0
[52480.204649] hci0 Event packet
[52480.204669] channel 3 len 47
[52480.204702] channel 2 len 53
[52480.204882] sock 000000007bf75a72, sk 000000004458fd39
[52480.205069] sock 0000000000be67f0 sk 00000000702bbe49
[52480.205080] channel 2 len 119
[52480.205191] sock 0000000000be67f0 sk 00000000702bbe49
[52480.205319] hci0 urb 000000000662b560 status 0 count 14
[52480.205374] channel 2 len 61
[52480.205462] sock 0000000000be67f0 sk 00000000702bbe49
[52480.205467] hci0
[52480.205515] hci0 Event packet
[52480.205575] channel 2 len 59
[52480.205677] channel 3 len 37
[52480.205686] channel 2 len 43
[52480.205792] sock 000000007bf75a72, sk 000000004458fd39
[52480.205943] sock 0000000000be67f0 sk 00000000702bbe49
[52480.205949] channel 2 len 119
[52480.206062] sock 0000000000be67f0 sk 00000000702bbe49
[52480.206213] channel 2 len 70
[52480.206317] sock 0000000000be67f0 sk 00000000702bbe49
[52480.206418] channel 2 len 71
[52480.206619] sock 0000000000be67f0 sk 00000000702bbe49
[52480.206625] channel 2 len 99
[52480.206796] sock 0000000000be67f0 sk 00000000702bbe49
[52480.206895] channel 2 len 61
[52480.206972] sock 0000000000be67f0 sk 00000000702bbe49
[52480.207141] channel 2 len 70
[52480.207285] sock 0000000000be67f0 sk 00000000702bbe49
[52480.207503] channel 2 len 66
[52480.207554] sock 0000000000be67f0 sk 00000000702bbe49
[52480.207748] channel 2 len 60
[52480.230470] hci0 urb 000000000662b560 status 0 count 31
[52480.230625] hci0
[52480.230631] hci0 Event packet
[52480.231264] hci0 urb 000000000662b560 status 0 count 14
[52480.231299] hci0
[52480.231303] hci0 Event packet
[52480.231351] channel 3 len 37
[52480.231409] channel 2 len 43
[52480.231545] sock 000000007bf75a72, sk 000000004458fd39
[52480.231639] sock 0000000000be67f0 sk 00000000702bbe49
[52480.231647] channel 2 len 119
[52480.231702] sock 0000000000be67f0 sk 00000000702bbe49
[52480.231740] channel 2 len 61
[52480.231763] sock 0000000000be67f0 sk 00000000702bbe49
[52480.231766] channel 2 len 60
[52480.323597] hci0 urb 000000000662b560 status 0 count 45
[52480.323810] hci0
[52480.323842] hci0 Event packet
[52480.323858] channel 3 len 51
[52480.323983] channel 2 len 57
[52480.324180] sock 000000007bf75a72, sk 000000004458fd39
[... 4142 lines cut ...]
[52580.945193] sock 0000000000be67f0 sk 00000000702bbe49
[52580.945194] channel 2 len 53
[52580.945203] sock 0000000000be67f0 sk 00000000702bbe49
[52580.945205] channel 2 len 60
[52580.945212] sock 0000000000be67f0 sk 00000000702bbe49
[52580.945214] channel 2 len 60
[52580.945240] sock 000000007bf75a72 sk 000000004458fd39
[52580.945242] got 13 bytes
[52580.945244] channel 2 len 19
[52580.945246] 0
[52580.945247] hci0 orig refcnt 8
[52580.945249] hci0: sock 000000004458fd39
[52580.955544] hci0 evt 2
[52580.955547] hci0 orig refcnt 9
[52580.955590] hdev hci0
[52580.955617] channel 3 len 13
[52580.955624] sock 000000004458fd39
[52580.955631] channel 2 len 22
[52580.955633] hci0 orig refcnt 8
[52580.955658] hci0 start
[52580.955661] length 0
[52580.955712] sock 000000007bf75a72, sk 000000004458fd39
[52580.955776] sock 0000000000be67f0 sk 00000000702bbe49
[52580.955779] channel 2 len 109
</details>

apiraino, 2021-02-11:
here's some more dmesg log after a reboot

<details><summary>logs</summary>

```

# dmesg | grep Bluetooth
[    4.520321] Bluetooth: Core ver 2.22
[    4.520349] Bluetooth: HCI device and connection manager initialized
[    4.520352] Bluetooth: HCI socket layer initialized
[    4.520354] Bluetooth: L2CAP socket layer initialized
[    4.520357] Bluetooth: SCO socket layer initialized
[    4.536458] Bluetooth: hci0: Firmware revision 0.1 build 50 week 12 2019
[    4.648226] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.648229] Bluetooth: BNEP filters: protocol multicast
[    4.648235] Bluetooth: BNEP socket layer initialized
[    5.932316] Bluetooth: RFCOMM TTY layer initialized
[    5.932320] Bluetooth: RFCOMM socket layer initialized
[    5.932323] Bluetooth: RFCOMM ver 1.11
[27312.002973] Bluetooth: hci0: Bootloader revision 0.0 build 26 week 38 2015
[27312.004397] Bluetooth: hci0: Device revision is 16
[27312.004402] Bluetooth: hci0: Secure boot is enabled
[27312.004405] Bluetooth: hci0: OTP lock is enabled
[27312.004408] Bluetooth: hci0: API lock is enabled
[27312.004410] Bluetooth: hci0: Debug lock is disabled
[27312.004414] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[27312.015751] Bluetooth: hci0: Found device firmware: intel/ibt-12-16.sfi
[27313.724055] Bluetooth: hci0: Waiting for firmware download to complete
[27313.724891] Bluetooth: hci0: Firmware loaded in 1682660 usecs
[27313.725039] Bluetooth: hci0: Waiting for device to boot
[27313.736876] Bluetooth: hci0: Device booted in 11693 usecs
[27313.737457] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-12-16.ddc
[27313.739996] Bluetooth: hci0: Applying Intel DDC parameters completed
[27313.741082] Bluetooth: hci0: Reading supported features failed (-16)          <------ ???
[27313.741093] Bluetooth: hci0: Telemetry exception format not supported
[27313.741968] Bluetooth: hci0: Firmware revision 0.1 build 50 week 12 2019
[37764.262094] Bluetooth: hci0: urb 00000000dd79c8d7 failed to resubmit (2)
[37764.547049] Bluetooth: hci0: Firmware revision 0.1 build 50 week 12 2019
[37770.943426] Bluetooth: hci0: urb 000000006c6f68bb failed to resubmit (2) <------ ???
[37771.224107] Bluetooth: hci0: Firmware revision 0.1 build 50 week 12 2019
[38000.317525] Bluetooth: hci0: urb 00000000986a44f6 failed to resubmit (2)
[38000.603219] Bluetooth: hci0: Firmware revision 0.1 build 50 week 12 2019
[38083.990097] Bluetooth: hci0: urb 00000000537d21bc failed to resubmit (2)
[38084.271315] Bluetooth: hci0: Firmware revision 0.1 build 50 week 12 2019
```

</details>

(I'll investigate a bit those `???` messages)

apiraino, 2021-02-11:
More logs showing a failed attempt to  connect

<details><summary>logs</summary>

```
[39886.745250] sock 00000000d26ce41a sk 0000000041b53739
[39886.745264] channel 2 len 110
[39886.745312] sock 00000000d26ce41a sk 0000000041b53739
[39886.745319] channel 2 len 71
[39886.745382] sock 00000000d26ce41a sk 0000000041b53739
[39886.745388] channel 2 len 112
[39886.745420] sock 00000000891f2933
[39886.745430] chan 000000000edaa45d
[39886.745434] chan 000000000edaa45d orig refcnt 1
[39886.745438] sk 00000000cbdb113d
[39886.745460] sk 00000000cbdb113d
[39886.745474] sk 00000000cbdb113d
[39886.745482] 04:ea:56:ae:95:a1 -> dc:2c:26:ec:d5:ef (type 0) psm 0x0001 mode 0x00
[39886.745486] 04:ea:56:ae:95:a1 -> dc:2c:26:ec:d5:ef
[39886.745490] hci0 orig refcnt 7
[39886.745495] hci0 dst dc:2c:26:ec:d5:ef
[39886.745499] hci0 orig refcnt 8
[39886.745504] conn 00000000077d15cf
[39886.745510] hcon 00000000077d15cf orig refcnt 0
[39886.745514] hcon 00000000077d15cf
[39886.745563] cache 000000001a2a713b, dc:2c:26:ec:d5:ef
[39886.745569] hci0 opcode 0x0405 plen 13
[39886.745575] skb len 16
[39886.745592] hci0 hcon 00000000077d15cf
[39886.745599] hcon 00000000077d15cf conn 000000005c95edda hchan 00000000b6fb6100
[39886.745606] conn 000000005c95edda, psm 0x01, dcid 0x0000
[39886.745616] chan 000000000edaa45d orig refcnt 2
[39886.745625] hci0 cmd_cnt 1 cmd queued 1
[39886.745664] hci0 type 1 len 16
[39886.745668] hcon 00000000077d15cf orig refcnt 1
[39886.745675] hcon 00000000077d15cf orig refcnt 2
[39886.745690] chan 000000000edaa45d BT_BOUND -> BT_CONNECT
[39886.745701] chan 000000000edaa45d state BT_CONNECT timeout 10000
[39886.745712] chan 000000000edaa45d orig refcnt 3
[39886.745725] hci0 orig refcnt 9
[39886.745823] sk 00000000cbdb113d
[39886.745951] sock 00000000a75db6ef sk 00000000d456c48e
[39886.745957] got 14 bytes
[39886.745965] channel 2 len 20
[39886.745971] 0
[39886.745974] hci0 orig refcnt 8
[39886.745979] hci0: sock 00000000d456c48e
[39886.746020] channel 3 len 14
[39886.746025] hci0 start
[39886.746042] channel 3 len 21
[39886.746097] channel 2 len 27
[39886.746111] sock 00000000d456c48e
[39886.746159] channel 2 len 22
[39886.746163] hci0 orig refcnt 9
[39886.746168] length 0
[39886.746273] sock 00000000a75db6ef, sk 00000000d456c48e
[39886.746365] sock 00000000a75db6ef, sk 00000000d456c48e
[39886.746426] sock 00000000d26ce41a sk 0000000041b53739
[39886.746433] channel 2 len 102
[39886.748366] sock 00000000d26ce41a sk 0000000041b53739
[39886.748428] channel 2 len 60
[39886.748473] sock 00000000d26ce41a sk 0000000041b53739
[39886.748480] channel 2 len 98
[39886.748584] sock 00000000a75db6ef sk 00000000d456c48e
[39886.748668] got 7 bytes
[39886.748700] channel 2 len 13
[39886.748704] 0
[39886.748708] hci0 orig refcnt 8
[39886.748742] hci0: sock 00000000d456c48e
[39886.748749] hci0 state 0 -> 1
[... 137 lines cut ...]
[39892.047885] chan 000000000edaa45d
[39892.051180] sock 00000000d26ce41a sk 0000000041b53739
[39892.051191] channel 2 len 104
[39892.051328] sock 00000000d26ce41a sk 0000000041b53739
[39892.051415] channel 2 len 61
[39892.051703] sock 00000000a75db6ef sk 00000000d456c48e
[39892.051786] got 13 bytes
[39892.051827] channel 2 len 19
[39892.051833] 0
[39892.051837] hci0 orig refcnt 8
[39892.051842] hci0: sock 00000000d456c48e
[39892.061654] hci0 hcon 00000000077d15cf handle 0
[39892.061664] hcon 00000000077d15cf
[39892.089547] hci0 orig refcnt 9
[39892.089604] hdev hci0
[39892.089662] channel 3 len 13
[39892.089672] sock 00000000d456c48e
[39892.089680] channel 2 len 22
[39892.089684] hci0 orig refcnt 8
[39892.089716] hci0 start
[39892.089722] length 0
[39892.089796] sock 00000000a75db6ef, sk 00000000d456c48e
[39892.089890] sock 00000000d26ce41a sk 0000000041b53739
```
<summary>

cschramm, 2021-02-12:
Looks like `btintel` (and or `btusb`) would have been the better choice for debug messages. :laughing:

Anyway, something is wrong at
>  [52574.503020] hcon 000000003c71cafd reason 19

It hints at the remote device, but that does not need to be true.

What happens when the `failed to resubmit (2)` message shows up and the firmware gets reset? Is that when you try to connect? Debugging in `btusb` should give some more insights there.

apiraino, 2021-02-12:
ok after another debug session I finally managed to connect && pair. I have isolated the issue on a case of PEBKAC, user was not actually enabling pairing mode on device, and `bluetoothd` was basically saying "oh I see the keyboard, I'll try to conn... oh it's gone".

my apologies for wasting your time, thanks again for the useful info, though.

closed 2021-02-12 by apiraino