blueman-project/blueman #1490: Missing German translation for "Rename Device" in openSUSE

tapwag, 2021-01-31:
blueman: 2.1.4
BlueZ: Unknown
Distribution: openSUSE Tumbleweed
Desktop environment: LxQt

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

tapwag, 2021-01-31:
Dear blueman-developers,

some strings do not seem to be integrated into openSUSE "Tumbleweed" which is a rolling release distribution. 

I am referring to the following bug: https://bugzilla.opensuse.org/show_bug.cgi?id=1102143

Things that I have done so far:

Checked https://github.com/blueman-project/blueman/blob/master/po/de.po - the string is translated there.
Visited: https://hosted.weblate.org/translate/blueman/stable/de/ - string also translated there.

I should have probably posted this issue here first than with openSUSE but could you sync the translation in the coming version so that the bug in openSUSE can be closed?

cschramm, 2021-02-03:
Are you sure about your blueman version and the issue? It was a known issue that's fixed in 2.1.4 but the bug report is more than two years older than that version.

closed 2021-02-18 by github-actions[bot]