blueman-project/blueman #1495: GSettings option to hide the status/indicator icon

tari01, 2021-02-13:
blueman: 2.1.4
BlueZ: 5.55
Distribution: Manjaro
Desktop environment: MATE

I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Currently we have no mechanism that would hide the blueman-applet icon (in either SNI or legacy mode), while keeping the daemon running. It would be really useful if the applet could expose a GSettings option that would allow the user to hide/show the applet.

cschramm, 2021-02-14:
As you found in the other thread, the blueman-applet process is vital. What is not and actually displays the icon is the blueman-tray process. It gets auto-started by the StatusIcon plugin.

We thought about making the StatusIcon plugin optional in #949. That is some effort, though, as some other parts rely on its presence.

What's the motivation behind this?

tari01, 2021-02-14:
Thank you for your reply.

> As you found in the other thread, the blueman-applet process is vital.

Yes. I see no reason why anyone would want to disable blueman-applet. It is a very stable all-in-one backend.

> We thought about making the StatusIcon plugin optional in #949

:+1: 

> What's the motivation behind this?

Applications that provide an alternate indication mechanism would need the daemon running, but would also like to have a means of disabling the icon to avoid confusion/double indicators.

cschramm, 2021-02-15:
Looks easier than I thought. #1496.

> Applications that provide an alternate indication mechanism

To be honest, I'm not sure if I like that as the icon provides functionality and is at least needed to manage applet plugins. We'd have to look into providing that in blueman-manager or something to avoid losing it.

ton, 2021-09-09:
What is the status of this, is this supposed to work now?

I would like to run `blueman-applet` in the background only (without a systray icon), and use `blueman-manager` for interacting with the tool. 

I've tried several options to disable the systray icon, without success:

* Disabling the `AppIndicator` plugin. It removes the systray icon temporarily but it immediately shows up again. After restarting `blueman-applet`, the `AppIndicator` plugin is listed as disabled, but the systray icon is still there.
* Removing `!AppIndicator` or `AppIndicator` from the dconf key.

I'm using the latest version of blueman.

cschramm, 2021-09-10:
@ton: You can test #1496.

closed 2022-10-07 by cschramm