blueman-project/blueman #1497: No Passkey Dialogue Box While Pairing Keyboard

sachmo-source, 2021-02-17:
blueman: 2.1.2
BlueZ: 5.53-0ubuntu3
Distribution: Linux Mint 20.1
Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The pairing dialogue box does not appear when I attempt to pair either of my Logitech K810 Keyboards. If I open blueberry bluetooth manager while I am pairing the keyboards via blueman, blueberry produces the appropriate pairing dialogue box. If I type in the passkey request while focused on the blueman window, the device pairs properly (and blueberry freezes). From
what I could understand of the last part of the loglevel debug readout (when I was attempting to pair the K810 keyboard model
with blueman alone), the request simply timed out due to lack of passkey input....

'AddressType': 'public', 'Name': 'Logitech K810', 'Alias': 'Logitech K810', 'Class': 9536, 'Icon': 'input-keyboard', 'Paired': False, 'Trusted': True, 'Blocked': False, 'LegacyPairing': True, 'Connected': True, 'UUIDs': [], 'Adapter': '/org/bluez/hci3', 'ServicesResolved': False}
blueman-manager 10.10.44 ERROR    Manager:201 error_handler:  Authentication Timeout
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 81, in callback
    value = proxy.call_finish(result).unpack()
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.AuthenticationTimeout: Authentication Timeout (36)
blueman-applet 10.10.46 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci3/dev_34_88_5D_3D_C4_0A {'Connected': False}

It would seem the pairing dialogue box simply does not appear in blueman for this keyboard model, for this disto/environment. I do not have other bluetooth keyboards to test with. This is the only issue I have with blueman, whatsoever. The full debug readout is below. That you for your time in reading this.

LOGLEVEL DEBUG FULL OUTPUT...

victor@computer:~$ blueman-applet --loglevel debug
blueman-applet version 2.1.2 starting
Stale PID, overwriting
blueman-applet 10.09.27 INFO     PluginManager:85 load_plugin: ['AppIndicator', 'StatusIcon', 'AuthAgent', 'Menu', 'StandardItems', 'Networking', 'SerialManager', 'ExitItem', 'RecentConns', 'KillSwitch', 'NMDUNSupport', 'DhcpClient', 'ShowConnected', 'NetUsage', 'DiscvManager', 'DBusService', 'GameControllerWakelock', 'NMPANSupport', 'TransferService', 'PPPSupport', 'PowerManager']
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 10.09.27 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci1 {'Address': '00:1A:7D:DA:71:12', 'AddressType': 'public', 'Name': 'computer #2', 'Alias': 'audio node', 'Class': 1835268, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001112-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0535'}
blueman-applet 10.09.27 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci3 {'Address': '00:1A:7D:DA:71:0D', 'AddressType': 'public', 'Name': 'computer', 'Alias': 'input node', 'Class': 1835268, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001112-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0535'}
blueman-applet 10.09.27 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Address': '00:1A:7D:DA:71:11', 'AddressType': 'public', 'Name': 'computer #1', 'Alias': 'gamepads [3 & 4]', 'Class': 1835268, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001112-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0535'}
blueman-applet 10.09.27 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci2 {'Address': '00:1A:7D:DA:71:05', 'AddressType': 'public', 'Name': 'computer #3', 'Alias': 'gamepads [1 & 2]', 'Class': 1835268, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001112-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0535'}
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 10.09.27 INFO     Networking:31 load_nap_settings: Loading NAP settings
blueman-tray version 2.1.2 starting
blueman-tray version 2.1.2 starting
Stale PID, overwriting
Stale PID, overwriting
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 10.09.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
[... cut ...]
blueman-manager 10.10.44 ERROR    Manager:201 error_handler:  Authentication Timeout
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 81, in callback
    value = proxy.call_finish(result).unpack()
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.AuthenticationTimeout: Authentication Timeout (36)
blueman-applet 10.10.46 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci3/dev_34_88_5D_3D_C4_0A {'Connected': False}

->

matthewfl, 2021-02-19:
I just encountered this same issue on archlinux.  I was able to get the pairing code by opening bluetoothctl at the same time as blueman.

blueman version: 2.1.4-1
bluez version: 5.55-2
Distribution: archlinux 5.10.15-arch1-1
Desktop environment: awesomewm

cschramm, 2021-02-23:
From the output it looks like blueman-applet does not receive any request. Anything in the bluetooth.service logs?