blueman-project/blueman #1500: Blueman applet can't be triggered from command line

fctorial, 2021-02-24:
It isn't possible to trigger blueman devices window to come up without using a mouse. So you can't connect to a bluetooth mouse without a usb mouse.

cschramm, 2021-02-24:
Regarding the tray icon: I'd say that depends on the desktop environment. If it does not allow to put focus on it, you cannot open the menu without a click. That is correct but not specific to blueman, is it?

Regarding opening the manager: Just run `blueman-manager` directly. You don't need the tray icon for it.

Regarding using the GUI without a pointer device: If it is about regularly connecting to one specific mouse, you're better off with a respective `bluetoothctl connect` command. There is no point in using the blueman GUI for that task.

fctorial, 2021-02-24:
blueman-manager is what I needed.

closed 2021-02-24 by fctorial