blueman-project/blueman #1508: Blueman freezes whole input system on connection using blueman-applet

Meister1593, 2021-03-14:
blueman: 2.1.4
BlueZ: 5.56
libinput: 1.17.0
xorg-server: 1.20.10
Distribution: Arch Linux
Desktop environment: AwesomeWM 4.3

When i connect to my headphones using blueman-applet (recent connections) and immediately try to open same tray menu and wait a bit, whole input for X freezes for about 30 seconds. 
Neither keyboard or mouse can do anything at this point (but i can see graphics is working fine).

This happened for me multiple times and it was quite frustrating.

Meister1593, 2021-03-14:
This issue does not happen in blueman-manager when i try to connect and open menu (by right clicking on device)

cschramm, 2021-05-14:
Sorry, this slipped through. Those things changed quite a bit in master. Could you try with recent code? AUR has blueman-git for that.

Meister1593, 2021-05-14:
Sure , I will check today, though I'm on gentoo now.

Meister1593, 2021-05-14:
Installed latest git version, same error still presented
Upon disconnect and connect it switches input completely off and i can't do anything in desktop overall from 15 seconds to 1 minute, even though i can see everything working fine.
This is what i noticed when this input freeze happened (first time this timeout was after some time with frozen input and second was right after i got input back again)
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/main/indicators/GtkStatusIcon.py", line 45, in <lambda>
    gtk_item.connect('activate', cast(Callable[[Gtk.MenuItem], None], lambda _, idx=index: activate(idx)))
  File "/usr/lib/python3.8/site-packages/blueman/main/Tray.py", line 45, in _activate_menu_item
    AppletService().ActivateMenuItem('(ai)', indexes)
  File "/usr/lib/python3.8/site-packages/gi/overrides/Gio.py", line 350, in __call__
    result = self.dbus_proxy.call_sync(self.method_name, arg_variant,
gi.repository.GLib.Error: g-io-error-quark: Timeout was reached (24)
```

cschramm, 2021-05-14:
I cannot replicate. Just to be sure:

1. Use recent connections menu to start connecting.
2. Immediately right-click the tray icon to open the menu again.
3. See everything getting unresponsive.

I suppose the exception is from blueman-tray. It looks like it fails to talk to blueman-applet and it makes sense that blueman-tray blocks in that case. We could solve that with an asynchronous call but blueman-applet should not block while connecting, so there seems to be more.

Please provide the output from `blueman-applet --loglevel debug` of when you're trying the steps. Maybe we can isolate where it blocks.

Meister1593, 2021-05-14:
> I cannot replicate. Just to be sure:
>
>     1. Use recent connections menu to start connecting.
>
>     2. Immediately right-click the tray icon to open the menu again.
>
>     3. See everything getting [...]

Full steps to reproduce:
For connection:
1. Powerup headset
2. Connect headphones through systray blueman menu 
2.1 You can connect headphones through autoconnection when headphones starts, doesn't matter. 
3. Open blueman systray menu again and wait for connection, after that system input will freeze. Note, you can open tray menu before systray icon changes to green or after it changed to green (but not too long). It will trigger this timeout anyway.

For disconnection:
1. Open blueman's systray menu, use *Disconnect device* or in main blueman applet program disconnect and open blueman's systray menu.
2. Wait for green icon to dissapear and system input will freeze

[Logs](https://gist.github.com/Meister1593/33d89d6d365cedc95de01bd678431ed3)
(Note in logs timing for certain actions after lockup timeout, it was like instant output for multiple messages at once. Also, disconnect didn't produce timeout error now)

Meister1593, 2021-05-16:
@cschramm Any progress? Sorry for pinging, I just thought this issue would stale again if I would not ping.

cschramm, 2021-05-17:
Thanks for your patience. :sweat_smile:

No hint in the logs. I have no idea where this could block. Let's try a thread dump: Run `PYTHONFAULTHANDLER=1 blueman-applet`, make it lock, switch to another terminal and run `killall -SIGABRT blueman-applet`. (In case it does not unblock, `killall blueman-tray` should help to get the desktop responsive again).

Meister1593, 2021-05-17:
> Thanks for your patience. sweat_smile
>
> No hint in the logs. I have no idea where this could block. Let's try a thread dump: Run `PYTHONFAULTHANDLER=1 blueman-applet`, make it lock, switch to another [...]

[dump_connect.txt](https://github.com/blueman-project/blueman/files/6496720/dump_connect.txt)
[dump_disconnect.txt](https://github.com/blueman-project/blueman/files/6496721/dump_disconnect.txt)

cschramm, 2021-05-17:
So when the device connects or disconnects, blueman tries to send a notification to the notifications service and seems to block on initialization of the proxy object for that.

Disabling the "ConnectionNotifier" plugin should probably avoid the problem.

I don't know what could block there. We could try a C backtrace: Get the blueman-applet PID, make it block, switch to another terminal and run `gdb -p [PID] -ex bt` with root privileges.

I could imagine some deadlock there where the window manager waits for blueman-tray, blueman-tray waits for blueman-applet, and blueman-applet waits for the notifications service. If the window manager and the notifications service live in a single process in a way where the waiting window manager blocks the notifications service, the loop is complete. That would explain why it's AwesomeWM specific.

Meister1593, 2021-05-17:
> So when the device connects or disconnects, blueman tries to send a notification to the notifications service and seems to block on initialization of the proxy object for that.
>
> Disabling the [...]

Disabling notifications helped only for disconnect, but for connection it's still a freeze (even if blueman won't have notifications).
Though i have another notification when connection happens (it's for connected headset, new sink), so i checked what if i would suspend notifications. It still making freezes even when no notifications are shown. I couldn't really disable it completely, so it well may be.
Coldn't find anything awesomewm-specific in c-backtrace, so still not entirely sure if it's awesomewm specific...
[gdb log](https://gist.github.com/Meister1593/e1c71c6c4fcdadac596fbb585a7fafba)

cschramm, 2021-05-18:
> Coldn't find anything awesomewm-specific in c-backtrace

Sure. The only thing I can think of at the point where it blocks is that blueman-applet / GLib waits for something on the D-Bus regarding notifications. If the notifications interface is handled by the same component that is waiting for UI stuff from blueman-tray, we could have a deadlock (three components waiting for the others in a circle, two via D-Bus, one via X). As far as a I know awesomewm has everything like the actual window manager and a notifications service in a single binary, so it could theoretically be possible that waiting for the UI stuff blocks it from handling things on D-Bus.

As expected, the process hangs in libgio, doing some polling via libglib. Unfortunately, to make sense of the backtrace, we'd need debug symbols for the glib2 package and as far as I get Arch's "concept", you'd have to rebuild the package for that: https://wiki.archlinux.org/title/Debug_-_Getting_Traces Could you go through that? :grimacing:

If it also locks up on connection without any notification from blueman, there might be more, of course. You could provide another Python thread dump of that to see where blueman-applet hangs, if a notification can't be it.

Meister1593, 2021-05-18:
> Sure. The only thing I can think of at the point where it blocks is that blueman-applet / GLib waits for something on the D-Bus regarding notifications. If the notifications interface is handled by [...]

As i said before, things changed and i'm now on Gentoo :smile:. 
Adding debug symbols to glib was quite easy and i got a lot more info with it. For example, there is a poll with timeout=25 seconds and it's indeed related somehow to d-bus. It leads to somewhere but all i can see here is address where it leads, and... i'm not so familiar with gdb debugging, but maybe starting to debug awesome is a good idea?
[dump](https://gist.github.com/Meister1593/2ef5be566830e6fc6c32aba872f86f97)

Meister1593, 2021-05-18:
Awesome is polling glib too... so it's deadlock?
[dump](https://gist.github.com/Meister1593/c985ff965bdab7f6c1bd543a61bb22d1)
It polls exactly [here](https://github.com/awesomeWM/awesome/blob/5da5d36178250335f748eed43551daad9e5af694/awesome.c#L443)

cschramm, 2021-05-18:
The awesomewm thread looks fine. That's just its main loop and it's polling for something to happen. That means, it is *not* actively waiting for blumean-tray. Probably X is.

blueman-applet is inside the GDBusProxy's init method, called from the Python code and waiting for something happening in its main loop as well, specifically something that triggers the callback passed to `async_initable_init_second_async` (https://github.com/GNOME/glib/blob/2.66.7/gio/gdbusproxy.c). That seems to be in vain. What should be called there is `async_init_call_start_service_by_name`. That effectively sends a call that can be triggered as
```
dbus-send --session --dest=org.freedesktop.DBus --print-reply /org/freedesktop/DBus org.freedesktop.DBus.StartServiceByName string:"org.freedesktop.Notifications" uint32:0
```
as well. I suppose it does not get a response which typically means that the D-Bus service fails to activate the notifications service.

However, you do get notifications in general, right? So that service obviously works and I guess if you try that call it will just work. Not sure if you could try it from a terminal while the desktop is unresponsive.

Am I right that it's awesomewm that provides the notifications service? You should probably find some reference by greping for `org.freedesktop.Notifications` in `/usr/share/dbus-1/services`.

cschramm, 2021-05-18:
One thing you could also try to make things visible is `dbus-monitor`.

Meister1593, 2021-05-18:
Here's what i got when i connected via blueman systray from `dbus-monitor` : [gist](https://gist.github.com/Meister1593/95d41194f5106e8587e9116650ba34cc)

cschramm, 2021-05-18:
Alright, so in that session blueman-tray is `:1.17`, blueman-applet is `:1.13` The call with serial 207 at 1621354748 blocks blueman-tray as it waits for a response until 1621354773, i.e. for 25 seconds. There are other signals and method calls at 1621354750 that indicate that the device is already connected but blueman-applet only shows up at 1621354773 where it talks to the notifications service at `:1.127`. That matches the thread dumps so far. It does not query the service by it's well-known name is it already knows its unique name `:1.127` at that point.

What's interesting is the communication between blueman-applet and the notifications service *before* the call from blueman-tray:

```
method call time=1621354748.223106 sender=:1.13 -> destination=org.freedesktop.DBus serial=2078 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=StartServiceByName
   string "org.freedesktop.Notifications"
   uint32 0
error time=1621354748.223122 sender=org.freedesktop.DBus -> destination=:1.13 error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=2078
   string "The name org.freedesktop.Notifications was not provided by any .service files"
method call time=1621354748.223463 sender=:1.13 -> destination=org.freedesktop.DBus serial=2079 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=GetNameOwner
   string "org.freedesktop.Notifications"
method return time=1621354748.223476 sender=org.freedesktop.DBus -> destination=:1.13 serial=60 reply_serial=2079
   string ":1.127"
method call time=1621354748.223766 sender=:1.13 -> destination=:1.127 serial=2080 path=/org/freedesktop/Notifications; interface=org.freedesktop.DBus.Properties; member=GetAll
   string "org.freedesktop.Notifications"
```

There is the expected `StartServiceByName` with the same error response that you got, then `GetNameOwner` that finds `:1.127` and then a call to get all properties.
```
dbus-send --session --dest=org.freedesktop.Notifications --print-reply /org/freedesktop/Notifications org.freedesktop.DBus.Properties.GetAll string:"org.freedesktop.Notifications"
```
From your d-feet screenshot, the notifications service does not implement `org.freedesktop.DBus.Properties`. That should usually be fine as GDbusProxy will just ignore the error but there is no error at all. What I would expect is something like `error time=... sender=:1.127 -> destination=:1.13 error_name=org.freedesktop.DBus.Error.UnknownMethod reply_serial=2080` but no. It looks to me like awesomewm fails to send that error and thus everything is blocked in blueman-applet until GDbusProxy just gives up on the call.

closed 2021-07-26 by github-actions[bot]

wereii, 2021-07-30:
Hey, I can confirm that I am having a pretty much same error.
Awesome WM (v4.3) too, the moment I click the tray when there is a connection or disconnection in progress my desktop freezes.

Reading thru, from the last comment of @cschramm it seems it's an issue with Awesome?

conte91, 2021-08-30:
+1 on this issue. I'm also affected and I'm happy to help with debugging if needed :)

cschramm, 2021-08-31:
@wereii @conte91: From my last post, it seems like Awesome WM does not give any response to
```
dbus-send --session --dest=org.freedesktop.Notifications --print-reply /org/freedesktop/Notifications org.freedesktop.DBus.Properties.GetAll string:"org.freedesktop.Notifications"
```
Can you confirm that?

wereii, 2021-08-31:
```sh
[~]> awesome -v
awesome v4.3 (Too long)
 • Compiled against Lua 5.3.5 (running with Lua 5.3)
 • D-Bus support: ✔
 • execinfo support: ✔
 • xcb-randr version: 1.6
 • LGI version: 0.9.2
[~]> dbus-send --session --dest=org.freedesktop.Notifications --print-reply /org/freedesktop/Notifications org.freedesktop.DBus.Properties.GetAll string:"org.freedesktop.Notifications"
Error org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```
Seems like it. I guess this it's time to introduce this over to the guys at awesomewm

conte91, 2021-09-01:
Thanks for opening the issue @wereii! I can confirm I see the same exact output on my machine [Arch + awesome 4.3-2].

wereii, 2021-09-06:
@conte91, FYI this problem seems to be resolved with latest awesome built against current master (`13cd20780`) - you can test it with `awesome-git` package from AUR or you can wait for next A.wm version:)

srodrigo, 2023-01-22:
Hey, I'm on Awesome 4.3, blueman-applet and I have the exact same issue with the X freezing for a few seconds when messing around with the tray applet. System is Ubuntu 22.10.

```
srodrigo@srodrigo ~ $ awesome --version
awesome v4.3 (Too long)
 • Compiled against Lua 5.3.6 (running with Lua 5.3)
 • D-Bus support: ✔
 • execinfo support: ✔
 • xcb-randr version: 1.6
 • LGI version: 0.9.2
```

cschramm, 2023-01-22:
5build1 is the Debian revision of that package. The upstream version is just 4.3 and there is no newer release that could contain the commit mentioned by @wereii.

srodrigo, 2023-01-22:
You are right, that version is so old actually, I didn't notice.

I followed the steps to build from sources https://github.com/awesomeWM/awesome#debian-based and now I can't reproduce the issue.

Thank you so much :)

cschramm, 2023-03-17:
@joelsleeba: As far as I know i3 does not even have a notification service built in, so it cannot be broken.

dm17, 2023-06-04:
This happens more or less sometimes. I'm using awesome as well, but I didn't think it could be awesome-specific. I figured it was a low level driver problem, but I don't see why a modern AMD system would have to hang everything when dealing with bluetooth.

```
awesome v4.3 (Too long)
 • Compiled against Lua 5.3.6 (running with Lua 5.3)
 • D-Bus support: ✔
 • execinfo support: ✘
 • xcb-randr version: 1.6
 • LGI version: 0.9.2
 ```

cschramm, 2023-06-04:
What's your point, @dm17? The problem is awesome 4.3's completely broken naughty.dbus, fixed in https://github.com/awesomeWM/awesome/pull/2722. blueman correctly waits for a reply to its notification which awesome just does not send and if I remember correctly that blocks the tray icon rendering which in terms blocks the complete window manager.

It's not your modern AMD system and Bluetooth, it's just your window manager that has not had any releases for 4 ½ years.

rodfersou, 2024-10-08:
I got the same issue with i3wm, and it starts after I add the `dunst` notification system.

Running Fedora 40 here:

```
$ dnf info dunst
RPM Fusion for Fedora 40 - Nonfree - NVIDIA Driver                                                                 3.3 kB/s | 4.9 kB     00:01    
RPM Fusion for Fedora 40 - Nonfree - Steam                                                                         1.3 kB/s | 1.5 kB     00:01    
RPM Fusion for Fedora 40 - Nonfree tainted                                                                         1.4 kB/s | 2.0 kB     00:01    
RPM Fusion for Fedora 40 - Nonfree - Updates                                                                        18 kB/s |  26 kB     00:01    
Installed Packages
Name         : dunst
Version      : 1.11.0
Release      : 1.fc40
Architecture : x86_64
Size         : 315 k
Source       : dunst-1.11.0-1.fc40.src.rpm
Repository   : @System
From repo    : updates
Summary      : Lightweight and customizable notification-daemon
URL          : https://dunst-project.org
License      : BSD-3-Clause
Description  : Dunst is a lightweight replacement for the notification daemons provided by
             : most desktop environments. It’s very customizable, isn’t dependent on any
             : toolkits, and therefore fits into those window manager centric setups we all
             : love to customize to perfection.
```

rodfersou, 2024-10-08:
I'm still puzzled.. first thought was because I start `dunst` wrongly, so I kill it, but now I turn on my `headphones` and it just connect, while my `keyboard` and `mouse` don't

I'm wondering if `Google Chrome` is the culprit, had my gmail tab frozen.. let me restart it

rodfersou, 2024-10-08:
yes, after kill `Google Chrome` my `mouse` and `keyboard` connect