blueman-project/blueman #1510: Trying to tether to an iPhone. Connects but fails to obtain IP address

JulianBane, 2021-03-16:
blueman: 2.1.4 (openSUSE & Manjaro)
BlueZ: 5.56-1  (openSUSE &Manjaro)
Distribution: openSUSE Tumbleweed & Manjaro (5.11.2-1)
Desktop environment: LXDE

I have been trying to get openSUSE to use my iPhone as it's internet gateway via bluetooth tethering. Device pairing and trusting was fine. Connecting to the iPhone ****** makes a connection, creates a network device bnep0 but gives an error message saying it was unable to get an IP address for the device.

I've tried manually configuring with dhclient but nothing.

I tried manually assigning an IP and Default route but no internet connectivity.

One clue maybe that when I start blueman-applet I get the following:
`blueman-applet version 2.1.4 starting
Stale PID, overwriting
blueman-applet 14.29.02 WARNING  PluginManager:148 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 14.29.03 WARNING  PluginManager:148 __load_plugin: Not loading DhcpClient because its conflict has higher priority
`
I then moved to a Manjaro system and got identical behaviour.

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

JulianBane, 2021-03-17:
OK, think I may have an issue with the iPhone. Tethering by wifi, which used to work, now fails with an IP of 196.254.xx.xx/16  being assigned (i.e. an automatic IP address and assigned when DHCP fails). As such, I think this is not blueman issue but will update this thread if I get more information.

infirit, 2021-03-17:
We have 2 plugins for pan, one uses NetworkManager in the other we set things up from blueman. The message you see is saying it is handles through NM. You can check this by starting bleuman-services

The full debug output of blueman-applet with the connection attempt should shed some light what is going on.

kasi042, 2021-03-22:
Hi there, 
Sorry, it took me some time. The issue is reproducible. It's not the phone. My setup:
openSUSE Leap 15.2
bluez: 5.48
blueman: 2.1.4
Desktop Environment: KDE
iPhone
It works brilliantly on the laptop with network manager. On the desktop I have wicked running like Julian, so we have to use the blueman plugins. Here's some journalctl:
[blueman_issue.txt](https://github.com/blueman-project/blueman/files/6185382/blueman_issue.txt)

cschramm, 2021-03-26:
The output that you posted shows that blueman fails to get the network interface for the BlueZ device. It says there would be no `Interface` D-Bus property (it seems like the object does have the `org.bluez.Network1` interface, though).

If you can't follow that path on your own, please actually provide the full debug output of blueman-applet (see the troubleshooting page you mentioned) so that we can see what's going on.

kasi042, 2021-03-28:
Whoops, sorry. I thought I'd get a notification. Please find the debug output here:
[blueman-debug.txt](https://github.com/blueman-project/blueman/files/6218017/blueman-debug.txt)
It was: starting blueman, connection the phone, trying to access the NAP. That's when I aborted. The phone, BTW, does show that the connection to the NAP has been established.

cschramm, 2021-05-06:
So blueman-applet uses it's `DhcpClient` in your case and blueman-mechanism reports a failed DHCP client invocation. Does `journalctl -u blueman-mechanism.service` show anything related?

kasi042, 2021-05-08:
Well, it seems  related:

`mars:~ # journalctl -u blueman-mechanism.service
-- Logs begin at Sat 2021-05-08 22:46:41 CEST, end at Sat 2021-05-08 23:07:00 CEST. --
May 08 23:06:36 mars systemd[1]: Starting Bluetooth management mechanism...
May 08 23:06:36 mars blueman-mechanism[5312]: Unable to init server: Could not connect: Connection refused
May 08 23:06:36 mars blueman-mechanism[5312]: Unable to init server: Verbindung ist gescheitert: Connection refused
May 08 23:06:36 mars blueman-mechani[5312]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
May 08 23:06:36 mars systemd[1]: Started Bluetooth management mechanism.`

cschramm, 2021-05-09:
Not really related. Nothing after the start when you connect the service?

What DHCP client do you have (`dhclient`, `dhcpcd`, `udhcpc`)?

kasi042, 2021-05-09:
Sorry, can't tell. That's what I got after starting blueman and triying to tether the phone. 
I have to say, on all desktops / "wired" connections I am using static IP adresses. My router is running a DHCP server for other devices. I have switched to DHCP just to check - with the above result. Switched back straight away. 
I just started the desktop and sshed in. That's what I get:
```
~> zypper se -i dhcp
Repository-Daten werden geladen...
Installierte Pakete werden gelesen...

S | Name        | Summary                                | Type
--+-------------+----------------------------------------+------
i | dhcp        | Common Files Used by ISC DHCP Software | Paket
i | dhcp-client | ISC DHCP Client                        | Paket

```

cschramm, 2021-05-10:
I guess `dhcpcd` should be in use then. The command, blueman tries should be `dhcpcd -m 100 bnep0` then. What happens when you run it manually after connecting?

kasi042, 2021-05-12:
Sorry, it took a while. I always need to switch that machine back to DHCP which gives it a different IP address. I'm actually not needing this. I'm just trying to help. ;-)
There is no `dchpcd`. There isn't even any package providing this for Leap 15.2. Here it's `wickedd-dhcp`:
mars:~ # journalctl | grep dhcp
```
May 12 22:08:04 mars wickedd-dhcp4[1351]: eth0: Request to acquire DHCPv4 lease with UUID 72349c60-4fb2-0a00-5b05-000009000000
May 12 22:08:05 mars wickedd-dhcp4[1351]: eth0: Committed DHCPv4 lease with address x.x.x.x (lease time 864000 sec, renew in 432000 sec, rebind in 756000 sec)

```

cschramm, 2021-05-12:
Sorry, I mixed something up there. ISC is `dhclient`. The command blueman uses there is `dhclient -e IF_METRIC=100 -1 bnep0`. (Note that it's only about `bnep0` and has nothing to do with your `eth0`.)

kasi042, 2021-05-14:
Sorry, just tried it but currently the phone wouldn't even connect. Switching my desktop to dhcp always gives me network trouble. The NFS mounts are rejected "no permission" due to wrong IP address. Interestingly, blueman tries to check these mounts in my home. (See textfile).
[journalctl_blue.txt](https://github.com/blueman-project/blueman/files/6479859/journalctl_blue.txt)

`dmesg` says:
`[  237.543210] Bluetooth: hci0: corrupted ACL packet`
The dhclient command didn't return anything - probably due to unsuccessful pairing. I'm not sure if I can be of much further help.
:-/

cschramm, 2021-05-17:
I suppose the connection is fine and available as the bnep0 interface. blueman only fails to run the DHCP client or that client fails to obtain an IP address, so that IP remains unconfigured.

The dhclient gives neither output nor an non-zero exit code (try `echo $?` afterward, if your shell does not indicate it)? :thinking:

kasi042, 2021-05-20:
OK, now...
Just updated the fruit to 14.5.1 and got the latest LEAP update.
Switched to DHCP in YaST.
Did the pairing, tried to access the private hotspot.
The phone does say it's connected! That's a change! It didn't before. Then:
```
mars:~ # dhclient -e IF_METRIC=100 -1 bnep0
mars:~ # echo $?
1
mars:~ # ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 24:4b:fe:5b:1e:27 brd ff:ff:ff:ff:ff:ff
    inet 10.42.42.70/24 brd 10.42.42.255 scope global eth0
       valid_lft forever preferred_lft forever
3: bnep0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 00:04:0e:8c:f2:86 brd ff:ff:ff:ff:ff:ff
mars:~ # route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
10.42.42.0      0.0.0.0         255.255.255.0   U     0      0        0 eth0
mars:~ # 
```
Hm, now wait. In YaST network settings the only interface configured is eth0 - which I switched to DHCP (and back). But I probably have to setup bnep0 manually?

kasi042, 2021-05-20:
I'm not sophisticated enough for this. I used YaST to add bnep0 as interface. As I didn't get any better option I used the "eth" type. I checked It's local IP and added that as route. 
I switched on DHCP for it and connected to the NAP. Then I got this:
```
mars:~ # route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
10.__.__.0      0.0.0.0         255.255.255.0   U     0      0        0 eth0
172.20.10.0     0.0.0.0         255.255.255.240 U     0      0        0 bnep0

```
I became very bold and unplugged the ethernet. The I got that:
```

mars:~ # route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
localhost       0.0.0.0         255.255.255.0   U     0      0        0 eth0
localhost       0.0.0.0         255.255.255.240 U     0      0        0 bnep0
mars:~ # ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 24:4b:fe:5b:1e:27 brd ff:ff:ff:ff:ff:ff
    inet 10.__.__._/24 brd 10.__.__.255 scope global eth0
       valid_lft forever preferred_lft forever
4: bnep0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 00:04:0e:8c:f2:86 brd ff:ff:ff:ff:ff:ff
    inet 172.20.10.9/28 brd 172.20.10.15 scope global bnep0
       valid_lft forever preferred_lft forever
mars:~ # ping 172.20.10.9
PING 172.20.10.9 (172.20.10.9) 56(84) bytes of data.
64 bytes from 172.20.10.9: icmp_seq=1 ttl=64 time=0.025 ms
64 bytes from 172.20.10.9: icmp_seq=2 ttl=64 time=0.035 ms
64 bytes from 172.20.10.9: icmp_seq=3 ttl=64 time=0.033 ms
64 bytes from 172.20.10.9: icmp_seq=4 ttl=64 time=0.029 ms
^C
--- 172.20.10.9 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3042ms
rtt min/avg/max/mdev = 0.025/0.030/0.035/0.006 ms
mars:~ # ping 8.8.8.8
connect: Network is unreachable
```
 So, the phone has applied itself an IP address which I can ping. But trying to get "out" it says "Network is unreachable". 

Apparently - at least now - it's less a blueman issue than a wicked issue or rather an issue of my ability to set up network.

...

kasi042, 2021-05-20:
Ah, sorry. The phone has applied the address to my computer - not itself. So I have just pinged my own IP address.

cschramm, 2021-05-20:
Nothing needs to be done manually. Blueman runs a DHCP client for you on the connection.

So `dhclient -e IF_METRIC=100 -1 bnep0` fails for you as well, according to the exit code. Surprisingly it doesn't print anything. Maybe `-v` sheds some light?

kasi042, 2021-05-22:
OK...
I tried the verbose option - no response. Blueman applet said it could not get an IP address. IP -a nevertheless showed 172.20.10.9 but again, status "unknown". 
As you said nothing need to be done manually, I have removed bnep0 from YaST network settings. Tried again the NAP. 
IP -a then showed bnep0 as "down", no IP address. In both cases the phone showed the connection to the NAP.
So the entry in the  YaST settings did make a difference. I just don't know how. I'll add it again and search for useful info on the settings.

kasi042, 2021-05-22:
So I added bnep0 again in YaST. Blueman again complains about not getting the IP address and `IP a` shows the same IP address again. This is` journalctl | grep blue` from that point on:
```
May 22 16:03:31 mars bluetoothd[1261]: No cache for 80:82:23:1F:EB:EF
May 22 16:03:49 mars bluetoothd[1261]: bnep0 connected
May 22 16:03:49 mars dbus-daemon[1280]: [system] Activating via systemd: service name='org.blueman.Mechanism' unit='blueman-mechanism.service' requested by ':1.94' (uid=1000 pid=5582 comm="/usr/bin/python3 /usr/bin/blueman-applet ")
May 22 16:03:49 mars blueman-mechanism[13060]: Unable to init server: Could not connect: Connection refused
May 22 16:03:49 mars blueman-mechanism[13060]: Unable to init server: Verbindung ist gescheitert: Connection refused
May 22 16:03:49 mars blueman-mechani[13060]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
May 22 16:03:49 mars dbus-daemon[1280]: [system] Successfully activated service 'org.blueman.Mechanism'
May 22 16:03:50 mars blueman-mechanism[13060]: blueman-mechanism 16.03.50 ERROR    DhcpClient:71 _check_client: dhcp client failed with status code 1
May 22 16:03:50 mars blueman-mechani[13060]: blueman-mechanism ERROR DhcpClient:71 _check_client: dhcp client failed with status code 1
May 22 16:03:50 mars blueman-applet[5582]: blueman-applet 16.03.50 WARNING  DhcpClient:51 err       : g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: 1 (0)
```

kasi042, 2021-05-22:
They are communicating, however. I assume 172.20.10.1 would be the iPhone's router address?  I have monitored bnep0 with wireshark. It goes berserk when I start `nmap 172.20.10.1` by all the packages sent for port scanning. 
The route is set:
~ # route
Kernel IP routing table
```
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
...
172.20.10.0     0.0.0.0         255.255.255.240 U     0      0        0 bnep0
```
But it wouldn't send anything out:
```
ping -I bnep0 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 172.20.10.9 bnep0: 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
6 packets transmitted, 0 received, 100% packet loss, time 5115ms
pipe 4

```
In wireshark it's just sending a broadcast "who has 8.8.8.8". 
I don't know if any of this is helpful...

cschramm, 2021-05-22:
> I assume 172.20.10.1 would be the iPhone's router address?

I don't know. Assuming it is, then you should be able to ping it. Your connection seems fine, it just lacks a default route. I assume the IP setup comes from whatever you do in YaST and I guess it uses `wickedd-dhcp` on the interface. Not sure why it would not set a default route.

Anyway, back to topic:

> I tried the verbose option - no response

With response you mean output? I just had a closer look at dhclient and the interesting stuff should actually go to syslog, even if verbose output is enabled, so `sudo journalctl -t dhclient` should hopefully reveal what's going on when blueman or you unsuccessfully invoke the command.

kasi042, 2021-05-26:
Hi, sorry for the late reply. Outside life is getting more demanding with Spring kind of kicking in.
The - somehow good - news first: I'm writing this via my iPhone's NAP. ... But: I have disabled DHCP. Via YaST I have set up bnep0 with the static IP address 172.20.10.9 I have formerly received from the phone's DHCP server and set up the default route to 172.20.10.1, deleted my "safe haven" default route via eth0 plus manually added a DNS server. `ping 8.8.8.8` as well as `ping google.com` worked. I did unplug the cable and I'm writing this. Well, that's a lot of fuss and I did bad things to my network settings. But at least it's usable.
I'll reestablish my regular settings and then try your "back on topic". ;-)

kasi042, 2021-05-26:
Oh man:
```
mars:~ # dhclient -e -v IF_METRIC=100 -1 bnep0
mars:~ # journalctl -t dhclient
-- Logs begin at Wed 2021-05-26 22:09:30 CEST, end at Wed 2021-05-26 23:02:12 CEST. --
May 26 22:30:32 mars dhclient[11199]: /sbin/dhclient-script: unable to pre-init requested interfaces -- see log messages
May 26 22:30:32 mars dhclient[11199]: 
May 26 22:30:32 mars dhclient[11199]: If you think you have received this message due to a bug rather
May 26 22:30:32 mars dhclient[11199]: than a configuration issue please read the section on submitting
May 26 22:30:32 mars dhclient[11199]: bugs on either our web page at www.isc.org or in the README file
May 26 22:30:32 mars dhclient[11199]: before submitting a bug.  These pages explain the proper
May 26 22:30:32 mars dhclient[11199]: process and the information we find helpful for debugging.
May 26 22:30:32 mars dhclient[11199]: 
May 26 22:30:32 mars dhclient[11199]: exiting.
May 26 23:01:47 mars dhclient[24504]: /sbin/dhclient-script: unable to pre-init requested interfaces -- see log messages
May 26 23:01:47 mars dhclient[24504]: 
May 26 23:01:47 mars dhclient[24504]: If you think you have received this message due to a bug rather
May 26 23:01:47 mars dhclient[24504]: than a configuration issue please read the section on submitting
May 26 23:01:47 mars dhclient[24504]: bugs on either our web page at www.isc.org or in the README file
May 26 23:01:47 mars dhclient[24504]: before submitting a bug.  These pages explain the proper
May 26 23:01:47 mars dhclient[24504]: process and the information we find helpful for debugging.
May 26 23:01:47 mars dhclient[24504]: 
May 26 23:01:47 mars dhclient[24504]: exiting.
```
But again, `ip a` shows:
```
4: bnep0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 00:04:0e:8c:f2:86 brd ff:ff:ff:ff:ff:ff
    inet 172.20.10.9/28 brd 172.20.10.15 scope global bnep0
       valid_lft forever preferred_lft forever
```
I couldn't find any log file. Where could it be, if not in /var/log/ ?

kasi042, 2021-05-26:
@cschramm: Thank you so much for your support. As I said before, I don't actually need this. I am happy I got it running with wicked. 
For sure, that is rather a hack / workaround. As you said it should work automatically with DHCP. However, I'm using wicked because I have it wired and even use static IP. So, bluetooth tethering for NAP - at least for me - doesn't make sense. I really don't need to push it further with DHCP. I may still provide information if needed - if you think it worth pursuing. But I don't want to waste your time.
I just posted the information in the openSUSE forum. If Julian is still interested, maybe we can get some additional input.

cschramm, 2021-05-27:
`unable to pre-init requested interfaces` leads me to two or three occurrences on the web where dhclient is just magically not working on openSUSE as in your case. My guess would be that something is sabotaged by those YaST or wicked things or something.

`see log messages` seems rather pointless as you're just looking at them when reading that. :smile: You could try `journalctl` without arguments, maybe there is more with another identifier (maybe `dhclient-script`? :shrug:).

JulianBane, 2021-05-27:
Hi everyone. Well done for all the good efforts on this. Sadly my only real (as opposed to VM) SUSE machine died over a month ago so have not been able to reproduce or test the things here but have been fascinated by the path travelled.

I ronically this all started on a Raspberry Pi where the tethering failed (actually failed due to CPU overload due to a background task I know now) and moved it onto the SUSE box for debugging when I hit this issue!

Thanks for all the hard work.

kasi042, 2021-05-27:
Well, here's something that explains a lot:
```
mars:~ # journalctl | grep dhc
May 27 21:59:31 mars wickedd-dhcp4[1268]: bnep0: Request to acquire DHCPv4 lease with UUID b8f8af60-1a89-0100-5d05-000006000000
May 27 21:59:31 mars dhclient-script[2819]: dhclient conflicts with enabled wicked.service and disabled
May 27 21:59:31 mars dhclient[2775]: /sbin/dhclient-script: unable to pre-init requested interfaces -- see log messages
May 27 21:59:31 mars dhclient[2775]: 
May 27 21:59:31 mars dhclient[2775]: If you think you have received this message due to a bug rather
May 27 21:59:31 mars dhclient[2775]: than a configuration issue please read the section on submitting
May 27 21:59:31 mars dhclient[2775]: bugs on either our web page at www.isc.org or in the README file
May 27 21:59:31 mars dhclient[2775]: before submitting a bug.  These pages explain the proper
May 27 21:59:31 mars dhclient[2775]: process and the information we find helpful for debugging.
May 27 21:59:31 mars dhclient[2775]: 
May 27 21:59:31 mars dhclient[2775]: exiting.
May 27 21:59:32 mars blueman-mechanism[2694]: blueman-mechanism 21.59.32 ERROR    DhcpClient:71 _check_client: dhcp client failed with status code 1
May 27 21:59:32 mars blueman-mechani[2694]: blueman-mechanism ERROR DhcpClient:71 _check_client: dhcp client failed with status code 1
May 27 21:59:32 mars wickedd-dhcp4[1268]: bnep0: Committed DHCPv4 lease with address 172.20.10.9 (lease time 85536 sec, renew in 42768 sec, rebind in 74844 sec)
May 27 22:01:59 mars dhclient-script[3875]: dhclient conflicts with enabled wicked.service and disabled
May 27 22:01:59 mars dhclient-script[3881]: dhclient conflicts with enabled wicked.service and disabled
May 27 22:01:59 mars dhclient[3869]: /sbin/dhclient-script: unable to pre-init requested interfaces -- see log messages
May 27 22:01:59 mars dhclient[3869]: 
May 27 22:01:59 mars dhclient[3869]: If you think you have received this message due to a bug rather
May 27 22:01:59 mars dhclient[3869]: than a configuration issue please read the section on submitting
May 27 22:01:59 mars dhclient[3869]: bugs on either our web page at www.isc.org or in the README file
May 27 22:01:59 mars dhclient[3869]: before submitting a bug.  These pages explain the proper
May 27 22:01:59 mars dhclient[3869]: process and the information we find helpful for debugging.
May 27 22:01:59 mars dhclient[3869]: 
May 27 22:01:59 mars dhclient[3869]: exiting.
mars:~ # 
```
Had I left out the "p" in the same command here: [https://github.com/blueman-project/blueman/issues/1510#issuecomment-840111739](url)
we might have saved a lot of time. 
> dhclient conflicts with enabled wicked.service
So, as blueman is relying on dhclient which conflicts with wicked - I think we're at the end here. I don't know how much fuss it is to add support for wickedd-dhcp and if there's enough demand. It was fun finding out, at least. 
:-))

cschramm, 2021-05-27:
That's weird. If I'd install a dhclient package I'd expect it to work, so it should probably just define a conflict with the wicked package. Not much of an improvement, though, as the YaST stuff probably needs wicked, so removing or disabling it is not an option to you.

You do not necessarily need dhclient, though. openSUSE also has a udhcp package and blueman has support for its client as well. Unless it's also inoperative due to wickedness, it should be a way.

Otherwise, yes, we might need to add wicked support here. I guess it should not be complex but if feels very edgy as it's only an openSUSE thing.

closed 2021-08-03 by github-actions[bot]