blueman-project/blueman #1512: Slackware(64) 14.2

inukaze, 2021-03-18:
blueman: 2.0.6
BlueZ: 5.47
Distribution: Slackware(64) 14.2
Desktop environment: Xfce

* [ X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

i trying to do the follow, Open a Terminal : 

```
su
/etc/rc.d/rc.bluetooth start
exit
```

`blueman-applet &
`
`bluetoothctl`
[bluetooth]# `power on`
[bluetooth]# `discoverable on`
[bluetooth]# `pairable on`
[bluetooth]# `agent on`
[bluetooth]# `default-agent`
[bluetooth]# `scan on`
[bluetooth]# `devices`

> Device FE:F9:E7:B9:E9:1E Dualpow
> Device 00:00:46:65:82:01 InuDroide

[bluetooth]# `pair 00:00:46:65:82:01`
Attempting to pair with 00:00:46:65:82:01
Failed to pair: org.bluez.Error.AuthenticationRejected

[bluetooth]# `pair FE:F9:E7:B9:E9:1E`
Attempting to pair with FE:F9:E7:B9:E9:1E
Failed to pair: org.bluez.Error.AuthenticationRejected

[bluetooth]# `exit`
exit

i had try a lot of another things like change lines of : 
/etc/dbus-1/system.d/bluetooth.conf

`editor /etc/dbus-1/system.d/bluetooth.conf`

```
<policy user="root">
<allow own="org.bluez"/>
<allow send_destination="org.bluez"/>
<allow send_type="method_call"></allow>

<policy context="default">
<allow send_destination="org.bluez"/>
</policy>
```

But nothing working. the Dualpow is a Headset ( Headphones with mic ) , and the another device is just Android SmartPhone Doogee Voyager 2 ( DG310 ) . 

When i try to connect the phone never appears PIN request on the phone. just inmediatly is a **org.bluez.Error.AuthenticationRejected**

inukaze, 2021-03-24:
Just trying to pair SmartPhone, Doogee Voyager 2 or to pair Headset via Bluetooh

cschramm, 2021-03-26:
In you description you start blueman-applet, then start bluetoothctl, enable its agent and set it as a default, then try to pair in bluetoothctl. You're neither using blueman's agent nor UI if you do that, so this is not a blueman issue and I do not get why you start blueman-applet in the first place.

inukaze, 2021-03-27:
I make a various test, the first i try to pair using the applet, but ever i got errors  and never pair, after of that i try via bluetoothctl, and neither pair the device.

cschramm, 2021-03-27:
Alright, if you actually do use blueman, please provide the debugging output of blueman-applet where you try to pair. Also check the bluetoothd logs.

closed 2021-04-11 by github-actions[bot]