blueman-project/blueman #1513: gdk_wayland_window_handle_configure_popup: assertion 'impl->transient_for' failed

nschloe, 2021-03-20:
blueman: 2.1.4
BlueZ: 5.56
Distribution: Ubuntu 21.04
Desktop environment: Sway

I'm using blueman in [Sway](https://github.com/swaywm/sway) (with Wayland). When I try to connect a new mouse, it will abort after a while with
```
Connection Failed:  Software caused connection abort
```
The log at `/var/log/syslog` says
```
Mar 20 14:36:27 ishanca blueman-manager[180635]: Window 0x15c8d30 is a temporary window without parent, application will not be able to position it on screen.
Mar 20 14:36:27 ishanca blueman-manager[180635]: gdk_wayland_window_handle_configure_popup: assertion 'impl->transient_for' failed
```
so I assume blueman tries to display a window, but can't.

cschramm, 2021-03-21:
Please start blueman-manager manually with `blueman-manager --loglevel debug` and check if it gives something relevant when that happens.

nschloe, 2021-03-22:
Thanks for the reply. I did that and got
```
blueman-manager 11.17.37 DEBUG    ManagerDeviceMenu:259 generate  : Satechi M1 Mouse
blueman-manager 11.17.38 DEBUG    ManagerDeviceMenu:259 generate  : Satechi M1 Mouse
blueman-manager 11.17.38 DEBUG    ManagerDeviceMenu:259 generate  : Satechi M1 Mouse
Gdk-Message: 11:17:38.325: Window 0x30c2d30 is a temporary window without parent, application will not be able to position it on screen.

(blueman-manager:228896): Gdk-CRITICAL **: 11:17:38.338: gdk_wayland_window_handle_configure_popup: assertion 'impl->transient_for' failed
blueman-manager 11.17.39 INFO     ManagerDeviceMenu:78 set_op    : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f385357bd00 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2eeae40)>
blueman-manager 11.17.39 INFO     ManagerDeviceMenu:78 set_op    : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f385357b7c0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2eeb120)>
blueman-manager 11.18.19 INFO     ManagerDeviceMenu:161 fail      : fail: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: blueman.bluez.errors.DBusFailedError:  Software caused connection abort
 (0)
blueman-manager 11.18.19 INFO     ManagerDeviceMenu:91 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f385357bd00 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2eeae40)>
blueman-manager 11.18.19 DEBUG    ManagerDeviceMenu:259 generate  : Satechi M1 Mouse
blueman-manager 11.18.19 INFO     ManagerDeviceMenu:91 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f385357b7c0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2eeb120)>
```
(No connection is established.)

cschramm, 2021-03-22:
So this is a question on BlueZ' `Software caused connection abort`. Check your `dmesg`, it's probably a firmware or similar thing. Also check the `bluetoothd` logs.

nschloe, 2021-03-22:
Thanks for the hint. I've enabled debugging on the bluetooth daemon. When trying to connect, dmesg remains silent, but syslog says:
```
Mar 22 21:51:29 ishanca bluetoothd[298359]: src/device.c:device_connect_le() Connection attempt to: DB:10:41:95:E6:06
Mar 22 21:52:10 ishanca bluetoothd[298359]: src/device.c:att_connect_cb() connect to DB:10:41:95:E6:06: Connection refused (111)
```

cschramm, 2021-03-26:
I don't know how to overcome that but the Internets offer e.g. `btmgmt le on`.

nschloe, 2021-03-29:
I tried the  whole procedure in Gnome and it works without problems there. It seems that the pairing window isn't shown in Sway (which is usually use) -- This is the cause of the problem.

cschramm, 2021-03-30:
Did you check `blueman-applet --loglevel debug` (see the troubleshooting page)? The reason why no notification shows up should probably be visible there. What notification-daemon do you use? dunst?

tedk0n, 2021-04-26:
output from ```blueman-applet --loglevel debug```

```
blueman-applet 11.09.39 ERROR    PluginManager:91 load_plugin: Unable to load plugin module GameControllerWakelock
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/blueman/main/PluginManager.py", line 89, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/lib/python3.9/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 14, in <module>
    raise ImportError('This is not an X11 screen')``
```

closed 2021-07-03 by github-actions[bot]