blueman-project/blueman #1515: Question about Arabic translations

mhmdanas, 2021-03-25:
Sorry for removing the template; I just found it to not be needed in this issue.

I was taking a look at Blueman's Arabic translations through Weblate, and then I found out that some of the strings are _totally_ incorrect. Like, even if you used a translation service, there's no way that they would be _that_ wrong. I then went into https://github.com/blueman-project/blueman/blob/1cf77b268dff784b30e70b97acfadaa424e469da/po/ar.po to investigate.

I found out that all of the mistaken translations have the following format:
```po
# A bunch of lines that I don't know the meaning of.
#| msgid "string1"
msgid "string2"
msgstr "{{ translation of string1 }}"
```

One of the most glaring examples ([link](https://github.com/blueman-project/blueman/blob/1cf77b268dff784b30e70b97acfadaa424e469da/po/ar.po#L996)):
```po
#. translators: device class
#: blueman/DeviceClass.py:104
#, fuzzy
#| msgid "Audio Source"
msgid "Video camera"
msgstr "مصدر الصوت"
```

Note that `string2` is the one shown in Weblate.

I then traced the changes using `git blame` to 8b22b2c011e0a8ebeeb6ae634aaf5eb441ff9ce6 ([direct link to file changes in commit](https://github.com/blueman-project/blueman/commit/8b22b2c011e0a8ebeeb6ae634aaf5eb441ff9ce6#diff-08eb20da420c098c66900bd02a7eceeadc23c656ab49c3186ca91c926a6aebc2)).

Any idea about what might be happening here? I could fix them myself (and probably will), but I'm curious as to what's the cause.

cschramm, 2021-03-25:
That looks like a funny (or fuzzy :hand_over_mouth:) Weblate feature that I do not really understand. It looks like `Video camera`, `Video monitor`, and `Video conferencing` appeared as new source strings and for some reason completely unknown to me Weblate thought they might be close to `Audio Source` and initialized the translations accordingly, adding the `fuzzy` attribute which means that it shall be reviewed by a translator.

infirit, 2021-03-26:
The reason on weblate is "source change" but I have no clue why weblate/gettext did it this way.

We could revert that change in arabic, update the translation again with gettext and hope it gets update properly.

If not @mhmdanas, I have no clue when it comes to arabic. You may have to into weblate and review and update these 😒 .

mhmdanas, 2021-03-26:
I think it's better for me to do it manually. Anyway, since I know the reason now, I'll be closing this issue. Thanks for the clarification!

mhmdanas, 2021-03-26:
Forgot to say something: the issue is possibly present in other languages' translations. Maybe someone should check?

closed 2021-03-26 by mhmdanas

infirit, 2021-03-26:
@mhmdanas I checked the dutch file and it is correct right now.

I tried reverting and updating and it still does the fuz and gets it wrong :disappointed:

cschramm, 2021-03-26:
Weblate did and does that throughout languages. All translations marked as fuzzy need translator review, which is exactly what fuzzy means. In most cases, though, we'd be better off with no translation instead of what Weblate copied there. :see_no_evil: