blueman-project/blueman #1518: Blueman-applet unfortunate gotcha

namgang, 2021-04-16:
blueman: 2.1.2-1ubuntu0.2
BlueZ: 5.53-0ubuntu3
Distribution: Ubuntu Studio
Desktop environment: Xfce

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

This is a UX issue. The default action invoked by a single left-click on the Blueman applet is "Turn Bluetooth Off". (A single right-click brings up the notifier menu in my environment.)

A mistaken left-click on the applet thus turns Bluetooth off. If your mouse is a BT mouse you just shot yourself in the foot. It's happened to me twice on a new box that does not have any other pointing device. It is very disruptive.

My suggestion is have something less disruptive there, like "Make Bluetooth Visible", or, alternatively, ask confirmation.

cschramm, 2021-04-17:
I'm not sure what you mean. Left-clicking the applet icon should open the device manager window (and probably do nothing if AppIndicator is used). It's by no means wired to "Turn Bluetooth Off". :thinking:

namgang, 2021-04-17:
Ouch. In such case the Xfce status notifier setup is the real culprit. Sorry to have taken up your time.

cschramm, 2021-04-17:
What kind of setup is that? Could you test #1382, if it supports StatusNotifierItem / AppInficator?

closed 2021-06-26 by github-actions[bot]