blueman-project/blueman #1523: [Feature request] Add keyboard shortcuts to blueman

erdnaxe, 2021-04-22:
blueman: 2.1.4
BlueZ: 5.55
Distribution: NixOS 20.09
Desktop environment: i3+XFCE

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Currently to close Blueman "Bluetooth Devices" window I need to input `ALT+A` then `E`. Would it be possible to bind at least some "exit" shortcuts such as `ESC`, `CTRL-Q` or `CTRL-W`?

This is also problematic with the "Plugins" and "Bluetooth Adapters" windows witch are impossible to close on i3.

cschramm, 2021-04-24:
#1406 added support for `CTRL-Q` and `CTRL-W`. It's in master and will be part of 2.2.

erdnaxe, 2021-04-24:
Oh I was not aware of #1406, this is nice.
Does #1406 only apply to the main window, or also to "Plugins" and "Bluetooth Adapters" windows?

cschramm, 2021-04-25:
Unfortunately not. It only applies to blueman-manager. The plugin dialog is part of blueman-applet and the adapters are blueman-adapters. It would be the same logic for blueman-adapters but blueman-applet is not a GtkApplication, so it does not have the same `set_accels_for_action` method.

erdnaxe, 2021-04-25:
> This is also problematic with the "Plugins" and "Bluetooth Adapters" windows witch are impossible to close on i3.

It is much less problematic as I don't open these windows everyday, but I believe it is still an issue.
Feel free to change the title/close this issue if you think this should not be fixed.

cschramm, 2021-04-27:
As I said, it's trivial for blueman-adapters, so I guess it's worth it. Regarding blueman-applet we'd have to switch it to GtkApplication to get the same. I'm not sure if @infirit had any specific reason not to use it in #1304.

infirit, 2021-11-10:
> As I said, it's trivial for blueman-adapters, so I guess it's worth it. Regarding blueman-applet we'd have to switch it to GtkApplication to get the same. I'm not sure if @infirit had any specific reason not to use it in #1304.

Short answer, there was no need for what GtkApplication provides over GApplication. It *should* be trivial to make it GtkApplication though.

infirit, 2022-11-24:
Closing blueman-adapters with a shortcut/escape per #1946

infirit, 2022-12-10:
I have added closing the pluginwindow and blueman-adapters with `Ctrl+w` and `Escape` in #1927. Are there other areas that need a shortcut for blueman 2.4?

PR #1982 (2022-12-31): Add accels for closing windows/dialogs
If more are needed let me know.

closes #1523
closed 2022-12-31 by infirit