blueman-project/blueman #1524: blueman-applet fails to start when used as systemd service

cherti, 2021-04-24:
blueman: 2.1.4
BlueZ: 5.58
Distribution: Arch Linux
Desktop environment: sway

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

# Steps to reproduce

1. `systemctl --user start blueman-applet`
2. `systemctl --user status blueman-applet`

Expected behavior: `blueman-applet` runs and shows up in system tray
Observed behavior: `blueman-applet` throws an exception:
```
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 42, in <module>
    BluemanApplet()
  File "/usr/lib/python3.9/site-packages/blueman/main/Applet.py", line 17, in __init__
    setup_icon_path()
  File "/usr/lib/python3.9/site-packages/blueman/Functions.py", line 155, in setup_icon_path
    ic.prepend_search_path(ICON_PATH)
AttributeError: 'NoneType' object has no attribute 'prepend_search_path'
```

Minimal reproduction example:
```
#!/usr/bin/env python3

import gi
gi.require_version("Gtk", "3.0")
from gi.repository import Gtk
ic = Gtk.IconTheme.get_default()
print(ic)
```

If run  manually, this will print an IconTheme-object, if run inside a systemd user service it yields `None`.

The icon theme can be found, if specified, in `~/.config/gtk-3.0/settings.ini`, however, the Gtk-module doesn't seem to provide a function to manually load an explicit icon theme, as far as I can tell.

As my experience with Gtk is limited, I am therefore not sure what's the best fix for this, as an icon would be rather useful for proper use of the `blueman-applet`. So unfortunately I can currently only provide a problem description instead of a fix.

cschramm, 2021-04-24:
Sounds like you lack the `DISPLAY` variable (and / or a `graphical.target`) there and reminds me of #1086.

cherti, 2021-04-24:
somehow this issue slipped my attention. Seems to be identical to this, hence closing here.

closed 2021-04-24 by cherti