blueman-project/blueman #1525: do_g_signal: The notification was dismissed by the user | device_confirm_passkey: Operation not permitted

b3wii, 2021-04-27:
When in bluetooth setting on my android phone and I tap on the found entry(my pc) to establish a connection, {'Connected': True} appears in the log but as soon as i accept the auth dialog it shows {'Connected': False}.

The notification was NOT dismissed by the user but the logs shows this message. And also this message: device_confirm_passkey: Operation not permitted.

LOGS:

BluezAgent:262 _on_request_confirmation: Agent.RequestConfirmation
blueman-applet 15.12.51 INFO     Notification:239 do_g_signal: 2
blueman-applet 15.12.51 DEBUG    Notification:245 do_g_signal: The notification was dismissed by the user.
blueman-applet 15.13.25 INFO     BluezAgent:207 _on_cancel: Agent.Cancel
[image]

blueman-manager version 2.1.2 starting
blueman-applet 15.40.29 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 15.40.41 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 15.40.45 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 15.41.02 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 15.41.13 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}

Apr 27 15:40:45 bluetoothd[78076]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr xXx type 0 status 0xe
Apr 27 15:40:45 bluetoothd[78076]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Apr 27 15:40:45 bluetoothd[78076]: src/device.c:device_bonding_failed() status 14
Apr 27 15:40:45 bluetoothd[78076]: src/adapter.c:resume_discovery()
Apr 27 15:41:02 bluetoothd[78076]: src/adapter.c:connected_callback() hci0 device xXx connected eir_len 21
Apr 27 15:41:02 bluetoothd[78076]: src/adapter.c:user_confirm_request_callback() hci0 xXx confirm_hint 0
Apr 27 15:41:02 bluetoothd[78076]: src/device.c:new_auth() Requesting agent authentication for xXx
**Apr 27 15:41:02 bluetoothd[78076]: No agent available for request type 2
Apr 27 15:41:02 bluetoothd[78076]: device_confirm_passkey: Operation not permitted**
Apr 27 15:41:02 bluetoothd[78076]: src/adapter.c:btd_adapter_confirm_reply() hci0 addr xXx success 0
Apr 27 15:41:13 bluetoothd[78076]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr xXx type 0 status 0x5
Apr 27 15:41:13 bluetoothd[78076]: src/device.c:device_bonding_complete() bonding (nil) status 0x05
Apr 27 15:41:13 bluetoothd[78076]: src/device.c:device_bonding_failed() status 5
Apr 27 15:41:13 bluetoothd[78076]: src/adapter.c:resume_discovery()
Apr 27 15:41:13 bluetoothd[78076]: src/adapter.c:dev_disconnected() Device xXx disconnected, reason 3
Apr 27 15:41:13 bluetoothd[78076]: src/adapter.c:adapter_remove_connection()
Apr 27 15:41:13 bluetoothd[78076]: plugins/policy.c:disconnect_cb() reason 3
Apr 27 15:41:13 bluetoothd[78076]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr xXx type 0 status 0xe
Apr 27 15:41:13 bluetoothd[78076]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Apr 27 15:41:13 bluetoothd[78076]: src/device.c:device_bonding_failed() status 14
Apr 27 15:41:13 bluetoothd[78076]: src/adapter.c:resume_discovery()

b3wii, 2021-04-27:
When i click on the bloetooth icon in system tray and select devices, a window opens and shows my phone. When i click on the button "create a pairing with the device" this happens:

Apr 27 16:04:23 bluetoothd[78076]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr xXx type 0 status 0xe
Apr 27 16:04:23 bluetoothd[78076]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Apr 27 16:04:23 bluetoothd[78076]: src/device.c:device_bonding_failed() status 14
Apr 27 16:04:23 bluetoothd[78076]: src/adapter.c:resume_discovery()
Apr 27 16:05:24 bluetoothd[78076]: src/adapter.c:connected_callback() hci0 device xXx connected eir_len 5
Apr 27 16:05:24 bluetoothd[78076]: src/profile.c:ext_confirm() incoming connect from xXx
Apr 27 16:05:24 bluetoothd[78076]: src/service.c:btd_service_ref() 0x560dab4062b0: ref=2
Apr 27 16:05:24 bluetoothd[78076]: src/profile.c:ext_confirm() :1.53/Profile/HSPHSProfile/00001108-0000-1000-8000-00805f9b34fb authorizing connection from xXx
Apr 27 16:05:24 bluetoothd[78076]: Authentication attempt without agent
Apr 27 16:05:24 bluetoothd[78076]: :1.53/Profile/HSPHSProfile/00001108-0000-1000-8000-00805f9b34fb rejected xXx: org.bluez.Error.Rejected
Apr 27 16:05:24 bluetoothd[78076]: src/service.c:btd_service_unref() 0x560dab4062b0: ref=1
Apr 27 16:05:25 bluetoothd[78076]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from xXx
Apr 27 16:05:25 bluetoothd[78076]: Authentication attempt without agent
Apr 27 16:05:25 bluetoothd[78076]: Access denied: org.bluez.Error.Rejected
Apr 27 16:05:25 bluetoothd[78076]: profiles/audio/a2dp.c:channel_remove() chan 0x560dab41f4b0
Apr 27 16:05:25 bluetoothd[78076]: src/profile.c:ext_confirm() incoming connect from xXx
Apr 27 16:05:25 bluetoothd[78076]: src/service.c:btd_service_ref() 0x560dab4062b0: ref=2
Apr 27 16:05:25 bluetoothd[78076]: src/profile.c:ext_confirm() :1.53/Profile/HSPHSProfile/00001108-0000-1000-8000-00805f9b34fb authorizing connection from xXx
Apr 27 16:05:25 bluetoothd[78076]: Authentication attempt without agent
Apr 27 16:05:25 bluetoothd[78076]: :1.53/Profile/HSPHSProfile/00001108-0000-1000-8000-00805f9b34fb rejected xXx: org.bluez.Error.Rejected
Apr 27 16:05:25 bluetoothd[78076]: src/service.c:btd_service_unref() 0x560dab4062b0: ref=1
Apr 27 16:05:25 bluetoothd[78076]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from xXx
Apr 27 16:05:25 bluetoothd[78076]: Authentication attempt without agent
Apr 27 16:05:25 bluetoothd[78076]: Access denied: org.bluez.Error.Rejected
Apr 27 16:05:25 bluetoothd[78076]: profiles/audio/a2dp.c:channel_remove() chan 0x560dab423100
Apr 27 16:05:29 bluetoothd[78076]: src/adapter.c:dev_disconnected() Device xXx disconnected, reason 2
Apr 27 16:05:29 bluetoothd[78076]: src/adapter.c:adapter_remove_connection()
Apr 27 16:05:29 bluetoothd[78076]: plugins/policy.c:disconnect_cb() reason 2
Apr 27 16:05:29 bluetoothd[78076]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr xXx type 0 status 0xe
Apr 27 16:05:29 bluetoothd[78076]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Apr 27 16:05:29 bluetoothd[78076]: src/device.c:device_bonding_failed() status 14
Apr 27 16:05:29 bluetoothd[78076]: src/adapter.c:resume_discovery()
Apr 27 16:06:51 bluetoothd[78076]: src/adapter.c:remove_temp_devices() /org/bluez/hci0

cschramm, 2021-05-06:
> The notification was NOT dismissed by the user but the log shows this message.

Your notifications service says so. It sends [`NotificiationClosed`](https://specifications.freedesktop.org/notification-spec/latest/ar01s09.html#signal-notification-closed) with reason 2.

> And also this message: device_confirm_passkey: Operation not permitted.

That's just the consequence of `No agent available for request type 2`. blueman-applet provides a default agent that can handle that type, so possible explanations are that you restarted BlueZ so that the default agent got lost or that you overrode it with another, less capable default agent.

cschramm, 2021-05-12:
I just double-checked with my notifications service (mate-notification-daemon) and found that the signal is expected after the signal for the invoked action. Nothing to worry there.

> If the default agent was overwritten then blueman should show that information.

As far as I know BlueZ neither exposes the default agent, nor does it notify agents about a change, so that's hard to get. Also, it's perfectly fine for blueman if the user wants to use another agent.

> the button "create a pairing with the device"

This is by no means a blueman text, by the way. Using two GUIs would be a perfect explanation for another agent jumping in.

The only thing you could do to get behind the agent situation is check bluetoothd debug logs for agent registrations.

b3wii, 2021-05-12:
> The only thing you could do to get behind the agent situation is check bluetoothd debug logs for agent registrations.

Please let me know how to find out if default agent was overwritten, i only get this output:
```
May 12 19:19:24 bluetoothd[1498]: src/adapter.c:connected_callback() hci0 device xXx connected eir_len 21
May 12 19:19:24 bluetoothd[1498]: src/adapter.c:user_confirm_request_callback() hci0 xXx confirm_hint 0
May 12 19:19:24 bluetoothd[1498]: src/device.c:new_auth() Requesting agent authentication for xXx
May 12 19:19:24 bluetoothd[1498]: src/agent.c:agent_ref() 0x55c3f44a2980: ref=2
May 12 19:19:24 bluetoothd[1498]: src/agent.c:agent_request_confirmation() Calling Agent.RequestConfirmation: name=:1.70, path=/org/bluez/agent/blueman, passkey=942065
May 12 19:19:26 bluetoothd[1498]: src/adapter.c:new_settings_callback() Settings: 0x00000ad3
May 12 19:19:26 bluetoothd[1498]: src/adapter.c:settings_changed() Changed settings: 0x00000008
May 12 19:19:26 bluetoothd[1498]: src/adapter.c:settings_changed() Pending settings: 0x00000000
May 12 19:19:57 bluetoothd[1498]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr xXx type 0 status 0x5
May 12 19:19:57 bluetoothd[1498]: src/device.c:device_bonding_complete() bonding (nil) status 0x05
May 12 19:19:57 bluetoothd[1498]: src/agent.c:send_cancel_request() Sending Cancel request to :1.70, /org/bluez/agent/blueman
May 12 19:19:57 bluetoothd[1498]: src/device.c:device_cancel_authentication() Canceling authentication request for xXx
May 12 19:19:57 bluetoothd[1498]: src/agent.c:agent_unref() 0x55c3f44a2980: ref=1
May 12 19:19:57 bluetoothd[1498]: src/device.c:device_bonding_failed() status 5
May 12 19:19:57 bluetoothd[1498]: src/adapter.c:resume_discovery() 
May 12 19:19:57 bluetoothd[1498]: src/adapter.c:dev_disconnected() Device xXx disconnected, reason 0
May 12 19:19:57 bluetoothd[1498]: src/adapter.c:adapter_remove_connection() 
May 12 19:19:57 bluetoothd[1498]: plugins/policy.c:disconnect_cb() reason 0
May 12 19:19:57 bluetoothd[1498]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr xXx type 0 status 0xe
May 12 19:19:57 bluetoothd[1498]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
May 12 19:19:57 bluetoothd[1498]: src/device.c:device_bonding_failed() status 14
May 12 19:19:57 bluetoothd[1498]: src/adapter.c:resume_discovery() 
```
> the button "create a pairing with the device"

https://ibb.co/S06kSgB

b3wii, 2021-05-12:
```
[bluetooth]# pair my_phone
Attempting to pair with my_phone
[CHG] Device my_phone Connected: yes
Request confirmation
[agent] Confirm passkey 369273 (yes/no): yes
[CHG] Device my_phone Modalias: bluetooth:v00E0p107Ed1436
[CHG] Device my_phone UUIDs: 123456
[CHG] Device my_phone UUIDs: 123456
...
[CHG] Device my_phone ServicesResolved: yes
[CHG] Device my_phone Paired: yes
Pairing successful
[CHG] Device my_phone ServicesResolved: no
[CHG] Device my_phone Connected: no
[CHG] Device my_phone RSSI: -73
[CHG] Device my_phone RSSI: -65
```

cschramm, 2021-05-12:
I am confused. Your latest bluetoothd output does not show "No agent available for request type 2" anymore. blueman's agent gets used there.

b3wii, 2021-05-16:
> I am confused. Your latest bluetoothd output does not show "No agent available for request type 2" anymore. blueman's agent gets used there.

I'm just confused it doenst work. Please let me know what else i can provide to narrow down the problem. Thanx.

cschramm, 2021-05-17:
Alright, so the two messages we're talking about do not seem to actually be the topic here. Let's reset. What I could narrow down so far:

---

The problem description reads something like: Trying to connect to or from my Android phone fails. When trying from the Android device and after confirming the pairing, the device just appears as disconnected again. Trying from blueman also fails. I replicated the same issue with bluetoothctl.

Our issue template should have been filled in with something like:
blueman: 2.1.2
BlueZ: 5.53
Distribution: Linux Mint 20 or 20.1
Desktop environment:

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

---

As you see the same behavior in bluetoothctl without involving blueman, this is clearly not a blueman issue, but let's try anyway.

Just to be sure: When initiating from Android, you do confirm the key on *both* sides? Do you get both confirmations as well, when you initiate from blueman?

Please provide the debugging output from blueman-applet as described on the troubleshooting page of when trying one or the other process.

b3wii, 2021-05-17:
> Just to be sure: When initiating from Android, you do confirm the key on both sides? Do you get both confirmations as well, when you initiate from blueman?

I only get the dialog on the android side.

Try to connect from android to pc:
```
blueman-applet 22.47.40 INFO     PowerManager:123 request_power_state: Requesting True
blueman-applet 22.47.40 INFO     KillSwitch:138 on_power_state_change_requested: True
blueman-applet 22.47.40 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 22.47.40 INFO     KillSwitch:112 io_event  : killswitch changed 3
blueman-applet 22.47.40 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 22.47.40 INFO     PowerManager:171 update_power_state: off True | foff False | on True | current state False | new state False
blueman-applet 22.47.40 INFO     PowerManager:108 check     : callbacks done
blueman-applet 22.47.40 INFO     PowerManager:79 set_adapter_state: True
blueman-applet 22.47.40 INFO     PowerManager:171 update_power_state: off False | foff False | on True | current state False | new state True
blueman-applet 22.47.40 INFO     PowerManager:175 update_power_state: Signalling True
blueman-applet 22.47.40 INFO     RecentConns:91 initialize: rebuilding menu
blueman-applet 22.47.40 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Class': 1835276}
blueman-applet 22.47.40 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Class 1835276
blueman-applet 22.47.40 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Powered': True}
blueman-applet 22.47.40 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Powered True
blueman-applet 22.47.41 INFO     PowerManager:171 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 22.47.52 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'DiscoverableTimeout': 60}
blueman-applet 22.47.52 DEBUG    DiscvManager:82 on_adapter_property_changed: prop DiscoverableTimeout 60
blueman-applet 22.47.52 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Discoverable': True}
blueman-applet 22.47.52 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discoverable True
blueman-applet 22.48.10 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_BxXx {'Connected': True}
blueman-applet 22.48.10 INFO     BluezAgent:262 _on_request_confirmation: Agent.RequestConfirmation
blueman-applet 22.48.10 INFO     Notification:239 do_g_signal: 2
blueman-applet 22.48.10 DEBUG    Notification:245 do_g_signal: The notification was dismissed by the user.
blueman-applet 22.48.44 INFO     BluezAgent:207 _on_cancel: Agent.Cancel
blueman-applet 22.48.44 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_BxXx {'Connected': False}
blueman-applet 22.48.44 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_BxXx {'Connected': False}
blueman-applet 22.48.54 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Discoverable': False}
blueman-applet 22.48.54 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discoverable False
blueman-applet 22.48.54 INFO     DiscvManager:99 on_adapter_property_changed: Stop
```
Tried connection from blueman to my android phone:
```
blueman-manager version 2.1.2 starting
blueman-applet 22.57.35 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 22.57.35 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 22.57.41 INFO     RecentConns:161 notify    : <Device.Device object at 0x7fe63027bd00 (blueman+bluez+Device+Device at 0x227e0a0)> 00000000-0000-0000-0000-000000000000
blueman-applet 22.57.41 INFO     RecentConns:91 initialize: rebuilding menu
blueman-applet 22.57.41 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 22.57.41 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 22.57.43 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 22.57.44 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 22.57.55 INFO     RecentConns:161 notify    : <Device.Device object at 0x7fe63027bd00 (blueman+bluez+Device+Device at 0x227e0a0)> 00000000-0000-0000-0000-000000000000
blueman-applet 22.57.55 INFO     RecentConns:91 initialize: rebuilding menu
blueman-applet 22.57.56 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 22.57.57 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 22.58.01 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 22.58.01 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 22.58.09 INFO     RecentConns:161 notify    : <Device.Device object at 0x7fe63027bd00 (blueman+bluez+Device+Device at 0x227e0a0)> 00000000-0000-0000-0000-000000000000
blueman-applet 22.58.09 INFO     RecentConns:91 initialize: rebuilding menu
blueman-applet 22.58.10 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 22.58.10 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
blueman-applet 22.58.13 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
blueman-applet 22.58.13 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
```

b3wii, 2021-05-17:
> > Just to be sure: When initiating from Android, you do confirm the key on both sides? Do you get both confirmations as well, when you initiate from blueman?
> 
> I only get the dialog on the android side.
> 
> Try to connect from android to pc:
> 
> ```
> blueman-applet 22.47.40 INFO     PowerManager:123 request_power_state: Requesting True
> blueman-applet 22.47.40 INFO     KillSwitch:138 on_power_state_change_requested: True
> blueman-applet 22.47.40 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: True
> blueman-applet 22.47.40 INFO     KillSwitch:112 io_event  : killswitch changed 3
> blueman-applet 22.47.40 INFO     KillSwitch:122 io_event  : State: True
> blueman-applet 22.47.40 INFO     PowerManager:171 update_power_state: off True | foff False | on True | current state False | new state False
> blueman-applet 22.47.40 INFO     PowerManager:108 check     : callbacks done
> blueman-applet 22.47.40 INFO     PowerManager:79 set_adapter_state: True
> blueman-applet 22.47.40 INFO     PowerManager:171 update_power_state: off False | foff False | on True | current state False | new state True
> blueman-applet 22.47.40 INFO     PowerManager:175 update_power_state: Signalling True
> blueman-applet 22.47.40 INFO     RecentConns:91 initialize: rebuilding menu
> blueman-applet 22.47.40 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Class': 1835276}
> blueman-applet 22.47.40 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Class 1835276
> blueman-applet 22.47.40 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Powered': True}
> blueman-applet 22.47.40 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Powered True
> blueman-applet 22.47.41 INFO     PowerManager:171 update_power_state: off False | foff False | on True | current state True | new state True
> blueman-applet 22.47.52 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'DiscoverableTimeout': 60}
> blueman-applet 22.47.52 DEBUG    DiscvManager:82 on_adapter_property_changed: prop DiscoverableTimeout 60
> blueman-applet 22.47.52 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Discoverable': True}
> blueman-applet 22.47.52 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discoverable True
> blueman-applet 22.48.10 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_BxXx {'Connected': True}
> blueman-applet 22.48.10 INFO     BluezAgent:262 _on_request_confirmation: Agent.RequestConfirmation
> blueman-applet 22.48.10 INFO     Notification:239 do_g_signal: 2
> blueman-applet 22.48.10 DEBUG    Notification:245 do_g_signal: The notification was dismissed by the user.
> blueman-applet 22.48.44 INFO     BluezAgent:207 _on_cancel: Agent.Cancel
> blueman-applet 22.48.44 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_BxXx {'Connected': False}
> blueman-applet 22.48.44 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_BxXx {'Connected': False}
> blueman-applet 22.48.54 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Discoverable': False}
> blueman-applet 22.48.54 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discoverable False
> blueman-applet 22.48.54 INFO     DiscvManager:99 on_adapter_property_changed: Stop
> ```
> 
> Tried connecting from blueman to my android phone:
> 
> ```
> blueman-manager version 2.1.2 starting
> blueman-applet 22.57.35 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> blueman-applet 22.57.35 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> blueman-applet 22.57.41 INFO     RecentConns:161 notify    : <Device.Device object at 0x7fe63027bd00 (blueman+bluez+Device+Device at 0x227e0a0)> 00000000-0000-0000-0000-000000000000
> blueman-applet 22.57.41 INFO     RecentConns:91 initialize: rebuilding menu
> blueman-applet 22.57.41 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
> blueman-applet 22.57.41 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> blueman-applet 22.57.43 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
> blueman-applet 22.57.44 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> blueman-applet 22.57.55 INFO     RecentConns:161 notify    : <Device.Device object at 0x7fe63027bd00 (blueman+bluez+Device+Device at 0x227e0a0)> 00000000-0000-0000-0000-000000000000
> blueman-applet 22.57.55 INFO     RecentConns:91 initialize: rebuilding menu
> blueman-applet 22.57.56 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
> blueman-applet 22.57.57 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> blueman-applet 22.58.01 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
> blueman-applet 22.58.01 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> blueman-applet 22.58.09 INFO     RecentConns:161 notify    : <Device.Device object at 0x7fe63027bd00 (blueman+bluez+Device+Device at 0x227e0a0)> 00000000-0000-0000-0000-000000000000
> blueman-applet 22.58.09 INFO     RecentConns:91 initialize: rebuilding menu
> blueman-applet 22.58.10 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
> blueman-applet 22.58.10 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> blueman-applet 22.58.13 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': True}
> blueman-applet 22.58.13 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_xXx {'Connected': False}
> ```

cschramm, 2021-05-18:
In the first case a notification or dialog should show up as blueman-applet does get a request for that. Did you check https://github.com/blueman-project/blueman/wiki/Troubleshooting#pairing-fails-on-the-remote-devices-side-wihout-a-pairing-dialog-showing yet?

In the second case blueman-applet does not get a request. I'd ask you to check bluetoothd debugging logs for that. The log from your second post seems to show such a case but it looks like BlueZ had no default agent registered there. You might have to restart blueman-applet after restarting bluetoothd to avoid.

b3wii, 2021-06-18:
I tried it with a different machine and up to date mint distro and it worked right away. @cschramm if your interested in debugging this via teamviewer or anydesk just let me know.

closed 2021-07-08 by b3wii