blueman-project/blueman #1530: Blueman does not respect Terminal=true in desktop files for default applications when receiving files

nabakolu, 2021-05-04:
blueman: 2.1.4-1
BlueZ: 5.58-1
Distribution: Arch Linux
Desktop environment: dwm

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When receiving files via Bluetooth (in my case a text file) where the default application for that file type is a terminal application (in my case vim), blueman does not open that application in a terminal. Instead the desktop environment freezes. I suspect, that the application gets started in the tty behind the desktop environment, because when killing the desktop environment (from another tty) I can see the end of a vim buffer in the tty.
When using other files like pdfs it works perfectly fine and when setting gvim as the default application for text files, it works fine as well.

**Steps to reproduce:**
Create a desktop entry `/usr/share/applications/vim.desktop` for vim with `Exec=vim` and `Terminal=true` and make that the default application for text files with `xdg-mime default vim.desktop text/plain`. Then send a text file from another system and open it via the notification that pops up.

cschramm, 2021-05-06:
Works fine for me (with the vim.desktop file from Arch's vim 8.2.2815-1).

What happens with
```
python3 -c 'from blueman.Functions import launch; launch("xdg-open", paths=["/tmp/test.txt"], system=True)'
```
Does it open a terminal window?

nabakolu, 2021-05-07:
> Works fine for me (with the vim.desktop file from Arch's vim 8.2.2815-1).
> 
> What happens with
> 
> ```
> python3 -c 'from blueman.Functions import launch; launch("xdg-open", paths=["/tmp/test.txt"], system=True)'
> ```
> 
> Does it open a terminal window?

It opens the file in the terminal in vim, but pressing any key gives the following error and then opens the file in firefox.

```
lukas ~ $ Vim: Warning: Input is not from a terminal
Vim: Error reading input, exiting...
Vim: Finished.

/usr/bin/xdg-open: line 881: x-www-browser: command not found
```

nabakolu, 2021-05-07:
Yea the same terminal. This seems to be a problem with xdg-open, I made a script run outside of a terminal, which called xdg-open, which did not spawn a new terminal. I'm just gonna rewrite my desktop files

closed 2021-05-07 by nabakolu