blueman-project/blueman #1531: connection failed:blueman.bluez.errors.DBusFailedError:No route to host.

harekrishna-108, 2021-05-06:
blueman:2.0.8-1+deb10u1
BlueZ:5.50-1.2~deb10u1+rpt2
Distribution:Twister OS 2.0.1
Desktop environment:Twister OS 2.0.1

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.
* [ ] i search for my bluetooth device and click on pair.
* [ ] but, there is no notification saying the device is paired.
* [ ] i close the blueman dialog box and reopen it. there it shows my deive is paired.
* [ ] then, if i go to connect it as an audio sink/headset,
* [ ] i get an error saying: connection failed:blueman.bluez.errors.DBusFailedError:No route to host.

note:
I had uninstalled and reinstalled blueman once and it had worked. 
But, after reboot, not working, tried many times...
Please help...

harekrishna-108, 2021-05-06:
I am a noob by the way...

harekrishna-108, 2021-05-06:
i am using a raspberry pi 4 and its bluetooth adapter is fine

cschramm, 2021-05-06:
(That's a fairly old version of blueman.) I'd guess that the adapter is not powered. Try `bluetoothctl power on`.

harekrishna-108, 2021-05-07:
> (That's a fairly old version of blueman.) I'd guess that the adapter is not powered. Try `bluetoothctl power on`.

As I said, the adapter is working fine. I am able to search for other Bluetooth device also.

It is just that I am not able to properly pair up and connect as an audio sink.

harekrishna-108, 2021-05-07:
I did try this from the trouble shooting page.
Pairing fails on the remote device's side wihout a pairing dialog showing
There might be some issue when blueman tries to display the pairing notification. To confirm, open bluetoothctl, enable its agent with the agent on command, and issue a pair command for the device. You should get a request there. If not, you're having a more generic issue.

To inspect the issue with notifications, first type:

head -1 $(which blueman-applet)
This should show you which python version blueman uses in case you have multiple. Then fire up a python console and type:

from blueman.gui.Notification import Notification
Notification("Title", "Message", 0, [["action1", "Action 1"], ["action2", "Action 2"]]).show()

I don't know how to open python.
I tried typing python in the terminal and then the first line of the above mentioned code. But, it said, blueman.gui.Notification not found...

cschramm, 2021-05-07:
That does not seem to be your problem, but regarding Python your distribution is obviously based on Debian buster, so `/usr/bin/python` most probably points to Python 2 but blueman is installed for Python 3. `python3` should work.

My guess was based on the fact that the only case where BlueZ itself uses EHOSTUNREACH (which translates to `No route to host.`) is when the adapter is not powered. Of course there are more components where it could come from, including the Bluetooth stack in the Kernel (however that case seems to boil down to that the adapter associated with the device is not available or powered) and pulseaudio. Check the logs of the `bluetoothd` and `pulseaudio` services. As it's an error, they probably have something even without debugging.

harekrishna-108, 2021-05-07:
> That does not seem to be your problem, but regarding Python your distribution is obviously based on Debian buster, so `/usr/bin/python` most probably points to Python 2 but blueman is installed for [...]

could you please guide me on how to find the log files.
[image]

harekrishna-108, 2021-05-07:
[image]
if i try to add a bluetooth device, this is the error i get. it is very random. i was able to pair twice while troubleshooting. but, after reboot , am not able to.

harekrishna-108, 2021-05-07:
[image]
[image]
After pairing: we need to get this:
[image]
but, i get this:
[image]

cschramm, 2021-05-07:
I don't know your system but I assume `journalctl -u bluetooth.service` and `journalctl --user -u pulseaudio.service` should show logs for those services.

harekrishna-108, 2021-05-07:
> I don't know your system but I assume `journalctl -u bluetooth.service` and `journalctl --user -u pulseaudio.service` should show logs for those services.

thankyou very much. but, as i said, it works on random occasions. so, when it worked, i paired up all my bluetooth devices. so, shouldnt be problem. thankyou very much for your support.

harekrishna-108, 2021-05-07:
[image]
[image]
here are the details.

harekrishna-108, 2021-05-07:
43

SAP stands for SIM Access Profile, so you have to disable it:

Open /etc/systemd/system/bluetooth.target.wants/bluetooth.service
Change:

ExecStart=/usr/lib/bluetooth/bluetoothd
To

ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap
Reload the systemd:

 sudo systemctl daemon-reload
Restart the bluetooth:

 sudo service bluetooth restart
Get the bluetooth status:

 sudo service bluetooth status

bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Sat 2016-04-30 10:38:46 UTC; 6s ago
     Docs: man:bluetoothd(8)
 Main PID: 12775 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─12775 /usr/lib/bluetooth/bluetoothd --noplugin=sap

harekrishna-108, 2021-05-07:
> 43
>
> SAP stands for SIM Access Profile, so you have to disable it:
>
> Open /etc/systemd/system/bluetooth.target.wants/bluetooth.service
> Change:
>
> ExecStart=/usr/lib/bluetooth/bluetoothd
> To [...]

i tried this and it works perfectly.

harekrishna-108, 2021-05-07:
i found an awesome solution.
sudo systemctl restart bluetooth
after every boot, if this command is executed, and bluetooth device removed and again paired, it works perfectly, even after reboot.

harekrishna-108, 2021-05-07:
[image]
after this popup, i again executed`sudo systemctl restart bluetooth` and its working flawless.
here are the logs.
[image]

like really, inspecting these logs are amazing. they are the source to know the problem to troubleshoot them.
i really thank you very much for your amazing help. 
HARE KRISHNA!

closed 2021-07-15 by github-actions[bot]