blueman-project/blueman #1535: Exception thrown when trying to remove a device

RudiAusBuddeln, 2021-05-10:
blueman: blueman-2.1.4-1
BlueZ: bluez-5.58-1
Distribution: Arch
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.
 
Log (bluetooth-manager): https://gist.github.com/RudiAusBuddeln/b8df650d85620cbc352419995ac161e2

So I accidentally connected to my headphones while they were not ready, which always fucks it up (not a problem generally). However, now I cannot remove the device from the list. This is very bad as I will need to use them soon. Is there a way to delete a cache file or something with the registered devices?

cschramm, 2021-05-11:
I would expect that to happen if the adapter that the device is connected to is not powered. Could it be that you have multiple adapters?

RudiAusBuddeln, 2021-05-12:
No, I only have a single bluetooth card.

But anyway; I can't reproduce it anymore because I cleared the directory. It has been working fine for months. So I'll close for now.

closed 2021-05-12 by RudiAusBuddeln