blueman-project/blueman #1537: blueman-applet menu : all icons left of the menu items are missing

souravdas142, 2021-05-17:
blueman: using `master` upto commit b1fb0459f9f1b22f09db3c725e3772dfd2e85008
BlueZ: 5.58
Distribution: Arch-Linux
Kernel : 5.12.4
Desktop environment: i3wm
Bar : polybar (3.5.5)

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

**Problem Description -**
I have 2 laptotp, Hp, Dell. Both are running on Arch-Linux (kernel 5.12.4) with `polybar`. On my hp laptop everything is ok, but on **my dell laptop all `blueman-applet` menu icons are missing.**

Giving the screenshot of the bug : from dell

machine | screenshot | Description
--- | --- | ---
Dell | [image] |  <p align=center> <b>This is Bug</b> <br /> <br /> <br /> You can Clearly see all icons from the menu item like `send files to device`,`Recent connection`, `Device`,`Adapters` etc are missing.
hp | [image] | <p align=center> <b>This is expected behaviour</b> <br /> <br /> <br /> And here on `hp` machine all `blueman-applet` menu icons are showing on the left of the `send files to device`,`Recent connection`, `Device`,`Adapters` this menu items.

Though on the Dell machine, all icons are shown under `blueman-manager` window. Here is the screenshot :

[image]
:information_source:  **But on the window menu bar there also all icons are missing**

**To Reproduce** 
Don't know how to reproduce, just compiled from the project with configuration options below : 

```bash
    ./configure \
        --prefix=/usr \
        --sysconfdir=/etc \
        --libexecdir=/usr/lib/${_pkgname} \
        --with-dhcp-config='/etc/dhcp.conf' \
        --enable-nautilus-sendto \
        --enable-appindicator \
        --disable-static
```

here is the log i get from running `blueman-applet --loglevel debug` [blueman-applet.log](https://pastebin.com/ddbJZPdB)

Please help man.

cschramm, 2021-05-18:
Which icon theme do you use?

Both systems up to date, so we can assume equal versions for libraries?

souravdas142, 2021-05-18:
@cschramm  The problem persists on any icon theme or gtk theme. Though for information currently using a custom icon theme which inherits the `breeze` icon theme. Yes have equal version of libraries.

cschramm, 2021-05-18:
Does any of the systems use AppIndicator? Easiest way to find out is to `killall blueman-tray` and then starting it in a terminal with `blueman-tray --loglevel debug`. It will tell you which indicator it uses.

souravdas142, 2021-05-18:
this is what i get from  `blueman-tray --loglevel debug`

```bash
blueman-tray 12.33.46 DEBUG    Tray:24 _on_name_appeared: Applet started on name org.blueman.Applet, showing indicator
blueman-tray 12.33.46 INFO     Tray:28 _on_name_appeared: Using indicator "AppIndicator"

(blueman-tray:1169): Gdk-CRITICAL **: 12:33:46.642: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
```

souravdas142, 2021-05-18:
:fire: **The Problem is Solved**

There is a similar issue had been reported on mate-desktop/mate-panel#521, and you guys were followed that issue too. Getting a hint from there, I had to put `gtk-menu-images=1` under `~/.config/gtk-3.0/settings.ini` to change the behaviour of menu icon under AppIndicator menu. Thank you @cschramm for you effort. You may close the issue now.

cschramm, 2021-05-18:
If I'm not mistaken, `GtkMenuItem`s do not exist in GTK 4 at all, do they?

I do not have that problem on MATE. Also no settings.ini anywhere as far as I see. So you had the setting in place on the HP machine but not on the Dell machine? :thinking:

souravdas142, 2021-05-19:
> If I'm not mistaken, GtkMenuItems do not exist in GTK 4 at all, do they?     

Ohh.. Sorry, I had not check it :sweat_smile: . Can it be still implemented with menu icon in gtk4+ `GtkPopoverMenu` I guess? thought don't know whether menu items icon will support or not.

> Also no settings.ini anywhere as far as I see      

from the arch [wiki](https://wiki.archlinux.org/title/GTK#Configuration) all gtk related confg files resides under `~/.config` or `/etc`

> So you had the setting in place on the HP machine but not on the Dell machine? :thinking:

I forget to included any `gtk` related settings file on my `dotfiles` thats my fault and the error occurred in looks and feel.

cschramm, 2021-05-19:
I have no idea about the problem, to be honest, especially because I do not get why I don't have it although I'm using MATE without any GTK configuration, but...

We have a StatusNotifierItem implementation for the indicator ready at #1382. It's meant to replace the AppIndicator implementation. What the AppIndicator implementation does is generate a GtkMenu including the GtkImageMenuItems and pass it on to libappindicator, which passes it to libdbusmenu, which then parses the widget and builds a `com.canonical.dbusmenu` D-Bus object out of it. In contrast, the StatusNotifierItem implementation builds a `com.canonical.dbusmenu` D-Bus object itself, so there is no GtkMenu involved (on our side) at all but we just pass the icon name via D-Bus.

I don't get the problem, so I'm not sure if it's related to the GtkImageMenuItem that's build, passed on and parsed in blueman-tray (then again: why don't we have the problem in blueman-manager menus as well?) or if it's just the panel's handling of the icon name it receives (I do not see *any* connection to blueman then). If it's the latter that's no improvement, of course, but if it's the first one, then that problematic flow will be gone completely.

souravdas142, 2021-05-20:
@cschramm  Do you believe vanilla `Arch Linux` follows pretty much bare install? Means for the first time there is no display server at all. yesterday I installed arch  on a vm and pass through my bluetooth adapter to it. Now there is only have xserver, i3, dunst, picom, polybar, gtk3, hicolor(icon theme), and blueman, I am giving a brief what is installed with blueman.
 
```bash
[2021-05-19T11:44:01+0530] [ALPM] installed libdbusmenu-glib (16.04.0-4)
[2021-05-19T11:44:01+0530] [ALPM] installed libdbusmenu-gtk3 (16.04.0-4)
[2021-05-19T11:44:01+0530] [ALPM] installed libindicator-gtk3 (12.10.1-9)
[2021-05-19T11:44:01+0530] [ALPM] installed libappindicator-gtk3 (12.10.0.r296-1)
[2021-05-20T07:22:55+0530] [ALPM] installed libical (3.0.10-1)
[2021-05-20T07:22:55+0530] [ALPM] installed bluez (5.58-1)
[2021-05-20T07:22:55+0530] [ALPM] installed python-cairo (1.20.0-3)
[2021-05-20T07:22:55+0530] [ALPM] installed gobject-introspection-runtime (1.68.0-1)
[2021-05-20T07:22:55+0530] [ALPM] installed python-gobject (3.40.1-1)
[2021-05-20T07:22:55+0530] [ALPM] installed blueman (2.1.4-1)
```
## Steps to reproduce with above situation
Till that there is no configs at `~/.config/gtk-3.0` or at `/etc/gtk-3.0`
1. Now if I start blueman, Voilla, In the tray blueman icon there is no menu items icon under appindicator menu.
2. Now Create a directory `gtk-3.0` under `~/config` 
3. create a `settings.ini` under `~/.config/gtk-3.0/` containg only below 2 lines.
```bash
     [Settings]
     gtk-menu-images=1
```
4. Now logout and login back to change the effect.
5. Click on the tray blueman icon (appIndicator), Voilla you can see all menu items icon there.
6. Go to step 1 with deleted the gtk `confdir` at `~/.config/gtk-3.0/`

for further information about what tools, apps are installed on my arch vm see [here](https://pastebin.com/jKQ396j0) for further debugging.

May be `Mate` modified their `panel` or something to change this behaviour natively, may you don't see any problem on `Mate` for this. I don't ever use `Mate` so I don't know about that.

cschramm, 2021-05-20:
Okay, I got it. Your reference to the MATE issue got me on a wrong track. So, gtk-menu-images is turned off by default in GTK+ 3. MATE actually has own overrides (GSettings `org.mate.interface`) and there it (`menus-have-icons`) is turned on by default. That's why icons work for me. https://wiki.archlinux.org/title/GTK#Configuration_tools sounds a lot like it's the same for Cinnamon, Enlightenment, GNOME, and Xfce as well. Not for i3.

The GTK+ 3 default means no icons in our current indicator implementations and should also mean no icons in the manager window (the window menu and devices' right-click menus). Using our new StatusNotifierItem draft mentioned earlier more or less avoids that for the tray icon as we just pass on the icon name via D-Bus. Actually, I'd say it depends on the application that implements the tray. If it uses a GtkImageMenuItems for the StatusNotifierItems, you have the same problem again.

souravdas142, 2021-05-20:
> Your reference to the MATE issue got me on a wrong track       

Sorry for that. I should have explained it clearly, actually from [this comment](https://github.com/mate-desktop/mate-panel/issues/521#issuecomment-276015640) on same `MATE` reference I got the hint to solve this, And you could see in the problem description he gave the same screenshot like I give here with the same menu items icon missing issue.

> Using our new StatusNotifierItem draft mentioned earlier more or less avoids that for the tray icon as we just pass on the icon name via D-Bus

When will this stuff get merged with `master`? have any chance shipped with v2.2? I want to test it. 

> Actually, I'd say it depends on the application that implements the tray. If it uses a GtkImageMenuItems for the StatusNotifierItems, you have the same problem again.

Exactly, the default value for this option is false, as the option is deprecated. And they give a alternative way to do this which i mentioned previously.

And lastly, have any chance to shift this project to gtk4?

cschramm, 2021-05-20:
> When will this stuff get merged with master? have any chance shipped with v2.2? I want to test it.

2.2 is around the corner, I'm basically just waiting for a thumbs up from @infirit in form of an approval of #1529. Unfortunately he's not very active lately (no pressure :grin:).

I'll merge #1382 into master right after 2.2 is done. I'm not confident enough in it to include it in the release but there are no (ambitious) plans for a 2.3, so that should follow soon.

> And lastly, have any chance to shift this project to gtk4?

Sure, we definitely want to do that at some point. There has been some discussion in #752 but not much work going on. Contributions are welcome.

souravdas142, 2021-05-21:
> 2.2 is around the corner, I'm basically just waiting for a thumbs up from @infirit in form of an approval of #1529. Unfortunately he's not very active lately (no pressure :grin:). 

 good luck man. :smile: 

> I'll merge #1382 into master right after 2.2 is done. I'm not confident enough in it to include it in the release but there are no (ambitious) plans for a 2.3, so that should follow soon.

is that PR good enough for have a try? I just want to try that.

**I have one more feature request regarding on tray AppIndicator icon.**
currently when bluetooth is connected with some devices, AppIndicator Icon is changed to blueman-active icon,
Now if we can detect a bluetooth audio device connected with bluetooth, so can we change the blueman tray AppIndicator icon something like [image] this icon? and for normal all other devices the icon remain same(blueman-active)

cschramm, 2021-05-21:
> is that PR good enough for have a try? I just want to try that.

You're absolutely welcome. Based on the blueman-git AUR package it should be as easy as

```
git clone https://aur.archlinux.org/blueman-git.git
cd blueman-git
sed -i 's/blueman-project\/${_pkgname}.git/cschramm\/blueman.git#branch=sni/' PKGBUILD
makepkg
sudo pacman -U blueman-git-2.0.r1323.g5b2332e7-1-x86_64.pkg.tar.zst
```

souravdas142, 2021-05-22:
Hi sorry for the late...!!

using your sni branch 5b2332e753a0717e6609d30955054028e13c39d1, I can't start  `blueman-manager`, there are following error.

```bash
Traceback (most recent call last):
  File "/usr/bin/blueman-manager", line 16, in <module>
    from blueman.main.Manager import Blueman
  File "/usr/lib/python3.9/site-packages/blueman/main/Manager.py", line 10, in <module>
    from blueman.gui.manager.ManagerDeviceList import ManagerDeviceList
  File "/usr/lib/python3.9/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 12, in <module>
    from blueman.gui.manager.ManagerDeviceMenu import ManagerDeviceMenu
  File "/usr/lib/python3.9/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 10, in <module>
    from blueman.config.AutoConnectConfig import AutoConnectConfig
ModuleNotFoundError: No module named 'blueman.config'
```
here is the log got from `blueman-applet --loglevel debug` [sni-blueman-applet.log](https://pastebin.com/DsR9Zwx4)

souravdas142, 2021-05-22:
> The GTK+ 3 default means no icons in our current indicator implementations and should also mean no icons in the manager window (the window menu and devices' right-click menus). Using our new [...]

This claim, As far I understand The menu items icon should show with the default (gtk-menu-images is false) setting of gtk3 right (correct me if i am wrong) ? But with your `sni` branch still Menu Items icon not showing and its effected with `gtk-menu-images=0|1` like before. The only difference I am noticing the `right` click on blueman-tray icon shows the tray menu and `left` click open the `blueman-manager` window. As in my case `blueman-manager` is crashed its not opening with the `left` click.

cschramm, 2021-05-22:
I just rebased the branch on master so that blueman-manager works.

> with your `sni` branch still Menu Items icon not showing and its effected with `gtk-menu-images=0|1` like before.

Too bad. There's nothing we can do about that. It's the applet implementation that uses GTK+ 3 there then.

While I'm not sure if the generation and parsing of the GtkMenu in the AppIndicator plugin (master and all releases) depends on the `gtk-menu-images` setting (as nothing gets actually rendered, just parsed), there's no GTK+ 3 involved in the StatusNotifierItem plugin (sni branch) at all.

In both cases we rely on an applet implementation that picks up the StatusNotifierItem and renders the menu accordingly. If that applet implementation happens to use GtkImageMenuItems to do so, it most probably depends on the `gtk-menu-images` setting again.

In your case it should be polybar, right? :thinking: From what I found on in, I'm not even sure if it supports StatusNotifierItem.

souravdas142, 2021-05-24:
A quick inspection, [blueman-manager crash](https://github.com/blueman-project/blueman/issues/1537#issuecomment-846408337) issue is fixed, as you rebased the branch on master, **this is a draft comment...**

cschramm, 2021-05-25:
Damn. I forgot that libappindicator has a GtkStatusIcon (XEmbed) fallback, so I guess when `blueman-tray --loglevel debug` says the AppIndicator implementation gets used, it's actually the fallback and when you try the sni branch, `blueman-tray --loglevel debug` will probably *not* show the `StatusNotifierItem` implementation but our `GtkStatusIcon` implementation.

How is master with snixembed? As far as I can see, snixembed just relies on libdbusmenu-gtk3 and libdbusmenu-gtk3 does not seem to use GtkImageMenuItem so it could work.

I just added an implementation of the GetGroupProperties method to the sni branch. Opposed to at least mate-panel and xfce4-statusnotifier-plugin, snixembed heavily relies on it instead of just the GetLayout method. The current implementation works for me in a quick snixembed test.

souravdas142, 2021-05-26:
> I just added an implementation of the GetGroupProperties method to the sni branch. Opposed to at least mate-panel and xfce4-statusnotifier-plugin, snixembed heavily relies on it instead of just the GetLayout method. The current implementation works for me in a quick snixembed test.    

Yes now witth `sni` branch menu items icon are showing and its not effected with `gtk-menu-images=0|1` like before. But there are still some problems, I will come one by one from critical to lowest order.
1. https://github.com/blueman-project/blueman/issues/1537#issuecomment-846408337 the same problem comes again in `sni` branch. May be you forget again to rebase on `master`.
2. There are some new errors. see [blueman-applet_sni-error2.log](https://pastebin.com/vQ8KnZvF), When I was tring to open `Devices` from menu, this error came.
3. Are they all harmless warnings/error [blueman-tray_sni-error.log](https://pastebin.com/HcgvQ1U5) ? 
 Specially ```/org/blueman/sni is not a service name, falling back to appindicator on :1.381```
```:1.381 set menu: /org/blueman/sni/menu```?
4. blumean-tray menu overlaps other tray icons, Details are below.
   >My tray size height is 16px from the top of the screen, and also tray-icon-size=16px

   If I right click on y=15 (px) [on xorg-cordinate system, 'x' is not important in this case] on blueman-tray icon, the tray menu 
   starts drawing from y=15 and if right clicked on y=1, menu starts drawing from y=1, Means blueman-applet menu overlapping 
   Other tray icons, This behaviour also have in master. for visualisation giving screenshots.

    Y=1 &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;  Y=15
    [image]  [image]

    We should respect tray-size-height. I want If i click on the blueman-tray icon on whatever 'Y' location, menu should draw always start  after `tray-size-height` means here from 16px ?

5. there are too much padding left of the menu-items icon If I start with `snixembed --fork` for both `master` and `sni`. For visualisation see below screenshots.

   Without `snixebeded` &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;   With `snixembed`
   [image]  [image] 

At last I am giving [blueman-applet_sni.log](https://pastebin.com/gQvvgmpP) for further debugging purpose.

---

> How is master with snixembed? As far as I can see, snixembed just relies on libdbusmenu-gtk3 and libdbusmenu-gtk3 does not seem to use GtkImageMenuItem so it could work.

with `snixembed` still effecting with `gtk-menu-images=0|1` as usual.

cschramm, 2021-05-26:
Nice. Thanks for all the feedback!

1. :facepalm: Done.

2. Not sure where that's coming from. Maybe fixed with the rebase?

3. "set menu" is just an info message. The service name thing is interesting. snixembed primarily expects a service name to be passed when registering a StatusNotifierItem and then expects the object path to be `/StatusNotifierItem`. I guess that was the initial KDE implementation. Funny enough passing a service name is specified in https://www.freedesktop.org/wiki/Specifications/StatusNotifierItem/StatusNotifierWatcher/ but the spec nowhere says anything about the object path. When snixembed detects that the passed string is not a service name, it uses it as object path and just uses the sender as the service name. That matches the appindicator behavior. I guess we should keep it as it is probably more compatible.

4. That seems to be specific to snixembed or polybar (probably more polybar). I don't see it with snixembed or any other way in mate-panel. I always get the menu below the area.

5. That is snixembed behavior. It also adds some weird graphical effects to the menu. I guess it tries to imitate the look of a very specific desktop or something. :shrug:

> with snixembed still effecting with gtk-menu-images=0|1 as usual.

Nice to know. So that's another bonus we get with dropping libappindicator. :slightly_smiling_face:

souravdas142, 2021-05-26:
Thank you  For the reply.

As `1.` is fixied `2.` is also fixed.
3. Okk.
5. Sorry for that. I also noticed some other menus have the same issue with `snixembed`, So it's not the issue of blueman.       

---
4.. No man, I have tested with `i3status`, `stalonetray`, `tint2` `mate-panel` all have the same issue mentioned in https://github.com/blueman-project/blueman/issues/1537#issuecomment-848465311

With `master` this isuue is **only** reproducible with `snixembed`.  (which is my previous claim was partially wrong, i.e natively both branch have the same issue) And is reproducible  with `sni` branch with or without `snixembed`. So its not an issue with `polybar` or [ `snixebed` ? ].

cschramm, 2021-05-26:
With StatusNotifierItems, there is nothing we could do about that as we have no influence on how the host renders and positions things. We might have an influence on it in our GtkStatusIcon implementation, I'm not sure.

`mate-panel` positions all my menus, no matter if native XEmbed, StatusNotifierItem or snixembed, below the icons. There is only a single exception (seemingly a native XEmbed implementation) that sometimes seems to trick mate-panel so that it overlaps the icons. I guess mate-panel manipulates the click events or something before forwarding them to the implementations. If the panel does not do that, then only the renderer could take care of it and in case of the discussed constellation that's snixembed, not blueman-tray.

souravdas142, 2021-05-27:
It's okey then. I could not find any more bugs nor have any feature suggestions more than that for sni. Thank you so much. You may close the issue.

closed 2021-05-27 by cschramm