blueman-project/blueman #1538: Connection faied: input/output error

LTWilliams58, 2021-05-20:
blueman: 2.1.2-1ubuntu-0.2
BlueZ: 5.53-0ubuntu3.1
Distribution: xubuntu 20.04
Desktop environment: xfce

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2021-05-20:
That's an error coming from BlueZ. Check the bluetoothd logs for context.

LTWilliams58, 2021-05-20:
The bluetooth entries from my syslog are already included, unless bluetoothd uses a different file.

As I mentioned, if there is some other log I need to include, please let me know.

cschramm, 2021-05-20:
Included? 🤔

LTWilliams58, 2021-05-20:
There are three pastebin links in my original report, the third one is a syslog grep. I will put those here as well:
blueman-applet --loglevel debug : https://pastebin.com/1ZNBVLWk
bluetoothctl : https://pastebin.com/7jV4RaEP
syslog | grep blue : https://pastebin.com/9qVyfBXg

cschramm, 2021-05-22:
You added that information to a comment that is not displayed. :wink: I just edited your initial message so that it shows up.

Looks like the remote device disconnects immediately so that the pairing fails. My guess would be that it is paired with some adapter already and needs a reset or something.

LTWilliams58, 2021-05-22:
Thank you for reformatting my initial message.

This same issue happens on a different PC with a different mouse, same bluetooth package versions, and started at about the same time.

cschramm, 2021-05-22:
From the bluetoothd logs I can only tell that the remote device disconnects (I'm surely not the best source for interpreting BlueZ stuff, though). You could gather more insights with `btmon`.

LTWilliams58, 2021-05-23:
Thanks for the advice. Output from btmon is in https://pastebin.com/aTY5Lx3A

Note this is btmon output from a different PC, but all package and OS versions are identical.

cschramm, 2021-05-23:
At 50.023959 BlueZ starts a connection attempt to the M557. At 55.335685 it gives up with a timeout. Did you actually get the `Input/output error` there? :thinking:

LTWilliams58, 2021-05-23:
So I'm not sure what changed, if anything. On the original PC, the BT mouse connects and remains connected. 
On my other PC, the same mouse fails to pair/connect, I see a timeout error. I collected fresh logs, all reporting the same timeframe.

sudo tail -f /var/log/syslog: https://pastebin.com/ziY4zpHP
sudo btmon: https://pastebin.com/9fiiJYeH
sudo bluetoothctl: https://pastebin.com/tX39HJiv
sudo blueman-applet: https://pastebin.com/djunQgds

Thanks for looking at all this. As always, if I can provide more info, please let me know.

cschramm, 2021-06-04:
Not sure what to look at. I'm surprised to not see a timeout in the btmon output.

From a quick Google search, I suspect this boils down to a firmware issue.

closed 2021-08-11 by github-actions[bot]