blueman-project/blueman #1541: Blueman-manager can't be called from context menu of blueman-applet

torwag, 2021-05-27:
blueman: 2.1.r442.gb1fb0459-1 (AUR Arch)
BlueZ: 5.58
Distribution: Arch Linux
Desktop environment: Sway

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

***What I expect:***
Using the context menu from blueman-applet to start blueman-manager
***What I get:***
Trying to start the blueman-manager from the context menu of the blueman-applet, I receive an error message from blueman-applet (tested by manually starting blueman-applet on a console). No window is created, the error is silent, if not catched in the logs manually.
***What else was tested:***
I can call blueman-manager (taken that blueman-applet is running) from the console without any issue.
Downgraded to 2.1.4 (community Arch package) I can call blueman-manager from within blueman-applet but context menus of blueman-manager does not work (maybe a wayland / sway problem)

***Error-message:***

```
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/blueman/main/DBusProxies.py", line 52, in call_finish
    proxy.call_finish(resp)
gi.repository.GLib.Error: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying (4)
```
to give a bit more context to it, here are the debug logs of the start-up of blueman-applet:
https://pastebin.com/ThEYhhS7

cschramm, 2021-05-30:
I'm not sure if I got the problem right. You get that exception in blueman-applet, right? I don't see it in the blueman-applet log output you provided, though. :thinking:

As a quick first check, please try if
```
python3 -c 'from blueman.main.DBusProxies import ManagerService; ManagerService().startstop()'
```
shows the same (exception + no window showing).

torwag, 2021-05-31:
Hi,
Yes that exception appears in the log, if I start blueman-applet manually from a command line (without debug option). To be more precious after opening the context menu with a right mouse click and select "Devices...". Other context menus seem to work. Executing your code, nothing happens, and no error log appears (in the same applet log, where I can reproduce the error message from within the context-menu).

cschramm, 2021-06-04:
> that exception appears in the log

Why isn't it in the pastebin content?

I'm rather clueless. Please get d-feet running and click on the org.blueman.Manager session bus service. What happens? If you're able to get there, navigate to `/org/blueman/Manager`, interface `org.freedesktop.Application`, method `ActivateAction` and try to execute it with input `"Activate", [], {}`.

torwag, 2021-06-07:
It might not be in the pastebin content, as I simply might have paste just the start but not the error, as additional information in case something went already wrong during startup.

cschramm, 2021-06-08:
Anything showing up in journalctl when you try that?

torwag, 2021-06-08:
Yes, if I use the applet and click on 'Devices...' journalctl shows:

```
Jun 08 22:12:15 thinklab5 dbus-daemon[3842]: [session uid=1000 pid=3830] Activating service na>
Jun 08 22:12:15 thinklab5 org.blueman.Manager[115784]: Unable to init server: Could not connec>
Jun 08 22:12:15 thinklab5 org.blueman.Manager[115784]: Unable to init server: Could not connec>
Jun 08 22:12:15 thinklab5 dbus-daemon[3842]: [session uid=1000 pid=3830] Successfully activate>
Jun 08 22:12:15 thinklab5 org.blueman.Manager[115784]: Unable to init server: Could not connec>
Jun 08 22:12:15 thinklab5 blueman-manager[115784]: cannot open display: 
```

Maybe it is repetitive as I clicked several times.

cschramm, 2021-06-08:
Looks like the `DISPLAY` variable is missing in your systemd environment. See #1086.

torwag, 2021-07-05:
Hi, so indeed I had sway slightly misconfigured and did not expose DISPLAY correctly to the system. After fixing this, it starts as intended. This issue can be closed and this post shall  remain for all eternity to remind me on my stupidity. 
If you face the same issue, please check that `echo $DISPLAY` points to the correct value resp. any value at all.

closed 2021-07-05 by torwag

goodvin, 2021-07-20:
I have the same issue, but the DISPLAY is correct.

goodvin, 2021-07-20:
I don't see org.blueman.Manager in d-feet.

goodvin, 2021-07-20:
I get the same exception when I click "/org/blueman/Manager" in d-feet.

cschramm, 2021-07-20:
@goodvin: If blueman-manager gives you `cannot open display: `, no `DISPLAY` variable is provided to it or it's empty.

goodvin, 2021-07-20:
blueman-manager doesn't give me "cannot open display:" & I can run it without any issues using a console.

I get an exception during activation of the dbus service via tray menu:
**gi.repository.GLib.Error: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying (4)**

goodvin, 2021-07-20:
Looks like everything works fine if I run `dbus-update-activation-environment DISPLAY` upfront.
Thanks for pointing me out to the env variables.