blueman-project/blueman #1543: [Question] Is there a keystroke to close blueman-manager?

WillieStevenson, 2021-05-30:
blueman:  blueman-manager 2.0.8
BlueZ: n/a
Distribution: Debian 10
Desktop environment: sway

Just a simple question on how to close the blueman-manger UI window.
Other programs (firefox, folder windows, etc) are often able to be closed with Ctrl+Q or Ctrl+W.
Does blueman-manager have such a keystroke? 

Thanks.

cschramm, 2021-05-30:
master and thus the upcoming 2.2 have Ctrl+Q and Ctrl+W accelerators. Usually window managers also have combinations to close an active window, typically Alt+F4, a quick Google search looks like Sway does not, though, at least not that one.

WillieStevenson, 2021-05-31:
I think that's the trade-off with sway (but well worth it due to it's other functionalities) - it seems each program has their own keystrokes for function's such as closing.
Thanks for the update, and I'll be awaiting the release.

cschramm, 2021-05-31:
Seems like it has Mod+Shift+Q.

closed 2021-05-31 by cschramm