blueman-project/blueman #1545: Missing NetworkManager is no longer handled gracefully

mgorny, 2021-06-01:
blueman: 2.2
BlueZ: 5.58
Distribution: Gentoo Linux
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

After upgrade to 2.2, the blueman service doesn't seem to be functional. When starting blueman-applet, I get the following traceback:  [blueman.txt](https://github.com/blueman-project/blueman/files/6574389/blueman.txt). Other tools just hang at start without displaying the window.

It seems that something has changed regarding loading NetworkManager. The previous versions seemed to handle it being missing gracefully while the new one seems to hit some exception loop.

cschramm, 2021-06-01:
Ouch. Can you please check #1546?

cschramm, 2021-06-01:
Released a 2.2.1. Thanks for the quick report.

closed 2021-06-01 by cschramm