blueman-project/blueman #1552: UI Bug

Sov-trotter, 2021-06-02:
blueman:2.1.4-1
Distribution:Manjaro Linux
Desktop environment: xfce

Apologies for the picture from phone.
[image]

cschramm, 2021-06-03:
There's a native StatusNotifierItem implementation in #1382 that replaces the AppIndicator implementation and should not have that issue.

saija, 2021-06-05:
Hi guys

Same UI issue here:

Distributor ID: Ubuntu
Description:    Ubuntu 20.10
Release:        20.10
Codename:       groovy
[image]

Desktop environment: KDE Plasma

cschramm, 2021-06-05:
Same solution. As far as I remember the issue is due to libappindicator and independent of the tray. It seems to be fine with our own StatusNotifierItem implementation in #1382 (some trays render the part bold, some just strip the tags).

closed 2021-06-05 by cschramm