blueman-project/blueman #1555: Bluetooth doesn't turn on automatically after suspend/resume

hamkg, 2021-06-13:
blueman: 2.2.1
BlueZ: 5.58
Distribution: Slackware64 -current
Desktop environment: MATE 1.24

After upgrading to blueman-2.2.1, bluetooth is turned off when suspending and remains so after resume. No big deal since I can easily turn it on with the blueman applet, but I would prefer it turning on automatically.

A bug, or intentional change? Something I can change in /etc/bluetooth/main.conf?

See lowlevel debug output got after starting blueman-applet, then going into suspend, then resume: https://pastebin.com/Jn9ZAssT

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2021-06-13:
Are you saying this behavior differs between 2.1.4 and 2.2.1 (without any other changes on them system!)? That's should not be intentional, no.

Am I right that there's actually no output when resuming?

hamkg, 2021-06-13:
Yes, I tried reverting to 2.1.4 and after suspend/resume Bluetooth was active at once. Upgraded once again to 2.2.1 and BT stays off after resume (red X on the systray icon, menu items greyed out). After clicking "Turn Bluetooth On" everything works as expected.

cschramm, 2021-06-14:
I guess the behavior highly depends on the system. For me, I always see the Powered property of the adapter changed to False on suspend and back to True on resume. In your log, it only gets changed to False but not changed back to True. Blueman can auto-power the adapter for you if you enable that option in the settings of the PowerManager plugin.

I do not get why 2.1 and 2.2 should show different behaviors here, though, as when the option is not enabled, they do not interfere with the power states at all. :thinking: Maybe the icon in 2.1 just tells you lies and it's actually not powered. You can check the state e.g. with `bluetoothctl show`.

hamkg, 2021-06-14:
The icon in 2.1.4 is truthful: no trouble at all getting a BT connection after suspend/resume.
I can see the same difference as you noticed when comparing logs from 2.1.4 and 2.2.1, but that doesn't help me understand why this is happening.
Maybe hardware-related? 
I've put a question in the Slackware section on LQ and noone has mentioned the same experience.
Also tried rebuilding 2.2.1 using the official slackware buildscript - no difference.

cschramm, 2021-06-14:
Very surprising. The changes logged by blueman-applet are just what it gets from BlueZ. If the blueman version is the only difference, then I would not expect that. You could check independent from blueman e.g. with `dbus-monitor --system 'path=/org/bluez/hci0,member=PropertiesChanged'` if you see the same difference. I'd try with 2.1.4 and 2.2.1 installed both with and without blueman-applet running.

Am I right that the PowerManager option is disabled for everything you described so far?

hamkg, 2021-06-14:
dbusmonitor shows a telling difference, see the pastebins (links below). For blueman-2.2.1 dbusmonitor stops at line 50 after suspend/resume. For blueman-2.1.4, dbusmointor stops at line 70 after resuming. For blueman-2.2.1, dbusmonitor stops at line 50. By clicking "Turn Bluetooth On" in the applet, the 20 crucial lines are added.

The "Auto power-on" option in PowerManager is unchecked.

blueman-2.1.4 dbusmonitor: https://pastebin.com/wMjpw33X
blueman-2.2.1 dbusmonitor: https://pastebin.com/8UKZm5B5

cschramm, 2021-06-14:
Double checking, as I'm not familiar with Slackware package management: The blueman version is definitely the only thing that changes? No hidden additional change to that?

If that's the case then it seems like blueman 2.1.4 takes care about powering the adapter even though the PowerManager option is disabled. I have no clue where that would come from, but if it's the case you should see it on the bus as well, i.e. e.g. `sudo dbus-monitor --system 'path=/org/bluez/hci0,member=Set'` should show a call (from blueman-applet but you'd need to know its bus name to identify it) that sets the `Powered` property to True.

In any case, the behavior of 2.1.4 is unexpected to me. 2.2.1 works as expected. To get the adapter powered automatically `/etc/bluetooth/main.conf` has the `AutoEnable` option for you and blueman-applet the one in the `PowerManager` plugin. Both should work to auto-power the adapter on resume.

hamkg, 2021-06-14:
So you mean that it's the expected, default behaviour that Bluetooth remains turned off after suspend/resume? I had no idea.
Checking the auto power-on option in the PowerManager plugin didn't change anything, though. Setting `AutoEnable=true` in `/etc/bluetooth/main.conf` did the trick though. The red X on the icon is visible for a second after resuming (I never saw it earlier) and then Bluetooth is active again. I'm happy with that solution, so I guess we can call the issue solved. Thanks for your help!

The slackware package does not contain anything apart from what's in the blueman-2.2.1.tar.xz file and two patches: blueman.allow.access.to.netdev.group.diff.gz. and dont-autostart-applet-in-kde.diff.gz. No static libraries or other stuff added.

cschramm, 2021-06-15:
> So you mean that it's the expected, default behaviour that Bluetooth remains turned off after suspend/resume?

Basically yes, that's the observed behavior of BlueZ with `AutoEnable` switched off. It might depend on Firmware or something.

I can confirm that our `PowerManager` option does not effect resume, so it's not an alternative to BlueZ' `AutoEnable` there. Not sure if it would be worth changing that.

areographe, 2021-07-25:
I've noticed this same issue on Manjaro Linux where the default behaviour for the last two years or so (going back to at least blueman 2.1.1) has been for the bluetooth on/off state to persist across suspend/resume. This means that if bluetooth was turned on, suspend and resume would result in bluetooth remaining on after resume is finished. However, a recent update from 2.1.4 to 2.2.1 means that bluetooth is always turned off on suspend/resume. There was no concurrent upgrade of bluez or bluez-libs packages as far as I'm aware, and I made no changes to the configuration of blueman or its plugins. The PowerManager plugin was configured to 'Auto power-on' for that two year period, and remains so currently, but it appears that the update to blueman 2.2.1 has changed the visible behaviour.

keylogthis, 2021-08-06:
I have the same thing. Previous version was working as expected but after updating to 2.2.1-1 my mouse wouldn't reconnect when resuming from sleep. The PowerManager was set to "Auto power on," but made no difference, changing /etc/bluetooth/main.conf and adding AutoEnable has now fixed the problem.

blueman: 2.2.1
BlueZ: 5.60-1
Distribution: Manjaro
Desktop environment: XFCE

cschramm, 2021-08-23:
@areographe @keylogthis: You're saying that blueman 2.1.4 did auto-power on resume for you? That's unexpected. :thinking:

JonnyHaystack, 2021-08-24:
I had this issue, but it seems like it's gone now. I'm on the 2.2.2-1 package in Arch. Not sure whether it was an update that fixed it or not though because I don't pay super close attention to every package that gets updated.

areographe, 2021-08-24:
also updated to 2.2.2-1 package in manjaro (identical to the package in arch i believe) and the issue i had is now gone. the behaviour for suspend/resume (i.e. bluetooth adaptor power on/off persists when going to sleep and waking up the laptop) is all correct, so i've got no issues now.

closed 2021-08-28 by cschramm

kewbish, 2021-09-14:
I had the same problem with the Bluetooth controller powering off after suspend after some updates. Strangely enough, 2.2.2-1 was the version that *didn't* work for me - I had to downgrade to 2.1.4. While I was on 2.2.2-1, I had PowerManager Auto Power-on enabled + I tried setting AutoEnable to true in `/etc/bluetooth/main.conf` and disabling USB autosuspend in TLP configuration settings, but none of those workarounds seemed to work. Only downgrading seemed to work.

nickoe, 2021-10-12:
I see the very same on archlinux as well. I guess this is a blueman thing, but I am not really sure. I recently updated as well and see blueman in there:
```
[2021-10-08T16:07:35+0200] [ALPM] upgraded bluez (5.59-2 -> 5.61-1)      
[2021-10-08T16:07:35+0200] [ALPM] upgraded bluez-libs (5.59-2 -> 5.61-1) 
[2021-10-08T16:07:36+0200] [ALPM] upgraded blueman (2.1.4-1 -> 2.2.2-1)
[2021-10-08T16:07:36+0200] [ALPM] upgraded bluez-utils (5.59-2 -> 5.61-1)
```

So I am on a x280 laptop and it used to work fairly well, but now bluetooth is off after I resume from suspend.

I not that there are multiple different observations in the later comments though... but it is certainly an inconvenient regression somewhere.

cyberpunkrocker-zero, 2021-12-11:
This regression still exist in blueman-2.2.3 on Arch Linux. I recently installed Arch on my ageing Samsung laptop, and spent several days trying to figure out why bluetooth stayed powered down after resuming from suspend. That is especially annoying in my case because I use a BT mouse, and touchpad is disabled while the mouse is in use.

The bluetooth adapter is a "Qualcomm Atheros Communications AR3012 Bluetooth 4.0", internally attached to USB bus.

As is the case with other people here, downgrading blueman to version 2.1.4 finally fixed the issue. Now bluetooth comes automatically up after resume like it is supposed to do.

Why is this thread marked "Closed", btw? This issue is far from being fixed!

cschramm, 2021-12-12:
Before the issue got closed, multiple users reported the desired but unexpected behavior to be back with blueman 2.2.2.

In any case, PowerManager's auto-power feature...
a) ...basically does not cover suspend and resume. I have no clue how blueman 2.1.4 affects resume and failed to replicate that but it might just depend on how suspend and resume happen on the specific system.
b) ...is superseded by BlueZ' `AutoEnable`. We'll have to think about dropping auto-power and possibly provide a frontend to the `AutoEnable` setting instead. A frontend is not worth too much, though, as most distros enable that by default anyway. It's more or less Arch Linux specific that the user has to set it as Arch Linux does not tamper with the upstream defaults.

infirit, 2021-12-12:
Blueman only automatically powers an adapter when it gets a new adapter added signal on dbus. If we had a debug applet log when this worked we can see all the events that lead to it being powered.

iant321, 2022-08-07:
I am on Fedora 36, BlueZ 5.65, blueman 2.3.1, kernel 5.18.13. The problem is still present - after resuming from sleep, bluetooth is off. The blueman tray icon is in place, showing the red cross. I can easily turn bluetooth on by clicking the icon, but this is still very annoying. AutoEnable=true in /etc/bluetooth/main.cfg. The AutoSwitch check box in the PowerManager plugin does not affect the issue. Unfortunately, I cannot downgrade to blueman 2.1.4 and BlueZ 5.58 without messing with many other packages in Fedora 36 and so cannot verify if this older version works. Any advice?

Go-rom, 2024-12-02:
Hello
i'm encountering the same issue with Blueman 2.3.5.
i'm confused because i see that PowerManager plugin was removed 2 years ago but i'm still seeing.
@cschramm Could you please tell me what's the right thing to do please ?

Note that if i try to disable PowerManager, it raises a dependency issue saying that it will also disable "KillSwitch".

infirit, 2024-12-03:
> Hello i'm encountering the same issue with Blueman 2.3.5. i'm confused because i see that PowerManager plugin was removed 2 years ago but i'm still seeing it. @cschramm Could you please tell me what's the right thing to do please ?

The PowerManager plugin wasn't removed, the auto powering of the adapter feature was.

You'll have to configure BlueZ to set the default power state of adapters, https://github.com/blueman-project/blueman/wiki/Troubleshooting#officially-supported-way

Go-rom, 2024-12-04:
Thank you @infirit for your answer
`AutoEnable` is already set to `True`

infirit, 2024-12-04:
> Thank you @infirit for your answer
> `AutoEnable` is already set to `True`

If it doesn't power on resume it may be a driver problem. Regardless, this is out of scope for blueman as a frontend and probably needs fixes in BlueZ and/or the kernel.

The only reason blueman was automatically  powering adapters was because it wasn't done by BlueZ. Since BlueZ changed the default to auto power adapters blueman dropped this feature.

Go-rom, 2024-12-05:
Thanks for the clarification.
I'll look into then a driver issue.

Denisgsv, 2025-04-30:
did you solved it ? 

> Thanks for the clarification. I'll look into a driver issue then.

Go-rom, 2025-05-06:
Hello @Denisgsv 
i didn't solve it technically speaking but i'm with a clean install of Linux Mint 22.1 now and i didn't encounter this issue for now.