blueman-project/blueman #1556: Too many notifications from Blueman

ShrirajHegde, 2021-06-15:
blueman:2.2.1
BlueZ:5.55
Distribution: Fedora 34
Desktop environment: Gnome 40

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

In the latest release, Blueman sends a notification for every connect/disconnect event, any way to disable that? it gets annoying after frequent connect and disconnect

ShrirajHegde, 2021-06-15:
A way to disable this would be nice, started in the latest update

[image]

cschramm, 2021-06-17:
You can disable the `ConnectionNotifier` plugin.

closed 2021-06-17 by cschramm

pjones, 2021-07-19:
@cschramm 

Disabling plugins doesn't seem to be persistent on my system.  In other words, after a reboot the `ConnectionNotifier` plugin is re-enabled.

I don't see anything in the gsettings schema file to control this via dconf.  When blueman starts how does it know which plugins to enable?  Is there some value I can feed to dconf to disable a plugin?

Thank you.

cschramm, 2021-07-19:
@pjones: Plugins are controlled by `/org/blueman/general/plugin-list`. When you uncheck the plugin, `!ConnectionNotifier` should immediately show up there and should still be unchecked after restarting blueman-applet.

pjones, 2021-07-19:
@cschramm Thank you very much.

I'm not running a desktop environment and I'm not running a dconf service so as far as I can tell any changes that the GUI is making just get thrown away.

What I can do is use `dconf load` to import an INI file with the correct settings before blueman starts.  I got it working with:

```ini
[org/blueman/general]
plugin-list=@as ['!ConnectionNotifier']
```

For any home-manager users that find this, here is the change to my config that I needed:

```nix
{
  services.blueman-applet.enable = true;

  dconf.settings."org/blueman/general" = {
    plugin-list = [ "!ConnectionNotifier" ];
  };
}
```

cschramm, 2021-07-19:
@pjones: Well, sure, if dconf changes do not get persisted, settings are not persistent. :laughing: Glad I could help. :+1:

vanitasvitae, 2022-05-17:
Hey! Thank you for your work on blueman!
A small nit: For blueman notifications in Gnome Shell it would be nice if stale notifications would be withdrawn.

See https://developer.gnome.org/documentation/tutorials/notifications.html#actions-with-parameters

(I have no oversight over the ecosystem so I don't know if this is easily doable for you, so don't feel pressured ;))

cschramm, 2022-05-19:
If I get it right, a stale notification in this context means that there is a newer connect / disconnect event for the same device. We can actually solve that with replacing (assuming that replacing leads to a new notification if the old one isn't shown anymore). The notification protocol also supports closing notifications but I don't see at which point we should consider them stale here.

ba32107, 2022-08-25:
Hi,

I still have this issue. I have tried disabling the `ConnectionNotifier` plugin, and it works until I restart my computer. Upon restart, I again see the notifications. When I open blueman, I go into plugins, I see that the `ConnectionNotifier` is disabled, but the app also crashes. After that happens, the notifications are not shown as expected.

So, it's as if I need to open blueman and the Plugins dialog for blueman to actually notice that this plugin should be disabled.

Anything I can try to troubleshoot?

cschramm, 2022-08-26:
@ba32107: Please open a new issue.

If I get it right, the disabled plugin seems to be working anyway. Terminate blueman-applet, run it with `blueman-applet --loglevel debug` from a terminal and provide the initial output.

Then you're saying that blueman-applet crashes shortly after opening the plugins dialog. Start it with the same arguments from the terminal (or continue with the previous session) and open the dialog to collect the crash output.

ba32107, 2022-08-28:
Thanks, raised https://github.com/blueman-project/blueman/issues/1838.

infirit, 2022-10-11:
I don't know exactly how it works. I'd test it but, that requires me to
actually use gnome, and that requires some self "motivating".