blueman-project/blueman #1557: Can buleman automatically respond to requests from serial data devices

xiaoye-cn, 2021-06-16:
blueman:2.0.8-1+deb10u1
BlueZ:5.50-1.2~deb10u1+rpt2
Distribution:
Desktop environment:Raspberry Pi OS with desktop (buster-armhf)

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2021-06-17:
blueman 2.2 brings an auto-connect feature that you can switch on for the serial service. blueman 2.0 cannot, no.

There is no official and stable interface to blueman but blueman-applet does expose a D-Bus method to connect services that you could use. You need the BlueZ D-Bus object path of the device, something like `/org/bluez/hci0/dev_00_00_00_00_00_00` (basically just built from the hci and the address).

dbus-send for blueman 2.2: `dbus-send --session --print-reply --dest=org.blueman.Applet /org/blueman/Applet org.blueman.Applet.ConnectService objpath:/org/bluez/hci0/dev_00_00_00_00_00_00 string:00001101-0000-1000-8000-00805f9b34fb`

I think in 2.1 the path should be `/org.blueman/applet` instead of `/org/blueman/Applet` and in 2.0 it's `/`. The method might be called `connect_service` in 2.0. It's best to just have a look e.g. with d-feet.

xiaoye-cn, 2021-06-20:
@cschramm 
Thank you very much for your reply!
Please forgive me, my level is not high, I still can't fully understand what you mean.
However, I found an easy way.
I wrote a line of command in the terminal: sudo rfcomm listen all
After this operation, the Raspberry Pi can respond to the Bluetooth request from the Lego EV3 and establish a virtual serial port. /dev/rfcomm0.
I use rfcomm0 directly in another python program. Complete the communication.
Thank you very much for your reply.
blueman is great!

closed 2021-06-20 by cschramm