blueman-project/blueman #1558: UI improvements suggestions

Jakobovski, 2021-06-26:
Here are a few UI suggestions which will greatly improve the usability of the app.

1) Add a connect button next to each item in devices so the user does not need to right click and open a context menu. This reduces 1 click and is simpler

2) Replace "search" button with "refresh" or "Search/Refresh". Its not clear that search is searching for devices and not searching within the already existing devices. 

3) Show the device list in the drop down menu on the toolbar. Why make the users open another window when they want to connect? Its really annoying

4) Don't show "Bluetooth devices ready" popup. Only show an error if they are not ready.

Thanks

cschramm, 2021-06-28:
1. Are you aware of the double-click?
2. Refresh clearly sounds too lightweight to me considering what it does. Scan could be an option.
3. The list could get rather big as it contains all known devices. I'm not sure if that would be a good idea.
4. What does that refer to?

fdmarcin, 2021-06-28:
> Are you aware of the double-click?

@cschramm Hi! I came here to suggest adding this. Do you mean this should already be there?
I'm on 2.1.4 and double-clicking a device in the list does nothing for me.

I think it would be a good UX improvement to this already great program to make it easier to connect to a device :)

cschramm, 2021-06-28:
2.2 supports double-click for connecting or disconnecting a device in blueman-manager, yes.

fdmarcin, 2021-06-28:
> 2.2 supports double-click for connecting or disconnecting a device in blueman-manager, yes.

Nice, thanks!

domenkozar, 2021-06-28:
I'd suggest inlining "Recent connections" in the tray menu if it has less than 3 entries, saves one click.

That's probably the most used action (for me?).

Jakobovski, 2021-06-29:
1) Double click does not work for me. I am using version 2.1.2
2) How about not having a search button at all. Just make it search automatically when the window is open and have a little spinner to indicate it is searching. This will save the user a click
3) Yes just show top 5-8  recent connections. That will do 99% of the work. Then you can get rid of the recent connections submenu of the dropdown menu
4) i think it might be a ubuntu dialog and not related to blueman: see screenshot: https://photos.app.goo.gl/5xQtjCXJSbywuiiS6

One more issues:

5) When i click on "devices" in the drop down it opens the window but it is not placed in front of my current windows

Thanks for the help and responsiveness!

cschramm, 2021-06-29:
1. Makes sense as 2.2 > 2.1.2.
2. Starting a discovery when blueman-manager gets opened could make sense (depending on which use cases one considers) but is a different story than changing the wording as the discovery will stop at some point and it does not make sense to me just immediately start another one.
3. Recent connections is another story as well. Makes sense to me. At least NetworkManager also just adds all available networks to its tray menu and that will typically be more than our recent connections.
4. and 5 are connected and a so called feature of gnome-shell, possibly combined with older bugs in gnome-shell and / or mutter that cause unexpected behavior in our case.

Jakobovski, 2021-06-30:
1) Ok thanks
2) Why not just keep it constantly searching? I think that is what MacOS and Ubuntu default UI use
3) Ok
4/5) Is there a workaround? 

Thanks!

cschramm, 2021-06-30:
2: Devices will keep popping up in the list, disturbing the user, it uses battery, and it might make the adapter visible (not quite sure to what extend). No real showstoppers, but if the user opens blueman-manager for something else than discovering new devices - and I'd say that's the case most of the times - they easily outweigh the then nonexistent benefits. I'm undecided. 🤷

4/5: You did not provide the minimal information asked for in our issue template, so I cannot give any specific hints, but in general what the Internets say is that there's a hidden `auto-raise` setting in `org.gnome.desktop.wm.preferences` as well as an extension called _Steal my focus_. Also there's a lot of stuff on the focus stealing prevention "feature" on gitlab.gnome.org, e.g. [gnome-shell#358](https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/358) or [mutter#505](https://gitlab.gnome.org/GNOME/mutter/-/issues/505). As you mentioned you're using blueman 2.1.2: I guess that Gnome behavior is not triggered by 2.2 due to #1304 (and probably even due to #1272 if that were not in place).

Jakobovski, 2021-07-01:
2) Ok. How about you make it a checkbox that is saved between sessions. So if the user `checks` to always search then it will do that everytime the window is open. I wonder what most people use blueman  for? I use it for connecting to my mouse and speakers.

cschramm, 2021-07-01:
If your use case is to connect to the same mouse and speakers again and again no discovery should be involved at all. As long as you do not remove them, BlueZ remembers both devices and as they are already paired (or do not even require it in the first place) you just have to connect. You can use the recent connections from the tray menu or open blueman-manager and double-click (2.2) or right-click and connect on the device. Why would you use the search button?

Jakobovski, 2021-07-03:
1) There seems to be a bug in recent connections as only one of my recent connections appears there
2) I have a few bluetooth mice. One at the office, one at home and each one can connect to 3 computers. (logitech mx master) Sometimes i use four computers (its a long story) and then the connections get reset and i need to find it again.

cschramm, 2021-07-05:
> There seems to be a bug in recent connections as only one of my recent connections appears there.

Please open a ticket for what does not seem to work as intended / expected.

> Also it only shows "auto connect profiles". What does that mean? I just want a simple "Connect button".

Recent connections are "per service". Services that blueman supports are auto-connect profiles (HID and audio), network services and serial services. It's a design choice of BlueZ 5 to connect to HID and audio profiles via a generic connect action on the device. I'd love to tell you that the connection is about input or audio profiles but we just do not know what it will find and connect to.

> Also how do i know that that device is even on and detected? I think it should work like the wifi menus. It shows you which of your recently paired items are available .

You don't. Bluetooth is not WiFi. If you want to connect to a known Bluetooth device, you just try and see if it responds.

>  each one can connect to 3 computers. (logitech mx master) Sometimes i use four computers

Using a device with more computers than it can remember is a case where you constantly need to re-pair, yes. I think you even have to remove the device from BlueZ to re-discover it in unpaired state for that. Starting a discovery automatically does not seem like a huge win there, as you still have the remove and pair operations, don't you? You might even fail to rediscover the device if you start the discovery before removing it.

InstanceOfMichael, 2021-12-13:
Suggestion 18: I think this program should show your bluetooth version (expected value 4.2 or 5.0 for my current device). It's not important to me where, it should generally be possible to find this information somewhere.

WhyNotHugo, 2021-12-13:
@InstanceOfMichael Oh, I really like that idea. I want to try streaming to multiple Headphones, and knowing their BT version at a glance would be super useful.

Any ideas if the D-Bus API exposes this?

cschramm, 2021-12-13:
I'd say it's not even part of the protocol and you could only apply heuristics on the observed features. Could you elaborate on the use case?

infirit, 2021-12-14:
It not useful imho as it won't tell you what profiles the device supports.

WhyNotHugo, 2021-12-14:
I sometimes want to use more than one pair of headphones (e.g.: both me and someone else listening to the same thing). AFAIK, this is supported on Bluetooth 5.0 (and usually works more on my phone than on my laptop).

I've had mixed experience using this on my laptop (sometimes sound is very choppy for both devices), and knowing what bluetooth version is being used for each device helps quickly understand if the issue is the just an "old bluetooth version", or something else.

InstanceOfMichael, 2021-12-16:
Ok, I realize my request was too ambiguous. I'd like to know my computer's BT version, but I'd also like to know my connected devices'd BT version, and I'd like to know what connection they were able to successfully negociate, but I don't understand the technical details, but I already know the specs are all optional and every manufacturer uses misleading information in all marketing materials, so maybe if I could see something more like:
```
computer: BT 5
connected device: (headphone, BT 4.2)
```
or 
```
computer: BT 5
connected device: (headphone, BT 5.0, supports low energy mode)
```

No specific use-case, aside from enabling the average non-technical user to have practical and accessible knowledge about their devices.

infirit, 2021-12-16:
That information is just not available over dbus afaics. We might be able to infer it's LE through secondary mean like certain behaviour/properties but I personally am not looking forward to doing or supporting that.

cschramm, 2021-12-19:
I don't think it's even available anywhere as devices don't state a standard version they pretend to be compatible with, they just implement mandatory and optional features specific to some.

What's available in that direction in the D-Bus API is a `LegacyPairing` flag that's always set for pre-2.1 devices. That's visible in our device info dialog. bluetoothd also knows if the device supports BR/EDR and / or LE (4.0 and later) (`SupportedTechnologies` in `/var/lib/bluetooth/*/*/info`) but does not expose that in the D-Bus API. Userspace doesn't know anything else, as far as I know.

cschramm, 2022-02-05:
Closing to stop re-reading this thread again and again. There is #1652 for the recent connections in the applet menu (still needs some work) and I do not see any other actionable topics in this thread.

PR #1652 (2022-09-10): Put recent connections in toplevel applet menu
Other applets like NetworkManager's do not try to keep menu small, so why should blueman...

Closes #1558

This was lying around for a while now. I'm not happy with the presentation yet and don't have any better idea for it but in general this should be good.
commit 46cd2831 (2022-09-10): Put recent connections in toplevel applet menu

Other applets like NetworkManager's do not try to keep menu small, so why should blueman...

Closes #1558
closed 2022-02-05 by cschramm