blueman-project/blueman #1559: pop-up: Shall bluetooth get enabled automatically?

mattrattus, 2021-06-27:
blueman: blueman 2.1.4-1
BlueZ: 5.59-2
Distribution: ArchLinux
Desktop environment: i3

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

My submission is not a problem just a question.
Right after installation, after starting blueman-manager, a pop-up appeared: Shall bluetooth get enabled automatically?
I selected "No". I manually enabled bluetooth.service in systemd and it works. I didn't change anything in /etc/bluetooth/main.conf
My question: which automatic inclusion is affected by the decision in this pop-up?
If I manually select "No" how can I change it to "Yes" and vice versa?

cschramm, 2021-06-28:
The popup is for the auto-connect option of blueman's PowerManager plugin. It's about auto-powering adapters, not enabling a service or anything. There are alternative ways to achieve that, e.g. BlueZ 5.35+'s `AutoEnable` setting, or udev rules that trigger `hciconfig hci0 up`. I would not expect any of that to be in place by default in your case, though, so initially blueman should indicate that Bluetooth is disabled if you chose No.

mattrattus, 2021-07-02:
Ok now I understand. Thx.
Maybe it's just my perception, but his "description" (I mean popup) is unclear.
Hence my question - because I was sure from description, that this Yes/No choice refers to changes of bluetooth settings in the system.

cschramm, 2021-07-02:
I actually thought about dropping blueman's auto-power feature in favor of BlueZ'. All versions of BlueZ that blueman 2.2 formally supports have the AutoEnable setting. An initial popup to enable it if it isn't already probably still makes sense, but it would just be a frontend to the BlueZ configuration (which people typically are not aware of).

In any case I agree that our wording is not ideal. We should just call it auto-powering adapters instead of enabling blueman. We actually do in the PowerManager option, so it's even inconsistent.

mattrattus, 2021-07-02:
I'm glad that my question was legitimate :)

But I still have little misunderstanding of dependencies - this plugin and bluetooth config file.
In the default setup AutoEnable is false and this options responsible for controllers.
But if I change PowerManager plugin to Yes - all connections are made.
How this is works?

cschramm, 2021-07-06:
The PowerManager logic just reacts to new adapters showing up and tries to power them in the BlueZ API. BlueZ' AutoEnable setting basically does the same (on a lower level and a little more, e.g. also auto-powering after suspend).

AutoEnable is not always false in the default setup. At least the bluez packages in Debian and Ubuntu set it to true. We would not need the dialog there at all. Distributions like Arch tend to be closer to upstream. I'm not sure how valuable the dialog is to those users. At least it could just modify or even just mention AutoEnable instead of shipping our own logic.

Regarding dependencies: Blueman 2.2. [formally requires](https://github.com/blueman-project/blueman/blob/2-2-stable/Dependencies.md) BlueZ 5.48 or later, so all supported version do have the AutoEnable setting.

closed 2021-09-12 by github-actions[bot]