blueman-project/blueman #1560: How to replace blueman-assistant in blueman version >=2.2?

heinrich-ulbricht, 2021-06-28:
blueman: 2.2.1
BlueZ: ?
Distribution: Fedora 33
Desktop environment: Xfce

I am using Fedora (on Qubes OS) and need to automate certain things [via script](https://m7i.org/tips/qubes-VM-bluetooth-audio/). The script uses `blueman-assistant` which unfortunately seems to be gone since blueman v2.2 according to the [v2.2 release notes](https://github.com/blueman-project/blueman/releases/tag/2.2):

> Drop blueman-assistant

What would be the equivalent for the following command in the current version of blueman?

`blueman-assistant --device=00:00:00:00:00:00`

Any hints are appreciated.

cschramm, 2021-06-28:
I'm not sure what that command is meant to do. If I'm not mistaken, it just opens either the pairing or the service connection page for the device, depending on its status. So it just triggers UI. As does the pavucontrol command issued next. 🤨

heinrich-ulbricht, 2021-06-28:
Since `pavucontrol` opens the UI I assumed that the `blueman-assistant` command did something important, like pre-selecting the correct device if multiple ones are connected. But your response does not seem to indicate that :D

I don't know either but assumed that functionality so far provided by `blueman-assistant` might have been migrated to one of the other tools - thus the question. If this is not the case and the command makes no immediate sense then the ticket can be closed I guess.

cschramm, 2021-06-28:
It clearly does not seem like it does anything else than opening UI. You could replace it with blueman-manager and just leave the user with the window.

For actual automation, blueman is of no use in that case and BlueZ got you covered. It seems to be about pairing the device and connecting to audio services, so just `bluetoothctl pair $headphones` and `bluetoothctl connect $headphones`.

heinrich-ulbricht, 2021-06-28:
`bluetoothctl connect $headphones` works like a charm. Thank you!

closed 2021-06-28 by heinrich-ulbricht