blueman-project/blueman #1566: Input/Output Error while connecting to BT headset

Pranjalmet, 2021-07-19:
blueman: 2.1.4
BlueZ: 5.60
Distribution: Arch
Desktop environment: DWM

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Pranjalmet, 2021-07-19:
[Blueman-applet logs](https://pastebin.com/Rdz504tz)

cschramm, 2021-07-19:
The bluetoothd logs should have some indication what's failing.

Pranjalmet, 2021-07-20:
[This is log of process of pairing and connecting](https://pastebin.com/PkbPCzzy)
it connects and disconnects after few seconds, in blueman-manager it says Input/Output error that's it

cschramm, 2021-07-20:
blueman only fowards the error reported by bluetoothd (BlueZ). Its logs typically show some more details on what went wrong. It's not about blueman, so do not look in blueman logs.

closed 2021-09-26 by github-actions[bot]