blueman-project/blueman #1569: Toolbar & Statusbar toggles dont work in Blueman 2.2.1

CTZxVULKAN, 2021-07-22:
blueman : 2.2.1
BlueZ : 5.60-1
Distribution : Artix Linux [Open-Rc]
Desktop environment: i3-gaps (wm)

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Issue : 

WIth the Blueman 2.2.1 update Toolbar/Status bar toggles dont function in their expected way.

My toolbar was set to `False` under `View ->Show Toolbar` which works succesfully when the application is launched for the first time. However after quiting the application and relaunching the `Show Toolbar` is set set to `True`. This cannot be changed by the user anymore using the toggles, it can be only resolved after a restart. The same applies to `status bar` under `View -> Show Statusbar` where its visibility is set to `True`.

Apart  from this notifications work fine however only one bar is showed in the connected devices ( previous versions 3 bars were shown on connected devices showing signal information)

This bug did not exist in the previous verison. Kindly look at this issue.

cschramm, 2021-07-23:
It seems like #1563 fixes those things.

CTZxVULKAN, 2021-07-23:
I see but what about the toggles?
Anyway i will test out the code and try to patch them myself if possible. :)

cschramm, 2021-07-23:
They actually work for me with that fix (I could replicate the problem without it). I guess the failing icon logic breaks the whole UI somehow, not sure. 🤔

CTZxVULKAN, 2021-07-23:
So apparantly after a system update idk why everything seems to be fixed. The toggles just work fine now for some reason. Yesterday i tested the problem multiple times and now it doesnt exist ?!!! 
To cross check i went to my manjaro XFCE system (also upto date)  and installed blueman there and the problem didnt exist there too.
Might have been an issue on my side with i3wm.

I think you should close this issue @cschramm and just work on #1563.
Thanks mate you provided great support 👍🏽

cschramm, 2021-07-23:
Got it: If you start blueman-manager with a connected device, the call at [blueman/main/Manager.py:161](https://github.com/blueman-project/blueman/blob/2.2.1/blueman/main/Manager.py#L161) fails due to the unexpected file names and the two dconf settings just do not get bound. So I can confirm that #1563 is a fix for this.

CTZxVULKAN, 2021-07-24:
Ahh i see thanks man great support ! :)

closed 2021-07-24 by CTZxVULKAN