blueman-project/blueman #1573: Connect / disconnect notifications don't close automatically

abelsromero, 2021-07-24:
blueman: 2.2.1
BlueZ: 5.60-1
Distribution: Manjaro
Desktop environment: bspwm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I suspect this is releated with the 2.2 change mentioned in the release notes as "Desktop notifications on connect / disconnect (plugin)".
Today, after a Manjaro update I get a popup with the message "Connected" or "Disconnected" that does not close and stays in the screen. Not only that, if I connect disconnect the devise, the popups acumulate as seen in the image attached.

Issues a part, I think displaying this notification should be an option that users should be able disable from the UI preferences panel. Som devices like headphones already have acustic clues confirming connection. Furthermore, the popup is intrusive in some activites like watching a movie (specially if you have to stand up to go the the PC and close it :%).

[image]

WholesomeDoktor, 2021-07-27:
I agree as well. I have to xkill the window everytime it pops up. This should be an option whether to have notification or not, even if it is at least a 5 sec timer to automatically close.

WholesomeDoktor, 2021-07-27:
> blueman: 2.2.1
> BlueZ: 5.60-1
> Distribution: Manjaro
> Desktop environment: bspwm
>
> * [x]  I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and [...]

If you check the View tab and then plugin settings, there you should be able to deselect the notification plugin. Hope it helps.

abelsromero, 2021-07-27:
Thanks! For anyone with the same issue, the exact plugin to disable is `ConnectionNotifier`

[image]

closed 2021-07-27 by abelsromero

ultratensai, 2021-08-03:
Is it possible to configure ConnectionNotifier to automatically close the pop up window after X amount of time?
It seems I need to click on the pop up to close it.

cschramm, 2021-08-04:
@ultratensai: blueman sends a standard desktop notification with no explicit expiration time, meaning the notification server's settings shall be used. What notification daemon do you use?

ultratensai, 2021-08-10:
@cschramm thanks, i had to adjust the settings on mako!

cschramm, 2021-08-11:
@ultratensai: :thinking: Looks like mako's default is no timeout. blueman's assumption currently is that there is a default timeout and that seems to hold for the common notification daemons. We might do better for mako and others by setting an explicit timeout from our side but we'd have to choose something the user has to live with while currently it's possible to set the preferred default.