blueman-project/blueman #1576: blueman bluetooth authentication authorization request dialogue pops ignores "Always Accept" or "Accept" input

kenfallon, 2021-07-26:
[image]
blueman: 2.2.1 (1:2.2.1-1.fc34.x86_64)
BlueZ: 
Name         : bluez
Version      : 5.60 
Release      : 1.fc34
Architecture : x86_64

Distribution: Fedora release 34 (Thirty Four)

Desktop environment: 
LXQt Version 0.16.0
Qt 5.15.2

Name         : openbox
Version      : 3.6.1
Release      : 17.fc34
Architecture : x86_64

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Turn on previously paired device 
The dialogue appears showing "Always Accept" or "Accept" click
```
blueman-applet 11.38.50 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_70_26_05_XX_XX_XX {'Connected': True}
blueman-applet 11.38.50 INFO     BluezAgent:310 _on_authorize_service: Agent.Authorize
```
The application then eventually times out and crashes

```
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/blueman/main/indicators/GtkStatusIcon.py", line 45, in <lambda>
    gtk_item.connect('activate', cast(Callable[[Gtk.MenuItem], None], lambda _, idx=index: activate(idx)))
  File "/usr/lib/python3.9/site-packages/blueman/main/Tray.py", line 45, in _activate_menu_item
    AppletService().ActivateMenuItem('(ai)', indexes)
  File "/usr/lib/python3.9/site-packages/gi/overrides/Gio.py", line 349, in __call__
    result = self.dbus_proxy.call_sync(self.method_name, arg_variant,
gi.repository.GLib.Error: g-io-error-quark: Timeout was reached (24)

```
The workaround is to cancel then disconnect the headphones and manually connect in blueman-manager.

```
blueman-applet 11.39.15 INFO     BluezAgent:221 _on_cancel: Agent.Cancel
blueman-applet 11.39.18 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_70_26_05_F2_66_B8 {'Connected': False}
blueman-applet 11.39.20 INFO     RecentConns:175 notify    : <Device.Device object at 0x7efed54a91c0 (blueman+bluez+Device+Device at 0x555722563960)> 00000000-0000-0000-0000-000000000000
blueman-applet 11.39.20 INFO     RecentConns:107 initialize: rebuilding menu
blueman-applet 11.39.22 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_70_26_05_F2_66_B8 {'Connected': True}
blueman-applet 11.39.22 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_70_26_05_F2_66_B8 {'ServicesResolved': True}
```
[image]

zaps166, 2021-08-17:
I have similar problem with Blueman 2.2 - buttons from notification are ignored. As a workaround you can install Blueman 2.1 or temporarily disable system notification manager: `dconf write /org/blueman/general/notification-daemon false`.

cschramm, 2021-08-28:
Confirmed bug. Caused by 17df1f22 and fixed in #1596.

PR #1596 (2021-10-03): Fix service authorization notifications
Closes #1576
commit d4603519 (2021-10-03): Fix service authorization notifications

Closes #1576
closed 2021-10-03 by cschramm