blueman-project/blueman #1582: Resource not ready

lxgr-linux, 2021-08-03:
blueman: 2.2.1-1
BlueZ: 5.60-1
Distribution: Arch Linux
Desktop environment: Xfce4

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hello,
sometimes when trying to connect to any Bluetooth device I get a "Resource Not Ready" banner in blueman. And when searching for devices I get the same thing, but am able to open the error message. This happens randomly. 

[image]

senorsnor, 2021-08-13:
I have the same using the same environment. In particular, this happens after my laptop suspends, and the result is that devices do not connect again after suspending. Running blueman with loglevel debug doesn't show any info.

This problem started occurring recently; downgrading blueman to 2.1.4 solves it for me.

lxgr-linux, 2021-08-16:
Since v2.2.2 this issue did not occur again.

closed 2021-08-16 by lxgr-linux