blueman-project/blueman #1585: Connect does not pair and connect

ChristFollower3, 2021-08-08:
blueman:
BlueZ:
Distribution:
Desktop environment:

When you click connect it does not pair and connect.  You have to pair first, then connect.  Either disable connect until paired or make connect pair first and then connect.

cschramm, 2021-08-23:
Paring and connecting are different things. The services that you can connect to with the connect action might not even need pairing. The devices' manuals should give directions what to do.

Byzantine-1, 2021-09-09:
Good day, 
For some reason, my devices show up on blueman, but won't pair or connect. I will attach the code I am receiving. Thank you.
[image]

cschramm, 2021-09-17:
@Byzantine-1: That does not seem related at all. Please open a separate issue and provide all the basic information that we ask for in the template.

closed 2021-11-24 by github-actions[bot]