blueman-project/blueman #1586: Doesn't show active devices

saugatsn, 2021-08-09:
The bluetooth adater seems to be working correctly but it doesn't show active bluetooth devices. So, I cannot connect to my device. How do I fix this?
blueman:
BlueZ:
Distribution:
Desktop environment: Parrot OS

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

tomazza, 2021-08-22:
Hi.

I've been having the same problems on three different installations, with Blueman 2.2.2-1 on Bluez 5.60-1. It is on three different Arch based distributions as well, Manjaro, Arco Linux and Arch Linux itself. 

I solved it by open bluetoothctl in terminal and and typed the command "scan on", without the quotes. When it had detected the device i wanted to connect, it was also detected in Blueman and I could connect to it. After that, I had no further issues connecting additional devices directly through Blueman.

cschramm, 2021-08-28:
So the search action in blueman did not work?

tomazza, 2021-08-28:
No not until I started the search through bluetoothctl. After that it worked just fine in Blueman.

cschramm, 2021-08-29:
When you hit the search button in blueman-manager, you should see `Discovering: yes` for the adapter in bluetoothctl and then devices showing up. If that does not work, please start blueman-applet with `--loglevel debug` and then blueman-manager with the same argument and provide the output starting where you hit the button.

closed 2021-11-05 by github-actions[bot]