blueman-project/blueman #1587: Unable to switch between High Fedility playback to Headset Unit

CTZxVULKAN, 2021-08-09:
blueman: 2.2.2
BlueZ: 5.60
Distribution: Artix (OpenRc)
Desktop environment: i3-wm

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I recently discovered that blueman can not switch audio profiles between A2DP and HSP. This does not allow headsets with built in mics to be used and can be only used for listening audio.
 
This issue was originally discovered in the release `2.2.1`. 
I was **not sure** wether it was an issue with **my headset** or blueman and waited for the next release to check if the problem still existed in there. This issue is still persistent in release  `2.2.2`. 

I am presented with the following message :
```
Failed to change profile to headset_head_unit
```
Below is a screenshot trying to recreate the problem.
Any help in this matter would be highly appreciated.

[image]

cschramm, 2021-08-11:
That menu is just an interface to pulseaudio's profile switching. You can find the very same e.g. in `pavucontrol`. If that fails as well, it's not a blueman issue.

You can find a couple of previous discussions of such issues at https://github.com/blueman-project/blueman/search?q=%22Failed+to+change+profile+to+headset_head_unit%22&type=issues.

One thing is limited HFP support. Connecting to HFP headsets is possible via oFono since pulseaudio 11.0 but not plug-and-play. To my understanding pulseaudio 15.0 finally has a native HFP backend that does not require oFono to be set up.

Native support for connecting to HSP headsets was already introduced in pulseaudio 6.0, so HSP should work in general.

closed 2021-08-12 by CTZxVULKAN

cschramm, 2021-08-12:
@CTZxVULKAN: Did you find a solution?