blueman-project/blueman #1590: Cannot use Bluetooth devices Ubuntu 20.04

liorko87, 2021-08-15:
blueman: 2.1.2
Distribution: Ubuntu 20.04.1

I'm trying to use my JBL live 300 TWS earphones.
Sometimes it works, sometimes it's not.
You can see the log from the blueman-applet [here](https://gist.github.com/liorko87/8797a42572b8c6b94f697f8452d3bf52).

* [V] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2021-08-23:
There's no connection attempt in the log. What happens?

liorko87, 2021-08-23:
When I'm trying ti open the `blueman-manager` via the terminal, this error appeared:
```
blueman-manager version 2.1.2 starting
blueman-manager 08.47.43 ERROR    Manager:118 on_dbus_name_appeared: Default adapter not found, trying first available.
blueman-manager 08.47.43 ERROR    Manager:122 on_dbus_name_appeared: No adapter(s) found, exiting
```

cschramm, 2021-08-29:
So blueman-manager does not work for you at all? How can you connect sometimes then?

liorko87, 2021-08-29:
Sometimes it's works and open correctly, sometimes it doesn't work at all.
When it's not working the error mentioned above appeared.

cschramm, 2021-08-29:
So this is actually about blueman sometimes not finding an adapter? Probably not a blueman issue, but let's try to get behind it. When it happens, does `bluetoothctl list` show an adapter / controller? If not, try to find it in lspci / lsusb and check `dmesg` for issue.

liorko87, 2021-08-29:
The `dmesg` outputs a lot of lines with:
`[ 5003.882438] usb 2-2-port3: Cannot enable. Maybe the USB cable is bad?`

When I'm looking for Bluetooth this is the only message that appears:
```
dmesg | grep blue
[    2.506528] thinkpad_acpi: rfkill switch tpacpi_bluetooth_sw: radio is unblocked
```

cschramm, 2021-08-29:
Well, if USB 2:2:3 is your Bluetooth adapter, you found the problem.

If you want to grep `dmesg`, use `Bluetooth`, with a capital `B`.

liorko87, 2021-08-29:
How can i check if USB 2:2:3 is indeed my Bluetooth adapter?
I'm using Lenovo T490 with docking station that connected with USB type C.

cschramm, 2021-08-29:
`lsusb` should tell you.

liorko87, 2021-08-30:
Thanks for the help, this is the problem, the type C adapter to the docking station.

closed 2021-08-30 by liorko87